dsmrreader/dsmr-reader #1761: 🙋k8s: opstarten zonder USB device

JimKlapwijk, 2022-12-02:
### Description

Hoi,

Ik gebruik DSMR Reader met ser2net. Ik wilde mijn PostgreSQL upgraden van 14 naar 15, maar dit lukte niet, en ben ik al mijn data kwijt, pech, dus nu opnieuw beginnen!

Ik verwijder mijn PostgreSQL en DSMR Reader namepsace, en rol mijn helm charts opnieuw uit. Echter blijft ik de error krijgen:

```
Starting DSMR Reader - datalogger...
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
2022-12-02 20:36:51,541 ERROR    mixins       run_once                        103 | dsmr_datalogger.management.commands.dsmr_datalogger: [!] Exception raised. Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/serial/serialposix.py", line 322, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/ttyUSB0'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/app/dsmr_datalogger/scripts/dsmr_datalogger_api_client.py", line 37, in read_telegram
    serial_handle = serial.serial_for_url(url=url_or_port, timeout=MAX_READ_TIMEOUT, **serial_kwargs)
  File "/usr/local/lib/python3.10/site-packages/serial/__init__.py", line 90, in serial_for_url
    instance.open()
  File "/usr/local/lib/python3.10/site-packages/serial/serialposix.py", line 325, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/app/dsmr_backend/mixins.py", line 92, in run_once
    self.run(data=self.data, **options)
  File "/app/dsmr_datalogger/management/commands/dsmr_datalogger.py", line 29, in run
    telegram = next(self.telegram_generator)
  File "/app/dsmr_datalogger/scripts/dsmr_datalogger_api_client.py", line 39, in read_telegram
    raise RuntimeError('Failed to connect: {}', error) from error
RuntimeError: ('Failed to connect: {}', SerialException(2, "could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'"))
```

Wat uiteraard logisch is, want ik heb helemaal geen /dev/ttyUSB0.
Hoe start ik een schone installatie zonder USB device op K8s/Docker?

Ik probeer dit met DSMR 5.4 en 5.9.

Alvast bedankt! 

### DSMR-reader version

5.4

### DSMR-reader platform

Docker (e.g. Xirixiz)

dennissiemensma, 2022-12-02:
Mijn kennis van Docker is helaas beperkt. Je kunt nog @Xirixiz vragen

xirixiz, 2022-12-03:
Yes, daar kan ik bij helpen :).

Check de documentatie hier even. Default draait DSMR Reader in standalone mode, wat inhoudt dat alles op 1 host draait. Dus dan wordt /dev/ttyUSB0 ook verwacht. 
https://github.com/xirixiz/dsmr-reader-docker

Ik denk dat je hier wel verder mee komt:
https://github.com/xirixiz/dsmr-reader-docker#dsmr-datalogger-related

JimKlapwijk, 2022-12-03:
Ik merkte inderdaad dat met `DSMRREADER_REMOTE_DATALOGGER_MODE=receiver ` de container wel doorstart.
Is dit een nieuwe optie? Ik kan me niet herinneren dat ik hier eerder problemen mee heb gehad.

Bedankt voor de snelle reactie in ieder geval! :)

closed 2022-12-04 by dennissiemensma