dsmrreader/dsmr-reader #1763: Drop support for PostgreSQL 10.x & 11.x

dennissiemensma, 2022-12-06:
### Feature

# https://www.postgresql.org/support/versioning/

- PostgreSQL 10.x is net EOL gegaan eind november.
- Overwegen om zelfs 11.x niet meer te ondersteunen, gezien die komende jaar EOL gaat.

dennissiemensma, 2023-01-31:
Overwegen zelfs 12.x niet te ondersteunen, in lijn met het project van Xirixiz. 13 of 14 is wel te doen.

dennissiemensma, 2023-07-21:
[image]

dennissiemensma, 2023-07-21:
Postgresql 14+ lijkt me prima.

dennissiemensma, 2023-07-21:
Ik houd het wel bij droppen van 3 major versies. v13 is nog tot eind '25

dennissiemensma, 2024-06-03:
Postgres 12 gaat dit jaar EOL, daarom is het denk ik handig om postgresl 15 minimaal te ondersteunen, al dwingt DSMR-reader dat (zelf) niet af.
[image]

dennissiemensma, 2024-06-03:
Onderdeel van v6, al kan ik het niet hard afdwingen, alleen aankondigen

niekniek89, 2025-12-06:
Enkel een vraag, en geen idee of dit daarvoor de juiste plaats is;
Ik draai al jaren tevreden de DSMR reader applicatie in Docker.
Daarbij zie ik dat mijn PostgreSQL versie 13.x is, en ik eigenlijk moet upgraden (maar weet niet zo goed hoe, en heb dat nooit zelf gedaan).

Is het een harde eis voor de werking, dat je perse op 15 of hoger moet zitten?
of kan ik straks gewoon upgraden naar v6.x met PostgreSQL 13.x?

dennissiemensma, 2025-12-07:
@niekniek89 upgrades van DSMR-reader upgraden ook onderliggende software, die weer andere nieuwere onderliggende software vereist (zoals Python of een nieuwere database). Het grootste deel zit in de container, waar je zelf niets voor hoeft te doen.

Alleen de database is een uitzondering. Gelukkig is die upgrade relatief makkelijk en hoeft die ook niet jaarlijks. Advies is om straks naar versie 17 te upgraden, dan zit je weer een paar jaar goed tot 2029:
- https://www.postgresql.org/support/versioning/

How-to staat hier, maar wacht nog even, want Xirixiz moet zelf ook ervoor zorgen dat de DSMR-reader-container wel kan verbinden met nieuwere postgresql versies. In DSMR-reader 6 wordt dat sowieso PG versie 17 als advies-versie als het lukt dat werkend te krijgen.
- https://github.com/xirixiz/dsmr-reader-docker?tab=readme-ov-file#postgres-upgrade-docker

dennissiemensma, 2025-12-30:
@niekniek89 het hangt er oa vanaf welke versie de container van Xirixiz ondersteunt, maar 16 zou inderdaad moeten werken!

Dan kun je er weer twee jaar mee vooruit en tegen die tijd upgraden naar 17, 18 of 19 is waarschijnlijk (bijna) even weinig werk.