dsmrreader/dsmr-reader #1766: 🙋Upgrade 5.9 gaat fout op raspberry pi, geeft tijdens upgrade foutmelding _psycopg.cpython-39-arm-linux-gnueabihf.so: undefined symbol: PQencryptPasswordConn

Tridax75, 2022-12-09:
### Description

Upgraden naar versie 5.9 gaat fout, hieronder complete logging:
--- You are currently running version: 
5.9

 --- Checking Python version.
Running Python 3.9.9 (main, Mar 20 2022, 11:32:05) 
[GCC 6.3.0 20170516]
[WARNING] Your Python version can still be used with your current DSMR-reader version. However, Python 3.11.0 or higher is preferred. Consider using it for new installations in the future.

 --- Checking for local file changes.

 --- Pulling remote repository for new commits...

 --- Merging/updating checkout.
Already up-to-date.

 >>> Running post-deployment script. <<<

 --- Checking whether '.venv' VirtualEnv is activated.
     [i] ----- Activating '.venv' VirtualEnv...

 --- Checking (minimum) Python version.
Running Python 3.9.9 (main, Mar 20 2022, 11:32:05) 
[GCC 6.3.0 20170516]
[WARNING] Your Python version can still be used with your current DSMR-reader version. However, Python 3.11.0 or higher is preferred. Consider using it for new installations in the future.

 --- Checking & synchronizing base requirements for changes.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: coreapi==2.3.3 in ./.venv/lib/python3.9/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 1)) (2.3.3)
Requirement already satisfied: crcmod==1.7 in ./.venv/lib/python3.9/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 2)) (1.7)
Requirement already satisfied: django==3.2.16 in ./.venv/lib/python3.9/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 3)) (3.2.16)
Requirement already satisfied: django-admin-rangefilter==0.8.8 in ./.venv/lib/python3.9/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 4)) (0.8.8)
Requirement already satisfied: django-admin-sortable==2.3 in ./.venv/lib/python3.9/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 5)) (2.3)
Requirement already satisfied: django-colorfield==0.7.2 in ./.venv/lib/python3.9/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 6)) (0.7.2)
Requirement already satisfied: django-debug-toolbar==3.7.0 in ./.venv/lib/python3.9/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 7)) (3.7.0)
Requirement already satisfied: django-filter==21.1 in ./.venv/lib/python3.9/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 8)) (21.1)
Requirement already satisfied: django-mathfilters==1.0.0 in ./.venv/lib/python3.9/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 9)) (1.0.0)
Requirement already satisfied: djangorestframework==3.14.0 in ./.venv/lib/python3.9/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 10)) (3.14.0)
Requirement already satisfied: django-solo==2.0 in ./.venv/lib/python3.9/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 11)) (2.0.0)
Requirement already satisfied: dropbox==11.36.0 in ./.venv/lib/python3.9/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 12)) (11.36.0)
Requirement already satisfied: gunicorn==20.1.0 in ./.venv/lib/python3.9/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 13)) (20.1.0)
Requirement already satisfied: influxdb-client==1.34.0 in ./.venv/lib/python3.9/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 14)) (1.34.0)
Requirement already satisfied: paho-mqtt==1.6.1 in ./.venv/lib/python3.9/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 15)) (1.6.1)
Requirement already satisfied: psycopg2-binary==2.9.5 in ./.venv/lib/python3.9/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 16)) (2.9.5)
Requirement already satisfied: pyserial==3.5 in ./.venv/lib/python3.9/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 17)) (3.5)
Requirement already satisfied: pyserial-asyncio==0.6 in ./.venv/lib/python3.9/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 18)) (0.6)
Requirement already satisfied: python-dateutil==2.8.2 in ./.venv/lib/python3.9/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 19)) (2.8.2)
Requirement already satisfied: python-decouple==3.6 in ./.venv/lib/python3.9/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 20)) (3.6)
Requirement already satisfied: pytz==2022.6 in ./.venv/lib/python3.9/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 21)) (2022.6)
Requirement already satisfied: pyyaml==6.0 in ./.venv/lib/python3.9/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 22)) (6.0)
Requirement already satisfied: requests==2.28.1 in ./.venv/lib/python3.9/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 23)) (2.28.1)
Requirement already satisfied: urllib3==1.26.13 in ./.venv/lib/python3.9/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 24)) (1.26.13)
Requirement already satisfied: uritemplate in ./.venv/lib/python3.9/site-packages (from coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 1)) (4.1.1)
Requirement already satisfied: itypes in ./.venv/lib/python3.9/site-packages (from coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 1)) (1.2.0)
Requirement already satisfied: coreschema in ./.venv/lib/python3.9/site-packages (from coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 1)) (0.0.4)
Requirement already satisfied: sqlparse>=0.2.2 in ./.venv/lib/python3.9/site-packages (from django==3.2.16->-r dsmrreader/provisioning/requirements/base.txt (line 3)) (0.4.2)
Requirement already satisfied: asgiref<4,>=3.3.2 in ./.venv/lib/python3.9/site-packages (from django==3.2.16->-r dsmrreader/provisioning/requirements/base.txt (line 3)) (3.5.0)
Requirement already satisfied: Pillow>=5.0 in ./.venv/lib/python3.9/site-packages (from django-colorfield==0.7.2->-r dsmrreader/provisioning/requirements/base.txt (line 6)) (9.0.1)
Requirement already satisfied: six>=1.12.0 in ./.venv/lib/python3.9/site-packages (from dropbox==11.36.0->-r dsmrreader/provisioning/requirements/base.txt (line 12)) (1.16.0)
Requirement already satisfied: stone>=2.* in ./.venv/lib/python3.9/site-packages (from dropbox==11.36.0->-r dsmrreader/provisioning/requirements/base.txt (line 12)) (3.3.1)
Requirement already satisfied: setuptools>=3.0 in ./.venv/lib/python3.9/site-packages (from gunicorn==20.1.0->-r dsmrreader/provisioning/requirements/base.txt (line 13)) (58.1.0)
Requirement already satisfied: reactivex>=4.0.4 in ./.venv/lib/python3.9/site-packages (from influxdb-client==1.34.0->-r dsmrreader/provisioning/requirements/base.txt (line 14)) (4.0.4)
Requirement already satisfied: certifi>=14.05.14 in ./.venv/lib/python3.9/site-packages (from influxdb-client==1.34.0->-r dsmrreader/provisioning/requirements/base.txt (line 14)) (2021.10.8)
Requirement already satisfied: idna<4,>=2.5 in ./.venv/lib/python3.9/site-packages (from requests==2.28.1->-r dsmrreader/provisioning/requirements/base.txt (line 23)) (3.3)
Requirement already satisfied: charset-normalizer<3,>=2 in ./.venv/lib/python3.9/site-packages (from requests==2.28.1->-r dsmrreader/provisioning/requirements/base.txt (line 23)) (2.0.12)
Requirement already satisfied: typing-extensions<5.0.0,>=4.1.1 in ./.venv/lib/python3.9/site-packages (from reactivex>=4.0.4->influxdb-client==1.34.0->-r dsmrreader/provisioning/requirements/base.txt (line 14)) (4.3.0)
Requirement already satisfied: ply>=3.4 in ./.venv/lib/python3.9/site-packages (from stone>=2.*->dropbox==11.36.0->-r dsmrreader/provisioning/requirements/base.txt (line 12)) (3.11)
Requirement already satisfied: jinja2 in ./.venv/lib/python3.9/site-packages (from coreschema->coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 1)) (3.1.1)
[... 212 lines cut ...]
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/contrib/auth/models.py", line 3, in <module>
    from django.contrib.auth.base_user import AbstractBaseUser, BaseUserManager
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/contrib/auth/base_user.py", line 48, in <module>
    class AbstractBaseUser(models.Model):
[... cut ...]
  File "/usr/local/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/db/backends/postgresql/base.py", line 29, in <module>
    raise ImproperlyConfigured("Error loading psycopg2 module: %s" % e)
django.core.exceptions.ImproperlyConfigured: Error loading psycopg2 module: /home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/psycopg2/_psycopg.cpython-39-arm-linux-gnueabihf.so: undefined symbol: PQencryptPasswordConn
```

dennissiemensma, 2022-12-09:
Ik heb hier helaas geen oplossing voor. Dit is meerdere malen gemeld:

- #1707
- #1649

En in beide gevallen is het mij niet duidelijk wat het precies was of heb ik geen reactie meer gehad.

Je kunt nog kijken of je hier wijzer van wordt:
- https://github.com/psycopg/psycopg2/issues/1462

Tridax75, 2022-12-10:
Probleem inmiddels opgelost, ik heb mijn Raspberry Pi handmatig geupgrade van Stretch naar Buster. 
https://raspberrypi.tilburgs.com/upgrade-stretch-to-buster/

Hiermee wordt Postgresql 9.6 vervangen en gaat de upgrade naar 5.9 weer goed.

closed 2022-12-10 by Tridax75