dsmrreader/dsmr-reader #1767: Na de upgrade naar 5.9 krijg ik een (foutmelding) melding in het over scherm

andyesten, 2022-12-11:
### Description

Na de upgrade naar 5.9 krijg ik in het ' over' scherm de volgende (fout) melding:

   Ongeveer None metingen opgeslagen, overweeg dataopschoning (indien nog niet ingeschakeld)   (nu)

Kan ik deze negeren? De dataopschoning staat op een maand.

Andy

### DSMR-reader version

5.9

### DSMR-reader platform

Docker (e.g. Xirixiz)

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                            v5.9 / v3.11.1 / mysql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                        ??? (mysql)
    Consumption records electricity / gas (est.)                                    ??? (mysql) / ??? (mysql)

UNRESOLVED ISSUES
    Ongeveer None metingen opgeslagen, overweeg dataopschoning (indien nog niet ingeschakeld)                                      nu
```

dennissiemensma, 2022-12-11:
Bedankt voor je melding. Werkt de rest verder wel? Zo ja, dan hoef je niets te doen.

Dit komt doordat je MySQL gebruikt. De kern van DSMR-reader zou overigens gewoon moeten werken, ook zonder PostgreSQL, maar dit is een onvoorzien bijeffect denk ik. Ik zal kijken of ik die specifieke check een keer wat kan verbeteren.

andyesten, 2022-12-11:
Hi,

Dank voor de snelle reactie.

> Werkt de rest verder wel? 

Ja zeker! Alles werkt verder zonder problemen.

> Ik zal kijken of ik die specifieke check een keer wat kan verbeteren

Top.

peternijssen, 2022-12-11:
Ik ervaar hetzelfde probleem, maar het doet inderdaad verder geen afbreuk aan de applicatie zelf :)

dennissiemensma, 2023-02-27:
In de volgende release wordt de melding niet meer getoond als het geen postgresql engine is.

closed 2023-02-27 by dennissiemensma