dsmrreader/dsmr-reader #1768: 🙋Nieuwe installatie met oude data (van v4.20 naar recente versie)

donnie-dominguez, 2022-12-11:
### Description

Ik draai DSMR-reader op een ubuntu server (v18) welke ik verprutst heb, en heb eigenlijk te weinig verstand van linux. DSMR reader draait overigens nog prima (v 4.20), maar krijg ik niet meer naar een nieuwere versie omdat ik python/postgresql niet meer naar een andere versie krijg.
Bestaat er een mogelijkheid om een verse installatie te doen van DSMR reader op een verse server en dan daarin de backup terug te zetten? De versie zal dan nogal verschillen neem ik aan. Ik ben ik veroordeeld om opnieuw te beginnen en afscheid te nemen van mijn historische data?

### DSMR-reader version

4.20

### DSMR-reader platform

Native (e.g. manual installation)

dennissiemensma, 2022-12-11:
Bedankt voor je vraag. Als je een goede backup hebt dan is er wel omheen te werken.

In principe kun je gewoon een verse installatie doen. Alleen moet je dan specifiek v4 kiezen:
- https://dsmr-reader.readthedocs.io/nl/v4/tutorial/installation/step-by-step.html

Ergens staat dan een stap als:
```
git clone --branch v4 https://github.com/dsmrreader/dsmr-reader.git
```

Ipv de laatste v5 versie:
```
git clone --branch v4 https://github.com/dsmrreader/dsmr-reader.git
```

----

De DB backup terugzetten doe je **eerder** in die stappen:
- https://dsmr-reader.readthedocs.io/nl/v4/tutorial/installation/step-by-step.html#optional-restore-a-database-backup

Er vanuitgaande dat de DB backup van v4.x is

dennissiemensma, 2022-12-11:
Zie dus overal expliciet `v4` in de bovenstaande URL's/command.

closed 2022-12-29 by dennissiemensma

donnie-dominguez, 2023-01-28:
Het heeft even mogen duren, maar ik ben eindelijk begonnen aan de klus. En het werkt als een zonnetje! Dat stappenplan is top. Ik draaide vorige week nog versie 4.20 op Ubuntu 18 lts en nu versie 5.10 op Ubuntu 22 lts.

Hoe ik dat heb aangepakt (wellicht hebben anderen hier nog wat aan):
- Verse Ubuntu 22 geïnstalleerd (als VM onder ESXi) en helemaal up-to-date gemaakt.
- Snapshot maken, zodat ik weer terug kan naar een 'kale' ubuntu mocht er iets mis gaan.
- Volledige nieuwe v4.20 installeren op de nieuwe server (met behulp van de tip 'git clone --branch v4 https://github.com/dsmrreader/dsmr-reader.git', en even de kabel van de slimme meter er aan gehangen om te kijken of het ook echt werkt.
- De database leegmaken:  
`sudo supervisorctl stop all

sudo -u postgres dropdb dsmrreader
sudo -u postgres createdb -O dsmrreader dsmrreader`
- Op de 'oude machine' een backup maken './manage.py dsmr_backup_create --full'
- Backup overzetten naar de nieuwe machine en importeren 'https://dsmr-reader.readthedocs.io/nl/v4/how-to/database/postgresql-restore-backup.html'
- Controleren of de nieuwe machine helemaal werkt (op v4.20)
- Nieuwe snapshot maken van de nieuwe machine (mocht er in de upgrade iets fout gaan)
- Daarna de upgrade van v4 naar v5 testen 'https://dsmr-reader.readthedocs.io/nl/v5/tutorial/upgrading/to-v5.html'
- Als deze goed lijkt te werken even de kabel van de slimme meter omzetten van de oude naar de nieuwe machine
- Oude machine uitzetten
- Weekje later, oude machine weggooien, snapshots van de nieuwe machine verwijderen
- Lekker genieten van de mooie applicatie, Dennis bedankt!

dennissiemensma, 2023-01-29:
Bedankt voor je uitgebreide stappen! En vooral fijn dat het je uiteindelijk gelukt is.