dsmrreader/dsmr-reader #1770: 🙋Totale dag gasverbruik in 'archive' wijkt af optelsom gasverbruik per uur (en vattenfall registratie)

nickgr6, 2022-12-18:
### Description

Beste,

Ik ben in het proces om mijn beeclear te vervangen door een device waarop ik dsmr reader ga draaien. Ter voorbereiding ben ik nu alle historische data aan het migreren mbv de api. Dit lijkt allemaal goed te zijn gegaan. 

Toen ik echter de totalen vergeleek tussen [beeclear](https://www.beeclear.nl/) (en vattenfall die gelijk zijn hieraan) en de dsmr interface, zag ik verschillen.

Het volgende toont in de archive voor 2 december:
[image]

De data view voor gasverbruik per uur toont:
```
 	Gas
0:00	0
1:00	0
2:00	0
3:00	0
4:00	0
5:00	0
6:00	0
7:00	0
8:00	0.019
9:00	0.219
10:00	0.598
11:00	0.485
12:00	0.149
13:00	0.112
14:00	0.279
15:00	0.007
16:00	0.044
17:00	0.076
18:00	0.07
19:00	0.026
20:00	0.02
21:00	0.136
22:00	0.083
23:00	0
0:00	0.475
```
Als ik die optel, kom ik uit op 2.798. Maar de interface toont 2.323. Het lijkt er dus op dat de laatste meting niet wordt meegenomen (0.475).  Dus mijn aanname was dat dat uur wellicht bij de volgende dag hoort aangezien de data view 2x 0:00 toont. Als ik echter de volgende dag bekijk zie ik: 
```
 	Gas
0:00	0.475
1:00	0
2:00	0
3:00	0
4:00	0
5:00	0
6:00	0
7:00	0
8:00	0.215
9:00	0.24
10:00	0.484
11:00	0.379
12:00	0.106
13:00	0
14:00	0.006
15:00	0.172
16:00	0.119
17:00	0
18:00	0.005
19:00	0.012
20:00	0.358
21:00	0.15
22:00	0.318
23:00	0.043
0:00	0
```
[image]
Hier komt het totaal van 2.607 echter wel overeen met mijn meting in beeclear/vattenfall en wordt 0.475 van de eerste 0:00 meting dus ook niet meegenomen in de optelsom. Het lijkt er dus op dat die meeting in beide dagen niet wordt meegenomen.

De dag rows in de `dsmr_stats_hourstatistics` db:
198069,2022-12-02 00:00:00.000000 +00:00,0.000,0.000,0.000,0.000,0.000
198070,2022-12-02 01:00:00.000000 +00:00,0.000,0.000,0.000,0.000,0.000
198071,2022-12-02 02:00:00.000000 +00:00,0.000,0.000,0.000,0.000,0.000
198072,2022-12-02 03:00:00.000000 +00:00,0.000,0.000,0.000,0.000,0.000
198073,2022-12-02 04:00:00.000000 +00:00,0.000,0.000,0.000,0.000,0.000
198074,2022-12-02 05:00:00.000000 +00:00,0.000,0.000,0.000,0.000,0.000
198075,2022-12-02 06:00:00.000000 +00:00,0.000,0.000,0.000,0.000,0.000
198076,2022-12-02 07:00:00.000000 +00:00,0.000,0.000,0.000,0.000,0.019
198077,2022-12-02 08:00:00.000000 +00:00,0.000,0.000,0.000,0.000,0.219
198078,2022-12-02 09:00:00.000000 +00:00,0.000,0.000,0.000,0.000,0.598
198079,2022-12-02 10:00:00.000000 +00:00,0.000,0.000,0.000,0.000,0.485
198080,2022-12-02 11:00:00.000000 +00:00,0.000,0.000,0.000,0.000,0.149
198081,2022-12-02 12:00:00.000000 +00:00,0.000,0.000,0.000,0.000,0.112
198082,2022-12-02 13:00:00.000000 +00:00,0.000,0.000,0.000,0.000,0.279
198083,2022-12-02 14:00:00.000000 +00:00,0.000,0.000,0.000,0.000,0.007
198084,2022-12-02 15:00:00.000000 +00:00,0.000,0.000,0.000,0.000,0.044
198085,2022-12-02 16:00:00.000000 +00:00,0.000,0.000,0.000,0.000,0.076
198086,2022-12-02 17:00:00.000000 +00:00,0.000,0.000,0.000,0.000,0.070
198087,2022-12-02 18:00:00.000000 +00:00,0.000,0.000,0.000,0.000,0.026
198088,2022-12-02 19:00:00.000000 +00:00,0.000,0.000,0.000,0.000,0.020
198089,2022-12-02 20:00:00.000000 +00:00,0.000,0.000,0.000,0.000,0.136
198090,2022-12-02 21:00:00.000000 +00:00,0.000,0.000,0.000,0.000,0.083
198091,2022-12-02 22:00:00.000000 +00:00,0.000,0.000,0.000,0.000,0.000
198092,2022-12-02 23:00:00.000000 +00:00,0.000,0.000,0.000,0.000,0.475

De daystatistics waardes in de database lijken ook correct:
[image]

Ook het opnieuw runnen (en stats opschonen en weer runnen) van `/app/manage.py dsmr_stats_reconstruct_missing_day_statistics` blijft hetzelfde resultaat geven.

Gezien dsmr reader al zo'n volwassen product is (complimenten voor alle opties :)!), ga ik er vanuit dat er ergens een fout zit in de migratie oid. Echter lijkt de data in de db correct. Vandaar dat ik toch dit ticket heb aangemaakt.

Enig idee waar het misgaat of wat ik zelf nog verder kan controleren om daar achter te komen? 

Alvast bedankt!

### DSMR-reader version

5.9

### DSMR-reader platform

Other

dennissiemensma, 2022-12-18:
Bedankt voor je melding en zeer uitgebreide beschrijving. De tijden van het gasverbruik _zouden_ altijd het begin van het uur moeten aangeven. `23:00` betekent dan "tussen 23:00 en 0:00". 
Echter is dat niet waterdicht, want gasmeters hebben soms wat minuten na de uurwisseling nodig voordat ze (of de slimme meter) bijwerken. Al *zou* DSMR-reader dat wel moeten afvangen.

- Heb je trouwens oude metingen geimporteerd en laten verwerken, of direct de uren en dagen geschreven?

- Ik denk dat het voor nu het makkelijkste is om te kijken of je dezelfde issues hebt wanneer je DSMR-reader wel live metingen laten verwerken.

nickgr6, 2022-12-19:
Oude metingen heb ik via de api ingeschoten en laten verwerken. Die tabel miste nog in de omschrijving:
`dsmr_datalogger_dsmrreading`:
```
2022-12-02 22:00:00.000000 +00:00,6151.057
2022-12-02 23:00:00.000000 +00:00,6151.532
```

Nu ik nog eens naar de dates kijk, valt me wel iets op. Volgens mij is deze kolom een UTC timestamp. Als ik `SHOW TIMEZONE;` query run, krijg ik namelijk `UTC`. Aangezien de wintertijd in NL UTC + 1 is, zou dit wellicht betekenen dat als je een 'nederlandse dag' queried, `2022-12-02 23:00:00.000000 +00:00`er buiten valt, aangezien dat in NL tijd neerkomt op `2022-12-03 00:00:00.000000 +01:00`, toch..? Maar ook nadat ik daarmee speel komen de totalen nog niet helemaal overeen na migratie. Ik verwacht dus dat er ook nog ergens een mismatch is in mijn script mbt jou opmerking "De tijden van het gasverbruik zouden altijd het begin van het uur moeten aangeven. 23:00 betekent dan "tussen 23:00 en 0:00".  

De nieuwe hardware waar ik dsmr reader op ga runnen is helaas nog niet gereed, dus kan ik nog geen live metingen mee testen.

Anyway, ik ben er weer even klaar mee voor vandaag. Ik hou je op de hoogte van de verdere ontwikkelingen

dennissiemensma, 2022-12-19:
Als het goed is slaat Postgres altijd timezone aware datetimes op als UTC. Wat je daarna terugkrijgt bij ophalen hangt af van je sql client en of je een tijdzone opgeeft bij verbinden.  Zie de queries in: https://github.com/dsmrreader/dsmr-reader/issues/1217#issuecomment-981156657

- Welke tijdzone gebruik je bij het inschieten via de API? DSMR-reader staat in principe alles toe, zolang ze relatief gezien maar kloppen (lokale tijd of UTC).

Verder kan het wel eens een edge-case zijn hoor, want als je de meterstanden uit je screenshot optelt, kom je op `6151.532 - 6148.734 = 2.798` van je sheet.

- Wat zijn de DB records in `dsmr_consumption_gasconsumption` rond middernacht tussen 2 en 3 december?

nickgr6, 2022-12-20:
Ik converteer het naar een UTC timestamp voor ik het in de api inschiet. Waarbij er rekening gehouden wordt met zomer/wintertijd. 

Aangezien ik merkte dat consequent alleen het laatste uur van de dag niet werd meegeteld in het totaal dagverbruik,  heb ik nog een extra uur van de beeclear tijd afgehaald bij omzetten van de timestamp naar utc Dit lost het probleem op voor dagen met verbruik in het laatste uur (zelfde geldt voor elektra). Echter werden daarna juist de dagen weer met verbruik in het eerste uur incorrect (wat daarvoor nog goed ging).

vertalen gaan de dagen goed waarbij er geen gas (zelfde geldt overigens voor elektra) verbruikt is in het eerste uur, dus dat geldt voor bijv 2 december:

[image]

De uur grafiek komt dan ook overeen met die van beeclear:
[image]

[image]

Nu gaat het dus bijv. weer mis op 4 december:
[image]
Hoewel ook hier de uur grafiek matched met die van beeclear:
[image]

Echter is de 'Meterstand bij eerste meting van de dag' anders dan in beeclear:
[image]
Hij pakt dus de 2e meterstand: 6.154,474 m3

Voorbeeld van conversie:
```
beeclear 03-12-2022 23:00
offset -120
result 2 2022-12-03T21:00:00.000Z
meter 6154.139

beeclear 04-12-2022 00:00
offset -120
result 2 2022-12-03T22:00:00.000Z
meter 6154.139

beeclear 04-12-2022 01:00
offset -120
result 2 2022-12-03T23:00:00.000Z
meter 6154.474

beeclear 04-12-2022 02:00
offset -120
result 2 2022-12-04T00:00:00.000Z
meter 6154.474
...

beeclear 05-12-2022 00:00
offset -120
result 2 2022-12-04T22:00:00.000Z
meter 6157.798
```

Bij deze even een export van de belangrijkste db tabellen:
[db-export.zip](https://github.com/dsmrreader/dsmr-reader/files/10271842/db-export.zip)

En de beeclear gas csv export wat de bron is voor de migratie:
[gas.csv](https://github.com/dsmrreader/dsmr-reader/files/10271853/gas.csv)

Wellicht valt je iets op?

dennissiemensma, 2022-12-21:
Bedankt voor je aanvulling. Ik kan nu niet zo 1-2-3 in de data kijken. 

Mijn vermoeden is nog steeds dat als er een optelling niet klopt, het een edge-case is qua bounds in DSMR-reader. 
Daar is eerder ooit iets in gekomen wat niet hoorde. Dus dat is gerevert via 4a1d40d3592d84c5dfb16fe47e071d88b9ce7c06.

Voor de duidelijkheid, wat klopt er nunog steeds niet?
- Het gasverbruik per uur?
- Het gasverbruik per dag?
- De eerste (gas)meterstanden van de dag?
- Alle bovenstaande?

Is het mogelijk om zowel DSMR-reader als BeeClear de meter te laten uitlezen voor een paar dagen? Je kunt DSMR-reader immers makkelijk leeggooien totdat de oorzaak boven water is.

nickgr6, 2022-12-21:
Het gasverbruik per uur gaat goed. 

Wat er nog niet goed gaat in het vb. van 2 december:

1. Meterstanden bij eerste meting van de dag Gas: 6.154,474 
in beeclear matched dit met de 2e meting van de dag: 01:00 04-12-2022. Ik zou echter de eerste waarde zoals in beeclear verwachten: 00:00 04-12-2022 6.154,139 m3  

2. Het totaal gasverbruik van deze dag toont 3,324 in dsmr. In beeclear is het totaal 3,659. Hier zit dus 0,335 verschil tussen. Dit is precies het verschil tussen 00:00 en 01:00 op die dag zoals wel weergegeven in de dsmr uur grafiek: 

[image]
En de data view:
```
 	Gas
0:00	0.335
1:00	0
2:00	0.188
3:00	0.282
4:00	0.213
5:00	0.14
6:00	0
7:00	0.121
8:00	0.44
9:00	0.498
10:00	0.311
11:00	0.005
12:00	0.005
13:00	0.005
14:00	0.104
15:00	0.256
16:00	0.261
17:00	0.247
18:00	0
19:00	0.012
20:00	0.074
21:00	0
22:00	0.162
23:00	0
0:00	0
```
Als je alle waardes van de dataview optelt, kom je ook uit op 3,659.

dennissiemensma, 2022-12-21:
Ik weet niet helemaal zeker of ik onderstaade goed intepreteer, maar als ik je SQL datadump goed lees dan komt de eerste meterstand op 4 december overeen met die `6.154,474`.

[image]

Dat lijkt de waarde op 23:00 UTC de dag ervoor, dus middernacht onze tijd/CET. Dat uur `0,335` zou dan zelfs nog bij de vorige dag 3 december horen. van `23:00 - 00:00` CET (of `22:00 - 23:00` UTC). 
Dan nog is het vreemd dat die in DSMR-reader op 4 december komt en niet 3 december.

Ik denk dat het goed is om dat extra uur er niet af te halen, zodat ze weer in lijn zijn, mits beeclear CET/CEST is en niet UTC. Anders is het sowieso niet te debuggen.

[image]

----

Heb je ook al eens geprobeerd om de tijden gewoon als CET/CEST in te schieten en niet UTC? Dat maakt het wellicht makkelijker, tenzij de beeclear in UTC is. 

Als het goed is kun je gewoon `+01:00` en `+02:00` opgeven aan de API ipv `Z`. https://github.com/dsmrreader/dsmr-reader/issues/1771#issuecomment-1360028032

nickgr6, 2022-12-22:
Net even via postman een testje gedaan. Als ik `2025-01-15T12:34:56+01:00` belandt dit in de DB gewoon in UTC format zonder offset: `2025-01-15 11:34:56.000000 +00:00`. Heeft dus denk weinig nut om mijn migratiescript daarop aan te passen. Jammer, want als het in de database met +1:00 of +2:00 notatie zou blijven zou het makkelijker te vergelijken zijn ipv steeds UTC met CET te moeten vergelijken. 

Beeclear is CET/CEST:
[image]

Als ik de extra offset eraf haal dan klopt idd het totaal van 4 dec, maar zoals eerder ook vermeld, gaat dan een dag met gasverbruik in het laatste uur weer fout 😅 :
[image]
Waarbij gas.csv de beeclear export betreft.
Volgens mij kloppen de waardes in deze tabel?

Maar je ziet dan echter in de uur grafiek, dat alles een uur 'verschoven' is en het laatste uur niet wordt meegeteld 
[image]
Totaal verbruik in dsmr: 5,390 en in beeclear 5,977m3, wat dus precies het verschil is van de laatste waarde.. 

Als ik kijk naar tabel `dsmr_stats_daystatistics` zie ik:
[image]
Volgens mij klopt deze tabel niet: 6179.756 - 6173.779 = 5.977 en niet 5.390. Beeclear geeft ook 5.977 aan. 

De meterstand van de dagen in de tabel komen wel overeen met beeclear:
[image]

Lijkt er dus op dat kolom `gas` incorrect berekend wordt?

dennissiemensma, 2022-12-22:
> Net even via postman een testje gedaan. Als ik 2025-01-15T12:34:56+01:00 belandt dit in de DB gewoon in UTC format zonder offset: 2025-01-15 11:34:56.000000 +00:00. Heeft dus denk weinig nut om mijn [...]

Bedankt voor je aanvulling. Voor de duidelijkheid, het gaat er mij niet zozeer om wat de database opslaat, want dat is altijd UTC. Het gaat er vooral om dat je dan zelf niet de conversie hoeft te doen, als je brondata niet UTC is. Als dat het makkelijker voor jezelf maakt.

Je kunt zelf in postgres aangeven in welke tijdzone je de data terug wilt hebben, dat staat los van de UTC-opslag. Zie hieronder als voorbeeld dezelfde data in andere tijdzones. 
Wellicht kun je je SQL client daarop instellen en anders handmatig via een query zoals hieronder. 

```
dsmrreader=# SET TIMEZONE='Europe/Amsterdam';
SET
dsmrreader=# select id, timestamp from dsmr_datalogger_dsmrreading order by timestamp desc limit 5;
    id    |       timestamp        
----------+------------------------
 19422064 | 2022-12-22 20:33:27+01
 19422063 | 2022-12-22 20:33:17+01
 19422062 | 2022-12-22 20:33:07+01
 19422061 | 2022-12-22 20:32:57+01
 19422060 | 2022-12-22 20:32:47+01
(5 rows)
```
```
dsmrreader=# SET TIMEZONE='UTC';
SET
dsmrreader=# select id, timestamp from dsmr_datalogger_dsmrreading order by timestamp desc limit 5;
    id    |       timestamp        
----------+------------------------
 19422064 | 2022-12-22 19:33:27+00
 19422063 | 2022-12-22 19:33:17+00
 19422062 | 2022-12-22 19:33:07+00
 19422061 | 2022-12-22 19:32:57+00
 19422060 | 2022-12-22 19:32:47+00
(5 rows)
```

----

> Lijkt er dus op dat kolom gas incorrect berekend wordt?

Ik geloof je helemaal dat er ergens iets niet klopt, maar voor het debuggen van dit issue zoek ik daarom naar de SQL export van je DSMR-reader-variant waar je ze in hebt geschoten als hoe ze ook in Beeclear staan, zonder verschil in uren. 
Ongeacht of je ze zelf omzet naar UTC of de lokale tijdzone gebruikt. 

Ik hoef alleen de `dsmr_datalogger_dsmrreading` export. Want dan kan ik inhoudelijk voor je kijken waar het misgaat.

dennissiemensma, 2022-12-22:
Mocht je trouwens DataGrip gebruiken (lijkt het op) dan kun je de tijdzone instellen in de optie van je DB-verbinding. Ik moest zelf wel even de verbinding verbreken en tabel sluiten voordat het effect had.

[image]

nickgr6, 2022-12-22:
Fair point, had je verkeerd begrepen, maar inderdaad het query'en met Amsterdam timezone maakt het een stuk makkelijker vergelijken. Ik gebruik IntelliJ dus kon idd bij options de timezone configureren 👍.

Export:
```
insert into dsmr_datalogger_dsmrreading (id, timestamp, electricity_delivered_1, electricity_returned_1, electricity_delivered_2, electricity_returned_2, electricity_currently_delivered, electricity_currently_returned, extra_device_timestamp, extra_device_delivered, processed, phase_currently_delivered_l1, phase_currently_delivered_l2, phase_currently_delivered_l3, phase_currently_returned_l1, phase_currently_returned_l2, phase_currently_returned_l3, phase_voltage_l1, phase_voltage_l2, phase_voltage_l3, phase_power_current_l1, phase_power_current_l2, phase_power_current_l3)
values  (294964, '2022-12-01 00:00:00.000000 +01:00', 6954.079, 0.000, 9326.590, 0.000, 0.000, 0.000, '2022-12-01 00:00:00.000000 +01:00', 6142.308, true, null, null, null, null, null, null, null, null, null, null, null, null),
        (294965, '2022-12-01 01:00:00.000000 +01:00', 6954.196, 0.000, 9326.590, 0.000, 0.000, 0.000, '2022-12-01 01:00:00.000000 +01:00', 6142.308, true, null, null, null, null, null, null, null, null, null, null, null, null),
        (294966, '2022-12-01 02:00:00.000000 +01:00', 6954.323, 0.000, 9326.590, 0.000, 0.000, 0.000, '2022-12-01 02:00:00.000000 +01:00', 6142.308, true, null, null, null, null, null, null, null, null, null, null, null, null),
        (294967, '2022-12-01 03:00:00.000000 +01:00', 6954.425, 0.000, 9326.590, 0.000, 0.000, 0.000, '2022-12-01 03:00:00.000000 +01:00', 6142.308, true, null, null, null, null, null, null, null, null, null, null, null, null),
        (294968, '2022-12-01 04:00:00.000000 +01:00', 6954.547, 0.000, 9326.590, 0.000, 0.000, 0.000, '2022-12-01 04:00:00.000000 +01:00', 6142.308, true, null, null, null, null, null, null, null, null, null, null, null, null),
        (294969, '2022-12-01 05:00:00.000000 +01:00', 6954.666, 0.000, 9326.590, 0.000, 0.000, 0.000, '2022-12-01 05:00:00.000000 +01:00', 6142.308, true, null, null, null, null, null, null, null, null, null, null, null, null),
        (294970, '2022-12-01 06:00:00.000000 +01:00', 6954.759, 0.000, 9326.590, 0.000, 0.000, 0.000, '2022-12-01 06:00:00.000000 +01:00', 6142.308, true, null, null, null, null, null, null, null, null, null, null, null, null),
        (294971, '2022-12-01 07:00:00.000000 +01:00', 6954.896, 0.000, 9326.590, 0.000, 0.000, 0.000, '2022-12-01 07:00:00.000000 +01:00', 6142.354, true, null, null, null, null, null, null, null, null, null, null, null, null),
        (294972, '2022-12-01 08:00:00.000000 +01:00', 6954.897, 0.000, 9326.772, 0.000, 0.000, 0.000, '2022-12-01 08:00:00.000000 +01:00', 6142.954, true, null, null, null, null, null, null, null, null, null, null, null, null),
        (294973, '2022-12-01 09:00:00.000000 +01:00', 6954.897, 0.000, 9327.227, 0.000, 0.000, 0.000, '2022-12-01 09:00:00.000000 +01:00', 6143.800, true, null, null, null, null, null, null, null, null, null, null, null, null),
        (294974, '2022-12-01 10:00:00.000000 +01:00', 6954.897, 0.000, 9328.008, 0.000, 0.000, 0.000, '2022-12-01 10:00:00.000000 +01:00', 6144.554, true, null, null, null, null, null, null, null, null, null, null, null, null),
        (294975, '2022-12-01 11:00:00.000000 +01:00', 6954.897, 0.000, 9328.572, 0.000, 0.000, 0.000, '2022-12-01 11:00:00.000000 +01:00', 6145.216, true, null, null, null, null, null, null, null, null, null, null, null, null),
        (294976, '2022-12-01 12:00:00.000000 +01:00', 6954.897, 0.000, 9329.099, 0.000, 0.000, 0.000, '2022-12-01 12:00:00.000000 +01:00', 6145.489, true, null, null, null, null, null, null, null, null, null, null, null, null),
        (294977, '2022-12-01 13:00:00.000000 +01:00', 6954.897, 0.000, 9329.478, 0.000, 0.000, 0.000, '2022-12-01 13:00:00.000000 +01:00', 6145.641, true, null, null, null, null, null, null, null, null, null, null, null, null),
        (294978, '2022-12-01 14:00:00.000000 +01:00', 6954.897, 0.000, 9329.844, 0.000, 0.000, 0.000, '2022-12-01 14:00:00.000000 +01:00', 6146.095, true, null, null, null, null, null, null, null, null, null, null, null, null),
        (294979, '2022-12-01 15:00:00.000000 +01:00', 6954.897, 0.000, 9330.371, 0.000, 0.000, 0.000, '2022-12-01 15:00:00.000000 +01:00', 6146.095, true, null, null, null, null, null, null, null, null, null, null, null, null),
        (294980, '2022-12-01 16:00:00.000000 +01:00', 6954.897, 0.000, 9331.043, 0.000, 0.000, 0.000, '2022-12-01 16:00:00.000000 +01:00', 6146.203, true, null, null, null, null, null, null, null, null, null, null, null, null),
        (294981, '2022-12-01 17:00:00.000000 +01:00', 6954.897, 0.000, 9331.570, 0.000, 0.000, 0.000, '2022-12-01 17:00:00.000000 +01:00', 6147.090, true, null, null, null, null, null, null, null, null, null, null, null, null),
        (294982, '2022-12-01 18:00:00.000000 +01:00', 6954.897, 0.000, 9332.067, 0.000, 0.000, 0.000, '2022-12-01 18:00:00.000000 +01:00', 6147.870, true, null, null, null, null, null, null, null, null, null, null, null, null),
        (294983, '2022-12-01 19:00:00.000000 +01:00', 6954.897, 0.000, 9332.238, 0.000, 0.000, 0.000, '2022-12-01 19:00:00.000000 +01:00', 6147.999, true, null, null, null, null, null, null, null, null, null, null, null, null),
[... cut ...]
        (295359, '2022-12-17 11:00:00.000000 +01:00', 7000.941, 0.000, 9412.470, 0.000, 0.000, 0.000, '2022-12-17 11:00:00.000000 +01:00', 6208.794, true, null, null, null, null, null, null, null, null, null, null, null, null),
        (295360, '2022-12-17 12:00:00.000000 +01:00', 7001.503, 0.000, 9412.470, 0.000, 0.000, 0.000, '2022-12-17 12:00:00.000000 +01:00', 6208.794, true, null, null, null, null, null, null, null, null, null, null, null, null),
        (295361, '2022-12-17 13:00:00.000000 +01:00', 7001.616, 0.000, 9412.470, 0.000, 0.000, 0.000, '2022-12-17 13:00:00.000000 +01:00', 6208.799, true, null, null, null, null, null, null, null, null, null, null, null, null);
```

dennissiemensma, 2022-12-22:
De meterstanden in de dagstatistieken-tabel worden niet gebruikt voor berekeningen. Deze zijn later toegevoegd puur ter historie.

De dataflow binnen DSMR-reader is:
```
dsmr_datalogger_dsmrreading  ->-> dsmr_consumption_electricityconsumption  ->-> dsmr_stats_daystatistics
                            \->-> dsmr_consumption_gasconsumption          ->-/
```

De twee tussentabellen voeren eventuele groepering uit. Als het ergens mis gaat, zit het vermoedelijk in `dsmr_consumption_gasconsumption`.

dennissiemensma, 2022-12-22:
De records voor 10 december in `dsmr_consumption_gasconsumption` komen inderdaad uit op `5,39`

```
,128+,226+,219+,141+,114+,450+,878+,331+,113+,006+,042+,020+,205+,524+,742+,451+,800
```

Dus ergens zit daar iets wat niet meegenomen wordt.

nickgr6, 2022-12-22:
Verschil met beeclear is de meting tussen 23:00 en 00:00 de volgende dag:
[image]
[image]

dennissiemensma, 2022-12-22:
Wat geeft deze query bij jou?

```
select dsmr_version from dsmr_datalogger_meterstatistics
```

dennissiemensma, 2022-12-22:
Als die leeg is, kun je deze eens proberen:

```
UPDATE dsmr_datalogger_meterstatistics SET dsmr_version = '42' WHERE true;
```

En dan de import/verwerking opnieuw doen.

dennissiemensma, 2022-12-22:
# `dsmr_version = '42'`
[image]

----

# `dsmr_version = '50'` (of leeg)
[image]

dennissiemensma, 2022-12-22:
Het is een ontzettend edge-case die ik zelf nog niet eerder had gezien (of vergeten was). 

Omdat de telegramversie onbekend is totdat je een echt telegram stuurt, trekt DSMR-reader geen uur af van de data en krijg je dus die verspringing.

- https://github.com/dsmrreader/dsmr-reader/blob/v5/dsmr_consumption/services.py#L330-L334

Ik denk dat de meeste gebruikers eerst DSMR-reader aan de slimme meter aansluiten en daarna pas de import doen. Want dit zit er al een eeuwigheid in en het is niet eerder opgevallen:

- https://github.com/dsmrreader/dsmr-reader/blame/cc089aba112149538c8990b4dc9cbc79e189a8fd/dsmr_consumption/services.py#L291-L295

[image]

Omdat het er zolang in zit, ga ik het ook niet aanpassen omdat de kans groot is dat het effect heeft op andere dingen. Desnietemin is het wel een soort van bug of uitzondering die iemand wellicht nog vaker tegen gaat komen. Dus mooi dat het nu een keertje helder is. Dank!

nickgr6, 2022-12-23:
`select dsmr_version from dsmr_datalogger_meterstatisticsz`geeft `50`

> Als die leeg is, kun je deze eens proberen:
> 
> ```
> UPDATE dsmr_datalogger_meterstatistics SET dsmr_version = '42' WHERE true;
> ```
> 
> En dan de import/verwerking opnieuw doen.

Gedaan en dat fixed inderdaad 1-12-2022, maar nu gebeurd hetzelfde als toen ik een extra uur van de offset had afgehaald: op 4-12-2022 wordt het eerste uur niet meer meegerekend.

Data in dsmr_consumption_gasconsumption table:
```
id,read_at,delivered,currently_delivered
250828,2022-11-30 23:00:00.000000 +01:00,6142.308,0.000
250829,2022-12-01 00:00:00.000000 +01:00,6142.308,0.000
250830,2022-12-01 01:00:00.000000 +01:00,6142.308,0.000
250831,2022-12-01 02:00:00.000000 +01:00,6142.308,0.000
250832,2022-12-01 03:00:00.000000 +01:00,6142.308,0.000
250833,2022-12-01 04:00:00.000000 +01:00,6142.308,0.000
250834,2022-12-01 05:00:00.000000 +01:00,6142.308,0.000
250835,2022-12-01 06:00:00.000000 +01:00,6142.354,0.046
250836,2022-12-01 07:00:00.000000 +01:00,6142.954,0.600
250837,2022-12-01 08:00:00.000000 +01:00,6143.800,0.846
250838,2022-12-01 09:00:00.000000 +01:00,6144.554,0.754
250839,2022-12-01 10:00:00.000000 +01:00,6145.216,0.662
250840,2022-12-01 11:00:00.000000 +01:00,6145.489,0.273
250841,2022-12-01 12:00:00.000000 +01:00,6145.641,0.152
250842,2022-12-01 13:00:00.000000 +01:00,6146.095,0.454
250843,2022-12-01 14:00:00.000000 +01:00,6146.095,0.000
250844,2022-12-01 15:00:00.000000 +01:00,6146.203,0.108
250845,2022-12-01 16:00:00.000000 +01:00,6147.090,0.887
250846,2022-12-01 17:00:00.000000 +01:00,6147.870,0.780
250847,2022-12-01 18:00:00.000000 +01:00,6147.999,0.129
250848,2022-12-01 19:00:00.000000 +01:00,6148.004,0.005
250849,2022-12-01 20:00:00.000000 +01:00,6148.510,0.506
250850,2022-12-01 21:00:00.000000 +01:00,6148.734,0.224
250851,2022-12-01 22:00:00.000000 +01:00,6148.734,0.000
250852,2022-12-01 23:00:00.000000 +01:00,6148.734,0.000
250853,2022-12-02 00:00:00.000000 +01:00,6148.734,0.000
250854,2022-12-02 01:00:00.000000 +01:00,6148.734,0.000
250855,2022-12-02 02:00:00.000000 +01:00,6148.734,0.000
250856,2022-12-02 03:00:00.000000 +01:00,6148.734,0.000
250857,2022-12-02 04:00:00.000000 +01:00,6148.734,0.000
250858,2022-12-02 05:00:00.000000 +01:00,6148.734,0.000
250859,2022-12-02 06:00:00.000000 +01:00,6148.753,0.019
250860,2022-12-02 07:00:00.000000 +01:00,6148.972,0.219
250861,2022-12-02 08:00:00.000000 +01:00,6149.570,0.598
250862,2022-12-02 09:00:00.000000 +01:00,6150.055,0.485
250863,2022-12-02 10:00:00.000000 +01:00,6150.204,0.149
250864,2022-12-02 11:00:00.000000 +01:00,6150.316,0.112
250865,2022-12-02 12:00:00.000000 +01:00,6150.595,0.279
250866,2022-12-02 13:00:00.000000 +01:00,6150.602,0.007
250867,2022-12-02 14:00:00.000000 +01:00,6150.646,0.044
250868,2022-12-02 15:00:00.000000 +01:00,6150.722,0.076
250869,2022-12-02 16:00:00.000000 +01:00,6150.792,0.070
250870,2022-12-02 17:00:00.000000 +01:00,6150.818,0.026
250871,2022-12-02 18:00:00.000000 +01:00,6150.838,0.020
250872,2022-12-02 19:00:00.000000 +01:00,6150.974,0.136
250873,2022-12-02 20:00:00.000000 +01:00,6151.057,0.083
250874,2022-12-02 21:00:00.000000 +01:00,6151.057,0.000
250875,2022-12-02 22:00:00.000000 +01:00,6151.532,0.475
250876,2022-12-02 23:00:00.000000 +01:00,6151.532,0.000
250877,2022-12-03 00:00:00.000000 +01:00,6151.532,0.000
250878,2022-12-03 01:00:00.000000 +01:00,6151.532,0.000
250879,2022-12-03 02:00:00.000000 +01:00,6151.532,0.000
250880,2022-12-03 03:00:00.000000 +01:00,6151.532,0.000
250881,2022-12-03 04:00:00.000000 +01:00,6151.532,0.000
250882,2022-12-03 05:00:00.000000 +01:00,6151.532,0.000
[... 320 lines cut ...]
251203,2022-12-16 14:00:00.000000 +01:00,6204.411,0.000
251204,2022-12-16 15:00:00.000000 +01:00,6204.560,0.149
251205,2022-12-16 16:00:00.000000 +01:00,6204.866,0.306
251206,2022-12-16 17:00:00.000000 +01:00,6205.177,0.311
251207,2022-12-16 18:00:00.000000 +01:00,6205.226,0.049
251208,2022-12-16 19:00:00.000000 +01:00,6205.231,0.005
251209,2022-12-16 20:00:00.000000 +01:00,6205.231,0.000
251210,2022-12-16 21:00:00.000000 +01:00,6205.236,0.005
251211,2022-12-16 22:00:00.000000 +01:00,6205.236,0.000
251212,2022-12-16 23:00:00.000000 +01:00,6205.236,0.000
251213,2022-12-17 00:00:00.000000 +01:00,6205.265,0.029
251214,2022-12-17 01:00:00.000000 +01:00,6205.265,0.000
251215,2022-12-17 02:00:00.000000 +01:00,6205.265,0.000
251216,2022-12-17 03:00:00.000000 +01:00,6205.377,0.112
251217,2022-12-17 04:00:00.000000 +01:00,6205.670,0.293
251218,2022-12-17 05:00:00.000000 +01:00,6205.965,0.295
251219,2022-12-17 06:00:00.000000 +01:00,6206.158,0.193
251220,2022-12-17 07:00:00.000000 +01:00,6206.543,0.385
251221,2022-12-17 08:00:00.000000 +01:00,6207.309,0.766
251222,2022-12-17 09:00:00.000000 +01:00,6208.445,1.136
251223,2022-12-17 10:00:00.000000 +01:00,6208.794,0.349
251224,2022-12-17 11:00:00.000000 +01:00,6208.794,0.000
251225,2022-12-17 12:00:00.000000 +01:00,6208.799,0.005
```
PS: hetzelfde lijkt te gebeuren voor elektra.

dennissiemensma, 2022-12-23:
Bedankt voor het checken! Ik ben er inmiddels ook achter waar het in zit, en dat is die tussentabel.

Alleen heb ik er geen makkelijke fix voor, omdat het e.e.a. raakt. Uiteindelijk had ik al wel plannen om DSMR-reader te versimpelen en alles te baseren op de metingen-tabel, wellicht in combinatie met een timeseries-database, die er iets lekkerder voor werkt. 
Alleen was dat zeker niet iets voor de korte termijn, dus ik zal komende tijd overwegen wat een goede fix is en niet alles direct op de schop gooit.

dennissiemensma, 2022-12-29:
Voor de korte termijn zal ik een fix maken die:
- voortaan de eerste meterstanden van de dag gewoon uit de metingen-tabel haalt
- voortaan voor het totale gasverbruik van een dag de individuele uren optelt (die wel lijken te kloppen)
- een script maken dit bovenstaande met terugwerkende kracht rechttrekt, mits de gegevens nog bestaan 
 
Want niet iedereen heeft alle brongegevens meer, vandaar die meterstanden in de dagstatistieken-tabel als last-resort. Daarnaast zal ik het script niet automatisch uitvoeren, maar optioneel maken, zowel met een dry-run modus als "echte" modus zodat gebruikers eerst zien of het ze raakt en wat er verandert.

Alle andere zaken neem ik wel mee bij de versimpeling van DSMR-reader later.

nickgr6, 2022-12-29:
Top, laat maar weten wanneer ik kan helpen met testen.

dennissiemensma, 2022-12-30:
Nu ik er weer wat dieper in graaf, zie ik waarom gas sowieso al afwijkt. De eerste meting van de dag bevat (bij oude meters) sowieso niet perse de nieuwe gasstand. 

Mijn gasmeter update de slimme meter elk uur, maar dat is niet om klokslag middernacht. Vanochtend duurde het bijna 5 minuten voordat de gasupdate geregistreerd was:

[image]

Ik denk dat o.a. dat de reden is zoals het nu werkt. Ik zal het meenemen in ieder geval, want anders klopt de gasstand nog steeds niet om middernacht. Die moet dus uit twee verschillende metingen komen, omdat anders de elektrastand ook niet klopt.

dennissiemensma, 2023-01-06:
Een deel heb ik kunnen fixen, want voortaan logt DSMR-reader ook per dag het tijdstip van de meterstanden. Daarbij is bij verschillende varianten ook goed te zien waar het euvel o.a. zit: om middernacht is de gasstand niet bijgewerkt.

Wat situaties voor terugwerkende kracht fixen. Het gaat telkens om het tijdstip rechtsonderin:

## 1. Gastijdstip gebaseerd op alle beschikbare recente metingen (tijdstip ELEKTRA telegram)
Voor installaties waarbij automatische opschoning **nog niet** gedaan is, is dat niet erg. Zoals bij deze zijn alle metingen beschikbaar en zit de nieuwe gasstand in het telegram van 8 over 12.

[image]

----

## 2. Gastijdstip gebaseerd op uurlijkse metingen na opschoning (tijdstip ELEKTRA telegram)
Deze situatie is lastiger en geldt standaard voor alle gebruikers met standaardinstellingen. Na X periode worden alleen de eerste en laatste (elektra)meting per uur bewaard.

Dus hier zijn maar 2 telegrammen per uur beschikbaar en omdat de kans nihil is dat binnen 10 seconden na middernacht de gasstand is bijgewerkt, is hier te zien dat de gasstand na een uur wordt gepakt.

Dat is dus niet erg, gezien 3. hieronder.

[image]

----

## 3. Gastijdstip gebaseerd op uurlijkse metingen na opschoning (tijdstip GAS telegram)

Idem als bij 2. hierboven, maar nu gebaseerd op het tijdstip dat de gasmeter telkens meestuurt. Hetzelfde telegram als hierboven, dus bijna een uur na middernacht. Echter, omdat om ~1 uur 's nachts de gasstand van dat tijdstip ook niet bekend is, is het niet erg.

[image]

----

Deze fix wordt automatisch geprobeerd bij de volgende release, maar werkt dus alleen voor gebruikers die ook de (opgeschoonde) metingen nog hebben.

Situaties 1 en 2 zijn puur ter indicatie, de fix wordt nummer 3. Voor v4-meters is dat exact op het uur, voor v5 meters vermoedelijk een tijdstip tussen 0:00 en 0:05.

dennissiemensma, 2023-01-07:
Ik heb nu ook een mogelijke fix voor de totalen per dag met terugwerkende kracht.

Er vanuitgaande dat de uurtotalen dus wel lijken te kloppen, krijg ik zoiets als resultaat over ~7 jaar aan data:

```
Difference by year:
2015:   3.971 m³
2016:   22.317 m³
2017:   12.948 m³
2018:   3.167 m³
2019:   3.573 m³
2020:   0.563 m³
2021:   8.278 m³
2022:   10.989 m³

Occurrences by year:
2015:   19 x
2016:   129 x
2017:   52 x
2018:   64 x
2019:   12 x
2020:   4 x
2021:   24 x
2022:   48 x

Total gas difference:   65.806 m³   (352 x)
```

dennissiemensma, 2023-01-09:
Ik ga het iets simpeler oplossen. Bovenstaande fixes + herberekenen prijzen (waar van toepassing) wordt automatisch gedaan.

De output van het script zal ik dan als Dashboard-notificatie zetten. Zodat het ook altijd terug te zien is:

----

[image]

dennissiemensma, 2023-01-17:
Ik draai de fix nu zelf een tijdje en het lijkt goed. De totalen komen overeen met de meterstanden, die door de fix ook betrouwbaarder zijn.

De volgende stap is nog even alles langslopen en wat docs/toelichting.

nickgr6, 2023-01-17:
Klinkt goed! Ik wacht het rustig af!

dennissiemensma, 2023-01-23:
Ik heb het nu een tijdje draaien en het lijkt beter.

Echter kan ik het niet makkelijk laten previewen voor Docker-gebruikers, omdat die niet makkelijk van branch kunnen wisselen. 
Wellicht dat ik Xirixiz kan vragen of er een experimentele build gemaakt kan worden op basis van de `development` branch hier.

dennissiemensma, 2023-01-29:
Bij 1 gebruiker levert de datamigratie wijzigingen op waarvan nog niet duidelijk is of dat klopt of niet. Ik zal nog een tweede hotfix uitbrengen die de datamigratie skipt, voor nu. Dan kan het later handmatig/optioneel.

dennissiemensma, 2023-01-29:
@FutureCow kan jij wellicht ergens een dump van (een deel van) je database delen zodat ik kan zien waar die grote verschillen vandaan komen?

https://github.com/dsmrreader/dsmr-reader/issues/1795#issuecomment-1407704731

FutureCow, 2023-01-29:
2017-03-11 | Before: 4.569 m³ - Recalculated: 1289.238 m³ | Difference: 1284.669 m³
2017-03-12 | Before: 2.677 m³ - Recalculated: 2121.984 m³ | Difference: 2119.307 m³
2017-03-13 | Before: 3.802 m³ - Recalculated: 2194.978 m³ | Difference: 2191.176 m³
2017-03-14 | Before: 2.903 m³ - Recalculated: 2282.002 m³ | Difference: 2279.099 m³
2017-03-15 | Before: 2.287 m³ - Recalculated: 1839.808 m³ | Difference: 1837.521 m³
2017-04-09 | Before: 2.229 m³ - Recalculated: 2.315 m³ | Difference: 0.086 m³
2017-04-10 | Before: 2.168 m³ - Recalculated: 29.986 m³ | Difference: 27.818 m³
2017-04-11 | Before: 2.192 m³ - Recalculated: 89.646 m³ | Difference: 87.454 m³
2017-04-12 | Before: 3.274 m³ - Recalculated: 141.726 m³ | Difference: 138.452 m³
2017-04-13 | Before: 2.966 m³ - Recalculated: 230.300 m³ | Difference: 227.334 m³
2017-04-14 | Before: 4.143 m³ - Recalculated: 307.467 m³ | Difference: 303.324 m³
2017-04-15 | Before: 2.905 m³ - Recalculated: 170.091 m³ | Difference: 167.186 m³
2017-04-30 | Before: 1.806 m³ - Recalculated: 20.250 m³ | Difference: 18.444 m³
2017-05-01 | Before: 3.202 m³ - Recalculated: 30.243 m³ | Difference: 27.041 m³
2017-07-08 | Before: 0.392 m³ - Recalculated: 8.534 m³ | Difference: 8.142 m³
2017-09-23 | Before: 0.027 m³ - Recalculated: 0.132 m³ | Difference: 0.105 m³

Ik wil eventueel wel helpen om te debuggen. Data in mijn database is nu natuurlijk aangepast met de nieuwe (verkeerde) gegevens. Ik heb wel eens iets met de hand toegevoegd, geen idee meer of dat deze data zijn? Zit in de backup (van gister) ook de oude database?

dennissiemensma, 2023-01-29:
Een van beiden is goed. Een backup van gisteren zou helemaal top zijn, omdat ik het dan relatief makkelijk kan laten "naspelen".

FutureCow, 2023-01-29:
Hier de backup van gister
[dsmrreader-postgresql-backup-Saturday.sql.gz](https://github.com/dsmrreader/dsmr-reader/files/10530293/dsmrreader-postgresql-backup-Saturday.sql.gz)

dennissiemensma, 2023-01-29:
Thank, ik heb hem binnengehaald

dennissiemensma, 2023-01-29:
@FutureCow als ik in het Archief kijk naar de data van **9 april 2017 tot 15 april 2017** (nav die meldingen in migratie) dan lijkt het gasverbruik per uur absurd veel.

Het begint de 9e heel laag:
[image]

De dag erna al op 2+ per uur:
[image]

Rond de 14e al 11+ kuub per uur:
[image]

En de 15e stopt het opeens:
[image]

Het is natuurlijk bijna 6 jaar geleden, maar zegt die week jou uberhaupt wat qua speciale dingen? Of wellicht een bug in DSMR-reader rond die tijd? Er staat me zelf niets bij.

----

De migratie telt namelijk al het uurverbruik op om (doorgaans kleine) afwijkingen recht te trekken. Strikt gezien klopt de fix bij jou, maar de brondata is uiteraard heel gek, waardoor de fix ook heel gek wordt.

dennissiemensma, 2023-01-29:
@FutureCow als ik trouwens naar de meterstanden kijk (is toevallig een andere fix in een andere migratie van dezelfde release), dan lijken die wel aannemelijk, namelijk 2 tot 6 kuub **per dag** verbruikt.

Verder lijkt het sowieso af te wijken qua gas per uur in die maand. Ook de week ervoor zou het een paar kuub per dag zijn, maar op zo'n dag komen de uren bij elkaar opgeteld niet eens op 1 kuub uit. 

Ik denk overigens ook niet dat we dan achter de oorzaak gaan komen, tenzij je zelf nog iets kan heugen. Wellicht dat ik de migratie voor nu dan laat voor wat het is, en nog wat andere gebruikers afwacht. 
Omdat de meterstanden nu ook rechtgetrokken zijn (waar mogelijk), kan ik altijd nog een keer een migratie maken die ook het verbruik per uur opnieuw berekent.

dennissiemensma, 2023-01-29:
In v5.10.2 is de migratie voor nu even uitgeschakeld. Ik denk dat ik voor een volgende release iets maak waardoor het toch met de hand te dry-runnen en uit te voeren is. 

Samen met een nog-te-maken migratie die gewoon naar de meterstanden kijkt en het (eventuele foute) uurverbruik negeert, zoals bij @FutureCow. Voor nu hoef je daar niets mee te doen, je lijkt immers nog de oude meterstanden te hebben.

FutureCow, 2023-01-29:
Zal dan idd een n=1 fout zijn. Geen idee meer waar de vreemde uur waarden vandaan komen, 15 m3 krijg ik er zelfs met een uur douchen er niet doorheen.
Makkelijkste voor mij om de waarden via de admin interface even aan te passen?

rodeho, 2023-01-29:
> In v5.10.2 is de migratie voor nu even uitgeschakeld. Ik denk dat ik voor een volgende release iets maak waardoor het toch met de hand te dry-runnen en uit te voeren is.
>
> Samen met een nog-te-maken [...]

Tnx, en voor mensen die nu op 5.10.1 zitten, gaat dat goed of is het verstandiger een VM restore uit te voeren en 12 uur terug in de tijd te gaan?

dennissiemensma, 2023-01-29:
> Zal dan idd een n=1 fout zijn. Geen idee meer waar de vreemde uur waarden vandaan komen, 15 m3 krijg ik er zelfs met een uur douchen er niet doorheen. Makkelijkste voor mij om de waarden via de admin interface even aan te passen?

@FutureCow je kunt ze handmatig aanpassen op `/admin/dsmr_stats/daystatistics/` of wachten tot er een nieuwe fix voor komt waarbij het automatisch kan.

dennissiemensma, 2023-01-29:
> Tnx, en voor mensen die nu op 5.10.1 zitten, gaat dat goed of is het verstandiger een VM restore uit te voeren en 12 uur terug in de tijd te gaan?

@rodeho de migratie/fix zelf is goed, dus je kunt daar op blijven. Alleen wil ik nog een tweede maken en daarna ze via een knop of iets handmatig laten uitvoeren, ipv automatisch voor iedereen. De belangrijkste fix is overigens de release zelf, die vanaf nu beter en meer bijhoudt.

Kwamen er bij jou nog (grote) meldingen uit qua afwijkingen nav de migratie?

dennissiemensma, 2023-02-12:
@nickgr6 je zou de v5.10.3 release kunnen proberen om te zien of het inmiddels al beter gaat of nog wat rework nodig heeft.

nickgr6, 2023-02-16:
Bedankt voor de update! Vanwege vakantie ga ik hier echter pas na 15 maart naar kunnen kijken. Ik hou je op de hoogte nadien.

Roukie686868, 2023-03-18:
Ik kijk in de dsmr_consumption_gasconsumption tabel en zie dat precies een maand terug alleen nog de waarden op het hele en 5 minuten voor het hele uur gemeld staan. Alle andere tussen waarden zijn weg. Is er een manier om het werkelijke (correcte) verbruik nog terug te halen uit de database? In het plaatje zie je boven de streek waarden elke 5 minuten daaronder nog maar 2 waarden per uur.
[image]

Roukie686868, 2023-03-18:
Ik realiseer me dat met de “delivered” waarden het allemaal per uur en dat terug te rekenen is dus kan ik het probleem zelf oplossen

dennissiemensma, 2023-03-18:
@Roukie686868 `dsmr_consumption_gasconsumption` is een tussentabel die niet helemaal klopt (waar dit issue over gaat), de bronmetingen in `dsmr_datalogger_dsmrreading` kloppen wel. Gas heet daarin `extra_device_*`

dennissiemensma, 2023-03-18:
@Roukie686868 en aanvullend daarop, de data wordt na verloop van tijd (standaard een maand) uitgedund voor performance. De eerste en laatste meting van elk uur worden bewaard, waardoor het altijd terug te rekenen moet zijn. 

DSMR-reader zal dat ook doen wanneer ik een fix heb voor het huidige issue.

Roukie686868, 2023-03-18:
Perfect dat de data ook in dsmr_datalogger_dsmrreading staat. Dan is het mogelijk om echt alles weer perfect terug te rekenen.
Ik hoop dat je de fix snel voorelkaar krijgt.

Roukie686868, 2023-03-18:
Dennis nog bedankt voor het wijzen naar de juiste kolom. Ik haal deze data binnen in PowerBI en met de volgende query krijg ik precies wat ik zocht. Ik gebruik wel de set met de gereduceerde regels, Hier krijg ik ook wat er per uur verbruikt is zonder de database onnodig te belasten. 

SELECT read_at, (delivered - lag(delivered,1) over (order by read_at)) as verbruik
FROM public.dsmr_consumption_gasconsumption

hcombee, 2023-04-10:
Ik kwam toevallig deze thread tegen toen ik zocht naar een verklaring voor mijn probleem. 
Tussen het overzicht van de maand maart van Vattenval en DSMR reader zat 15 m3 verschil in het gasverbruik. Het stroomverbruik klopte wel.

Toen ik verder ging kijken zag ik dat het overzicht in Homeassistant en Mindergas.nl wel klopt met het overzicht van Vattenval. Beide krijgen de meterstanden door van DSMR reader.

Ik gebruik versie  v5.10.3. In de screenshot van het gasverbruik van gisteren zie je  het verschill tussen HAS en DSMR.

[image]

StevenM1, 2024-01-07:
Ik heb hetzelfde probleem ervaren. Bij het vergelijken van de jaarstatistieken zoals geregistreerd in DSMR reader, met handmatig bijgehouden meterstanden, bleek dat er een aanzienlijk verschil was in stroom- en gasverbruik (~10% minder gasverbruik in DSMR reader dan op de meterstanden). 
Hier wordt ook geen gebruik gemaakt van de standaard datalogger met live readings, maar in plaats daarvan injecteer ik zelf de telegramdata in de readings table van de database.

Volgens mij gaat het mis in het berekenen van de consumptions, en specifiek in de manier waarop een tijdsrange wordt gequery'd in `consumption_by_range()` in `dsmr-reader/dsmr_stats/services.py`:

```
    electricity_readings = ElectricityConsumption.objects.filter(
        read_at__gte=start,
        read_at__lt=end,
    ).order_by("read_at")

    gas_readings = GasConsumption.objects.filter(
        read_at__gte=start,
        read_at__lt=end,
    ).order_by("read_at")
```

Hier query't DSMR alles binnen een tijdsrange, waarbij de eerste reading wordt afgetrokken van de laatste reading om te bepalen wat de consumptie was binnen die range. `end` en `day` verschillen 1 uur of 1 dag, afhankelijk van of er een uur- of dagstatistiek wordt berekend. Maar dit kan misgaan:
Stel dat je eerste reading van de dag op 00:00 (middernacht) is, en de laatste reading om 23:55, dan wordt dus de reading van 00:00 afgetrokken van de reading van 23:55 om te bepalen hoeveel gas/elektriciteit er is verbruikt op deze dag. En dus worden de laatste 5 minuten van die dag níet meegerekend. Aangezien de jaarstatistieken een optelsom zijn van de day statitieken, kan dit behoorlijk oplopen.

In plaats daarvan werkt het bij mij om simpelweg te query'en op `read_at__lte` in plaats van `read_at__lt`. Dan neem je ook het laatste interval van de range mee in je metingen. Hiermee komt mijn jaarverbruik zoals geregistreerd in DSMR reader exact overeen met de handmatig bijgehouden meterstanden.

Overigens laat ik `create_daily_statistics` (dsmr-reader/dsmr_stats/services.py) gewoon de gasmetingen uit `consumption` returnen (en niet de nieuwe `hours_gas_sum`; waarom deze daar wordt berekend in plaats van de `consumption['gas']` is me niet helemaal duidelijk)

dennissiemensma, 2024-01-08:
@StevenM1 bedankt voor je uitgebreide aanvulling.  Het specifieke stukje code is al meerdere maken aangepast en weer teruggezet (#1385):

- @ 2019 `lt` -> `lte`:  https://github.com/dsmrreader/dsmr-reader/commit/2ff635033df54089c89b8bd7a257f7480abe5ef0#diff-79b655a9839c8d926e6778808d4f0ffe7a31f9d6038143b972023d3dd9f5c56d
- @ 2021 `lte` -> `le`: https://github.com/dsmrreader/dsmr-reader/commit/4a1d40d3592d84c5dfb16fe47e071d88b9ce7c06#diff-79b655a9839c8d926e6778808d4f0ffe7a31f9d6038143b972023d3dd9f5c56d

De kern van het hele probleem is uberhaupt die tussentabel in DSMR-reader, de consumption.

Er is sowieso geen garantie dat er een meting is tussen 23:55 en middernacht, wat de huidige opzet in DSMR-reader per definitie dus kan laten "lekken". 
Eigenlijk wil ik daar van af en die complete tussentabel er uit hebben en alles direct berekenen op de metingen. Hoewel dat niet alles oplost, zou ik dat wel kunnen gebruiken voor de dag/maand/jaar-totalen. En zou het dan in ieder geval beter moeten werken.

Alleen is dat niet te doen qua werkgrootte. Plus dat de huidige situatie onder (unit)tests staat die niet kloppen (want het klopt immers niet), dus die moeten ook over de kop. Het liefste naar een integratietest die meer naar het eindresultaat kijkt, voor DSMR v3, v4 en v5 meters.

En hoewel ik wel neig om weer terug te gaan van de situatie van 2019-2021 (dus jouw voorstel), voor nu, weet ik niet wat de gevolgen daarvan zijn. 
Een verbetering tov toen is wel dat er in de laatste DSMR-reader versies bij de dagtotalen ook de meterstanden opgeslagen worden. Zodat die altijd gebruikt kunnen worden voor ofwel een fix met terugwerkende kracht of gewoon als "waarheid".

Dus veel keuzes en veel wegen,

StevenM1, 2024-01-10:
Ik snap je overwegingen, en ben het er helemaal mee eens dat de beste oplossing (maar waarschijnlijk wel veel werk) zou zijn om gewoon direct de readings te query'en en aan de hand daarvan de consumption te berekenen.

Wat ik niet helemaal begrijp is wat de reden is geweest om van `lte` naar `lt` te gaan; ik kan iig zelf niet zo goed bedenken in welke situatie dat de juiste getallen zou opleveren. Maar ik heb niet de hele commit history gelezen en ken de situatie van 2021 niet, allicht was er een goede reden voor.

Zonder direct de tussentabellen eruit te gooien, lijkt het mij dat de meeste accurate benadering is om een range te pakken (zeg, 24 uur), en dan de eerste meting ná die range te gebruiken als eindpunt. Dat is iets complexer dan de `lte` die ik eerder voorstelde, omdat het vermoedelijk echt een tweede query nodig heeft, inderdaad voor de situatie dat geen reading voldoet aan de 'equal' conditie (dus dat er bijv. geen reading is van middernacht de volgende dag). Bij mij lijkt het goed te gaan omdat er altijd wel een reading is die exact overeenkomt met het eindpunt v/d range (ook omdat de timestamps van de readings in de database iets te lijken zijn opgeschoond t.o.v. de ruwe metingen in de telegrams), maar dat hoeft niet altijd het geval te zijn.

Indien er géén reading is op exact het eind van de range, lijkt het me dat de eerstvolgende reading de meest accurate benadering daarvan is. Het kan dan wel zijn dat die reading een tijdje na de gespecificeerde range is (hypothetisch, als je slechts om het uur een meting hebt, kan het zo zijn dat er op dag `t` een laatste meting is om 23:00, en op dag `t+1` pas een om 01:00). In dat geval moet er hoe dan ook een keuze worden gemaakt waaraan de consumptie in het interval 23:00-01:00 moet worden toebedeeld (tenzij je gaat interpoleren o.i.d. maar dat lijkt me het onnodig complex maken). Altijd de eerstvolgende reading pakken is in ieder geval consistent.

Maar goed - ik weet niet alles van de structuur van DSMR reader, en weet dat dit soort aanpassingen riskant zijn dus dat ik het begrijp dat je voorzichtig bent in het aanpassen van dit soort dingen. Maar misschien heb je iets aan wat ruggespraak.

dennissiemensma, 2024-01-10:
Bedankt voor je aanvulling. Het nadeel is dat ik ook niet meer exact weet welke wijziging waarvoor is, ondanks dat ik redelijk probeer om referenties in de commit messages te zetten. Alleen weet ik niet de echte achterliggende reden.

Wat betreft de edge-cases bij missende data, daar kan DSMR-reader sowieso niets in betekenen, dus ik ben het met je eens dat buiten de scope blijft. 

Inmiddels is wel duidelijk dat de wijziging hoe dan ook ingrijpend gaat worden, dus dat scheelt wat in de voorzichtigheid. Ook omdat er inmiddels ook weer qya Python/Django versies gebumped moeten worden, dus er komt sowieso een nieuwe major DSMR-reader versie uit, die mag afwijken van de vorige 5.x versie (mits data forward compatible, maar dat is zelden een issue). 

Ik denk dat uiteindelijk het slopen van de tussentabellen een hoop code scheelt, maar ook wel aardig wat kan raken qua API, MQTT, GUI, etc. Als in dat die data er straks niet meer is. Maar dat is dan maar zo. 
Helemaal omdat het verschil dusdanig groot is (icm hoge prijzen) dat het niet meer een goede benadering is van wat men verbruikt. Een kleine afwijking vond ik nog acceptabel, maar klein is het niet meer helaas.

Rudios81, 2024-01-22:
Allereest moet ik toegeven dat ik niet de gehele historie van gesprek hier gelezen heb, maar ik stuit ook op verschillen in dagelijks gas verbruik.
Wat mij met name opvalt is dat het dagelijkse gebruik binnen DSMR berekend wordt dmv de uurgetallen die uit de meter komen.
Vanuit DSMR maak ik gebruik van de mogelijkheid om data naar een influxDB te pushen. Daarin wordt alleen het cumulatieve getal van de gasmeter doorgestuurd, en dus in influxDB bewaard.
Nu, als ik vanuit die cumulatieve getallen een dagelijks getal bereken, komt dat niet overeen met hetgeen er door DSMR wordt berekend. Dat zou er dus eigenlijk op neer komen dat de uurgetallen die de meter afgeeft niet accuraat zijn???
Uiteindelijk zijn de € die er mee gemoeid gaan afhankelijk van de cumulatieve teller van de meter, daar wordt immers op afgerekend.

massis87, 2025-01-27:
> [@massis87](https://github.com/massis87) dank voor het checken van de meterstanden, want dat zou mijn volgende vraag zijn. Uiteindelijk zal de fix namelijk moeten kijken naar die meterstanden.
> 
> Kan het zijn dat je in jouw geval redelijk wat gasverbruik hebt rond middernacht? Want de bug zit in de grens tussen dagen, verbruik van overdag zou wel goed moeten registeren.

Helaas niet, dat was mijn eerste idee ook. Tussen 11h 's nachts en 1h 's ochtends is het nooit meer dan 1m3...
Ik had ook al vergeleken tussen 23h en 6h, omdat fluvius om 6h uitleest, maar over die 7 uren was het ook max 1.5m waar ik keek, niet genoeg om het maandtotaal te verklaren...

savagegek, 2025-02-18:
Hoi, hier even een inbraak want ik denk dat ik deze bug ook ervaar. Alleen wil ik dat wel even controleren omdat ik ook issue #1811 zag die er op lijkt. Mocht ik fout zitten hoor ik het graag.
Allereerst dank voor dit mooie project en ik ga niet roepen dat ik dit zsm asap opgelost wil hebben want ik ben allang blij dat dit mooie project bestaat, maar het wel fijn zou zijn.

Probleem wat ik tegenkom is dat de totalen van gasverbruik die dsmr laat zien in bijvoorbeeld het archief, niet kloppen. 
Meterstanden kloppen wel en worden goed opgehaald en opgeslagen door dsmr zie ik, dit heb ik bevestigd door zowel de ruwe data uit postgres te vergelijken als handmatig de meterstand bij te hebben gehouden de afgelopen dagen.

Er zit een fluctuerend verschil in. En hier boven lees ik over middernacht, overgang van dagen waar het door ontstaat. Wat het voor mij plausibel maakt omdat de vloerverwarming hier altijd aan staat en bij koudere dagen dus kan resulteren in gas gebruik precies in die uren.
Bijv:
 2025-01-01 00:00:02+01 |               2516.887
 2025-01-01 00:59:54+01 |               2517.208

Dit heb ik bij mijzelf even vergeleken en lijkt erop dat hier wel een verklaring in zit alleen is het niet 100% het gemiste gasverbruik waar ik dan op uitkom.

Als voorbeeld heb ik januari 2025:
dsmr archief: 171,504 m3
Energieleverancier: 187,2 m3
Wat ik kan bevestigen met de data uit postgres wat zegt 187,176m3. Wat afgerond dus 187,2 m3 is.

Dit is een verschil van 15,696 m3. En als ik de metingen van het tijdvak 00:00 - 00:59 optel kom ik op 11,427.
Punt is dat ik dus niet op die genoemde 15 m3 uitkom, maar wel al een stuk meer in de buurt.

dennissiemensma, 2026-03-05:
Ik heb hier nu een goede agent tool naar kunnen laten kijken, op mijn eigen dataset van inmiddels bijna 10 jaar, en die bevestigt de oorzaak:

[image]

[image]

---

De volgende stap is een fix proberen te laten maken en die een tijdje draaien. Als die sluitend blijkt, dan kan het later met terugwerkende kracht gedaan worden (voor zover de eerste meterstanden van elke dag beschikbaar zijn na #1301).

Het gaat niet sluitend worden als er niet een meterstand vrijwel direct na middernacht is, bijvoorbeeld doordat iemand het aantal metingen hard throttled met de datalogger- of backend-sleep. 

Verder kan dit ook de andere kant op werken, namelijk dat het verbruik tussen middernacht en de eerste meting, door de fix op de voorgaande dag geteld wordt. Maar ik denk dat dat minder kwalijk is dan het verschil nu. Wordt ook kwestie van proberen in de praktijk.

dennissiemensma, 2026-08-18:
Onderdeel van [v6.2](https://github.com/dsmrreader/dsmr-reader/releases/tag/v6.2.0), nieuwe container build volgt komend uur

[3 more replies]

closed 2026-08-18 by dennissiemensma