dsmrreader/dsmr-reader #1771: 🙋

cyberjunky, 2022-12-20:
### Description

I upload my energy data using v2 datalogger api (from within HA to DSMReader add-on, I read the values from P1 with KNX...

But I get http error 400, even though it works fine, the data is parsed.
Any idea what is wrong to get the errors out of my logs?
I have added +01 to timestamp, but that only gave 415 http and non working upload

This is example of a log entry without +01 (works but gives error 400)
```
2022-12-20 12:12:40.335 WARNING (MainThread) [homeassistant.components.rest_command] Error. Url: http://192.168.x.y:7777/api/v2/datalogger/dsmrreading. Status code 400. Payload: b'{"electricity_currently_delivered": 0.41000000000000003, "electricity_currently_returned": 0.0, "electricity_delivered_1": 19667.34, "electricity_delivered_2": 20721.84, "electricity_returned_1": 1678.061, "electricity_returned_2": 4234.9, "timestamp": "2022-12-20T12:12:40", "extra_device_timestamp": "2022-12-20T12:12:40", "extra_device_delivered": 8755.154}'```

Entry after adding +01 in timestamps:
```
2022-12-20 11:36:00.311 WARNING (MainThread) [homeassistant.components.rest_command] Error. Url: http://192.168.2.101:7777/api/v2/datalogger/dsmrreading. Status code 415. Payload: b'{"electricity_currently_delivered": 0.58, "electricity_currently_returned": 0.0, "electricity_delivered_1": 19667.34, "electricity_delivered_2": 20721.52, "electricity_returned_1": 1678.061, "electricity_returned_2": 4234.9, "timestamp": "2022-12-20T11:36:00+01", "extra_device_timestamp": "2022-12-20T11:36:00+01", "extra_device_delivered": 8754.677}'
```

### DSMR-reader version

5.9

### DSMR-reader platform

Docker (e.g. Xirixiz)

cyberjunky, 2022-12-20:
Fixed, I now sent the UTC time with the Z in the end...

cyberjunky, 2022-12-20:
Argg.. it seems not, weird ```2022-12-20 12:42:25.529 WARNING (MainThread) [homeassistant.components.rest_command] Error. Url: http://192.168.x.y:7777/api/v2/datalogger/dsmrreading. Status code 400. Payload: b'{"electricity_currently_delivered": 0.41000000000000003, "electricity_currently_returned": 0.0, "electricity_delivered_1": 19667.34, "electricity_delivered_2": 20722.16, "electricity_returned_1": 1678.061, "electricity_returned_2": 4234.9, "timestamp": "2022-12-20T11:42:25Z", "extra_device_timestamp": "2022-12-20T11:42:25Z", "extra_device_delivered": 8755.154}'```

dennissiemensma, 2022-12-20:
The HTTP 400 means the request sent is invalid. Usually the response body returns a hint, any idea what the response body is? It's probably something small.

You can try using:
```
2022-12-20T11:36:00+01:00
```

Instead of:
```
2022-12-20T11:36:00+01
```

See [one of the tests](https://github.com/dsmrreader/dsmr-reader/blob/v5/dsmr_api/tests/v2/test_dsmrreading.py#L53).

cyberjunky, 2022-12-21:
I changed it to below (was no : in timezone offset) for one of my tests but I still get error 400 and data seems to be accepted still, I have no clue whats wrong with it, could it be the number of digit's or something else? Thanks.

2022-12-21 16:51:25.190 WARNING (MainThread) [homeassistant.components.rest_command] Error. Url: http://192.168.x.y:7777/api/v2/datalogger/dsmrreading. Status code 400. Payload: b'{"electricity_currently_delivered": 0.9400000000000001, "electricity_currently_returned": 0.0, "electricity_delivered_1": 19668.84, "electricity_delivered_2": 20734.52, "electricity_returned_1": 1678.061, "electricity_returned_2": 4234.9, "timestamp": "2022-12-21T16:51:25+01:00", "extra_device_timestamp": "2022-12-21T16:51:25+01:00", "extra_device_delivered": 8757.752}'

dennissiemensma, 2022-12-21:
You will need to have a look at the **response body** to see what the exact error is. It _should_ have more details.

cyberjunky, 2022-12-21:
> You will need to have a look at the **response body** to see what the exact error is. It _should_ have more details.

Ah yes forgot about that, home assistant doesn't show, will use python snippet

cyberjunky, 2022-12-21:
```
[!] API error: {"electricity_currently_delivered":["Ensure that there are no more than 9 digits in total."]}
```
This will fix it i guess ;-)

closed 2022-12-21 by cyberjunky

cyberjunky, 2022-12-21:
I have imported some wrong data, is there an option to delete certain measurements like between 18:55-19:00 or by manual selecting them? I removde some meterreadings inside datalogger menu, but the graphs is not changed...

dennissiemensma, 2022-12-21:
You will need to do the same for:
- `/admin/dsmr_consumption/electricityconsumption/`
- `/admin/dsmr_consumption/gasconsumption/`

cyberjunky, 2022-12-22:
That worked, thanks!