dsmrreader/dsmr-reader #1772: Notificaties activeren lukt niet

syphernl, 2022-12-22:
### Description

Als ik onder Admin > Notificaties de service op Telegram, de gegevens invul en op "Save" druk krijg ik "Notification Apps Configuration was changed successfully." 
Maar als ik de pagina dan ververs staat de selectie weer op "-- Disabled --" en missen de gegevens die ik voor Telegram heb ingevuld.

Ik kan hem wel op Dummy zetten, daar blijft hij ook op. Idem voor de rest (Pushover & Prowl). 
Als ik hem vervolgens _daarna_ op Telegram zet, blijft hij daar wel op staan. En worden de instellingen daarvan ook gewoon opgeslagen.

### DSMR-reader version

v5.9

### DSMR-reader platform

Native (e.g. manual installation)

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                       v5.9 / v3.10.9 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 0.5s / 168h / False
    Latest telegram version read / Parser settings                                                 "42" / "4"

DATA
    Telegrams total (est.)                                                                             369215
    Consumption records electricity / gas (est.)                                               111160 / 33545
```

dennissiemensma, 2022-12-22:
Bedankt voor je melding. Zie je ook fouten in de logfiles? Omdat het opslaan van de instellingen niet direct iets versturen, komt de fout op de achtergrond.

syphernl, 2022-12-22:
De `dsmr_backend` gaf aan:

```
2022-12-22 09:16:15,620 ERROR    services     send_notification               177 |  - Notification API returned client error, wiping settings...
2022-12-22 09:16:15,655 ERROR    schedule     dispatch_signals                 20 | (AssertionError) <bound method NotificationAppConfig._on_backend_called_signal of <NotificationAppConfig: dsmr_notification>> errored: Notify API call failed: {"ok":false,"error_code":400,"description":"Bad Request: chat not found"} (HTTP 400)
2022-12-22 09:16:43,471 ERROR    services     send_notification               177 |  - Notification API returned client error, wiping settings...
2022-12-22 09:16:43,496 ERROR    schedule     dispatch_signals                 20 | (AssertionError) <bound method NotificationAppConfig._on_backend_called_signal of <NotificationAppConfig: dsmr_notification>> errored: Notify API call failed: {"ok":false,"error_code":400,"description":"Bad Request: chat not found"} (HTTP 400)
2022-12-22 09:16:52,786 ERROR    services     send_notification               177 |  - Notification API returned client error, wiping settings...
2022-12-22 09:16:52,810 ERROR    schedule     dispatch_signals                 20 | (AssertionError) <bound method NotificationAppConfig._on_backend_called_signal of <NotificationAppConfig: dsmr_notification>> errored: Notify API call failed: {"ok":false,"error_code":404,"description":"Not Found"} (HTTP 404)
2022-12-22 09:17:45,884 ERROR    services     send_notification               177 |  - Notification API returned client error, wiping settings...
2022-12-22 09:17:45,907 ERROR    schedule     dispatch_signals                 20 | (AssertionError) <bound method NotificationAppConfig._on_backend_called_signal of <NotificationAppConfig: dsmr_notification>> errored: Notify API call failed: {"ok":false,"error_code":404,"description":"Not Found"} (HTTP 404)
2022-12-22 09:18:16,466 ERROR    services     send_notification               177 |  - Notification API returned client error, wiping settings...
2022-12-22 09:18:16,490 ERROR    schedule     dispatch_signals                 20 | (AssertionError) <bound method NotificationAppConfig._on_backend_called_signal of <NotificationAppConfig: dsmr_notification>> errored: Notify API call failed: {"ok":false,"error_code":400,"description":"Bad Request: chat not found"} (HTTP 400)
```

dennissiemensma, 2022-12-22:
Dank voor het delen. De oorzaak lijkt dan: `Bad Request: chat not found`

DSMR-reader gooit dan de instellingen weer weg om te voorkomen dat het systeem oneindig externe diensten gaat spammen met foute gegevens en je geblacklist wordt.

dennissiemensma, 2022-12-22:
Ik neem aan dat je deze hebt geprobeerd? 

- https://dsmr-reader.readthedocs.io/en/v5/how-to/admin/notifications.html#telegram-how-do-i-obtain-the-chat-id

Let wel dat dit een lange tijd geleden is gemaakt, dus wellicht is het outdated als Telegram iets in de API heeft gewijzigd.

syphernl, 2022-12-22:
Ik had de waardes gekopieerd vanuit HASS.

Wellicht dat er een spatie achter stond?

Hij wilde aanvankelijk ook niet opslaan met alleen de type op Telegram en geen API key/chat ID zonder melding in de frontend dat die waardes verplicht zijn.

dennissiemensma, 2022-12-22:
De hele admin-interface in DSMR-reader is helaas verre van gebruiksvriendelijk en de notificatie-opzet helpt daar niet bij. Dus het is voor nu trial & error, maar weet dat ik deze Telegram-notificaties zelf niet gebruik, dus het kan ook zijn dat het simpelweg niet meer werkt hoor.

niekniek89, 2022-12-23:
Ik krijg elke ochtend netjes een telegram melding, met alle verbruik/kosten van de dag ervoor. Werkt bij mij al enkele jaren prima.

dennissiemensma, 2022-12-29:
@syphernl ben je hier nog verder mee gekomen?

syphernl, 2022-12-29:
> @syphernl ben je hier nog verder mee gekomen?

Nee, helaas niet. Ik blijf de foutmelding krijgen dat de chat id niet geldig is, terwijl HASS die ook gewoon gebruikt.

dennissiemensma, 2022-12-29:
En bovenstaande link werkt ook niet qua stappen? Wellicht zelfs een nieuwe chat proberen ter vergelijking?

syphernl, 2022-12-30:
✅ Probleem opgelost! 

Je moet _eerst_ op "Start" klikken (of `/start` sturen) voor je bot je iets mag sturen 😅

closed 2022-12-30 by syphernl