dsmrreader/dsmr-reader #1773: 🙋Vanaf 01-01-2023 geen "dubbele meter" meer.

niekniek89, 2022-12-22:
### Description

Goedemorgen,

Momenteel heb ik nu nog een "dubbele meter" (normaal/laag tarief).
Vanaf 01-01-2023 (als het prijsplafond actief word), gaat bij mij (Essent) de "dubbele meter" weg.
Dit betekend dus geen normaal/laag tarief meer.

Nu zie ik in DSMR dat dit netjes per dag word bijgehouden in de kolommen.
Moet ik nog iets doen binnen DSMR, zodat normaal/laag straks verdwijnt, en er gewoon 1 normaaltarief over blijft?

[image]

### DSMR-reader version

DSMR-reader: v5.9

### DSMR-reader platform

Docker (e.g. Xirixiz)

Truesh00ter, 2022-12-22:
Volgens mij kun je gewoon hier:
http://IP_of_hostname/admin/dsmr_consumption/energysupplierprice/1/change/

Je daltarief verwijderen, daar staat namelijk ook bij dat je het enkel hoeft in te vullen indien relevant.

dennissiemensma, 2022-12-22:
@niekniek89 je kunt in de frontend/weergave-instellingen in de admin een vinkje aanzetten voor het samenvoegen van tarief 1 & 2.

Qua prijzen kun je gewoon dezelfde prijs voor tarief 1 & 2 invullen.

niekniek89, 2022-12-23:
Bedankt.
Ik heb inderdaad een optie gevonden voor het samenvoegen van de tarieven.

closed 2022-12-23 by niekniek89