dsmrreader/dsmr-reader #1775: 🚨API telegram updates only once per minute in graphs

slinkingant, 2022-12-29:
### Description

Telegram-waardes welke via de V2 api (api/v2/datalogger/dsmrreading) naar DSMR-reader gestuurd worden, komen maar 1x per minuut in de 'live graphs' terecht.

Ik heb Pi met DSMR-reader, welke ik probeer te vervangen door een ESP32 welke de telegram-data doorstuurt naar een DSMR-reader in docker. 
Op basis van deze dongle (https://smart-stuff.nl/product/p1-dongel-slimme-meter-esp32/) en wat eigen software in ESPHome, kan ik nu elke seconde de telegram-data van de meter doorsturen. Bij het refreshen van DSMR, zie ik ook elke seconde nieuwe waardes in het dashboard verschijnen, echter de grafiek gebruikt laat maar 1 waarde per minuut zien.

Door een splitter op de P1 poort heb ik nu beide aangesloten:
Pi met USB-kabel:
[image]
ESP32 naar docker:
[image]
Dit is voor alle grafieken, niet alleen voor de 'recent electricity usage'

Voor de lange-termijn statistieken maakt dit natuurlijk niet uit, maar vooral de korte piekjes zijn wel interessant om terug te kunnen zien op zoek naar sluipverbruikers.

Sidenote: in de ENV 'DSMRREADER_REMOTE_DATALOGGER_INPUT_METHOD' heb ik nu nog 'serial' staan, maar doordat er geen device gevonden wordt, staat de logfile vol met deze foutmeldingen: 'serial.serialutil.SerialException: Could not open port socket://None:23: [Errno -2] Name does not resolve'
Idee om er naast serial/ipv4 een api-optie in te maken, welke de uitleesfunctionaliteit niet aanroept?

### DSMR-reader version

5.9

### DSMR-reader platform

Docker (e.g. Xirixiz)

dennissiemensma, 2022-12-29:
Je kunt even dubbelchecken of data groepering toevallig aan staan. DSMR-reader groepeert metingen per minuut, maar ik weet even niet zeker of dat standaardgedrag is of niet. Het is in te stellen in de webconfiguratie.

Voor de datalogger kun je wellicht deze proberen: https://github.com/xirixiz/dsmr-reader-docker#dsmr-datalogger-related
```
DSMRREADER_OPERATION_MODE=api_server
```

slinkingant, 2022-12-29:
Ah, dat was inderdaad het probleem. Kan me niet herinneren dat ik die instelling aangepast heb, maar nu werken de grafieken weer per meting.

operation_mode aangepast naar api_server, en log is nu ook vrijwel 100% schoon. Alleen nog een 
`pg_dump: error: server version: 15.1 (Debian 15.1-1.pgdg110+1); pg_dump version: 14.5
pg_dump: error: aborting because of server version mismatch
`
Maar dat komt zo te zien door een mismatch in de pgdump van dsmr-reader/django vs postgres:latest docker image.

dennissiemensma, 2022-12-29:
Je kunt nog overwegen Postgres 14 te gebruiken voor nu, support tot eind '26: https://www.postgresql.org/support/versioning/

closed 2022-12-29 by dennissiemensma