dsmrreader/dsmr-reader #1782: Buienradar API lijkt geen weerstations meer te kunnen vinden🚨

stroep, 2023-01-02:
### Description

De Buienradar API komt momenteel terug met onderstaande meldingen:

2023-01-02 10:00:11,559 ERROR    services     run                              21 | Buienradar: Selected station info not found: 6280
2023-01-02 11:00:11,897 ERROR    services     run                              21 | Buienradar: Selected station info not found: 6279

(Ik heb hier geswitcht van Groningen naar Hoogeveen, voor de test, maar beide kan hij niet vinden)

Vanochtend geupdate naar de nieuwste versie, maar dit lost het niet op.

### DSMR-reader version

5.9

### DSMR-reader platform

None

### Debug info dump

```shell
<!-- COPY OUTPUT BELOW -->

DSMR-READER
    App / Python / Database                                                       v5.9 / v3.10.6 / postgresql
    BE sleep / DL sleep / Retention / Override                                    1.0s / 5.0s / 4032h / False
    Latest telegram version read / Parser settings                                                 "42" / "4"

DATA
    Telegrams total (est.)                                                                             346013
    Consumption records electricity / gas (est.)                                                  55810 / 937

<!-- COPY OUTPUT ABOVE -->
```

niekniek89, 2023-01-02:
Geen idee of dit ermee te maken heeft:

Ik gebruik Home Assistant, met daarin een addon van Buienradar (om het weer op te kunnen halen).
Echter heeft deze addon (tot eind van de ochtend) het niet gedaan (vanaf 30e december).
Heeft er dus ruim 2 dagen uitgelegen....
Op de website van buienradar stond bij "laatst geupdate" ook een outdate datum (2 dagen lang).

Leek er dus op alsof buienradar (of een specifieke locaties issues had).

niekniek89, 2023-01-02:
edit:

Leken toch issues bij Buienradar (of API), want mijn weerstation in Home Assistant werkt weer.
Zojuist gecheckt bij mij in DSMR:

[image]

Lijkt weer wat binnen te krijgen.

stroep, 2023-01-02:
Denk dat je gelijk hebt, bij mij is hij ook weer up-to-date.  Dan kan hij gesloten worden, vermoed ik. Misschien een andere bron als optie toe voegen?

niekniek89, 2023-01-02:
Top dat het weer werkt, merk overigens wel wat vaker (1x in de 3/4 maanden) dat de service van Buienradar een dipje heeft, en het na het weekend/feestdagen weer opgelost is.

Denk dat het bij Buienradar blijft.... volgens mij was het nooit echt de bedoeling een dergelijke functie als weerstation toe te voegen aan DSMR (valt volgens mij buiten de scope van het algemene doel DSMR (enkel uitlezen slimme meter)).

stroep, 2023-01-02:
Met het gevaar volledig off topic te raken: De temperatuurindicatie in de historie is wel heel waardevol, wil je gegevens vergelijken. Dus wat dat betreft, vind ik het zeker een goeie toevoeging.

dennissiemensma, 2023-01-02:
Bedankt voor het melden en de updates. Buienradar ligt er soms even uit, het blijft uiteraard een gratis dienst waar we dankbaar gebruik van maken.

> Denk dat het bij Buienradar blijft.... volgens mij was het nooit echt de bedoeling een dergelijke functie als weerstation toe te voegen aan DSMR (valt volgens mij buiten de scope van het algemene doel DSMR (enkel uitlezen slimme meter)).

Dat is inderdaad wel waar dit project uiteindelijk heen gaat op alle vlakken. Dit project is nu grofweg 6 a 7 jaar oud en heeft door de jaren heen wat zijdelingse features gekregen. 
In die jaren is er ook dusdanig veel veranderd in de wereld (in positieve zin) dat DSMR-reader nu allerlei dingen ernaast doet, die eigenlijk inmiddels op andere manieren veel beter gedaan worden (of kunnen worden).

Temperatuurmetingen zijn bijvoorbeeld leuk, maar eigenlijk zeggen ze weinig zonder [graaddagen](https://www.mindergas.nl/degree_days/explanation). En om dat allemaal in te bouwen, verschuift de focus van het project. Dus dan is een project zoals HA of MinderGas daar veel beter voor geschikt (qua temperatuur). 
Idem voor teruglevering zonnepanelen, dan is een PVOutput veel beter geschikt daar voor. Of notificaties/thresholds, daar zie ik iemand via HA veel specifieker de eigen wensen kunnen instellen, gecombineerd met andere componenten.

Een ander voorbeeld zijn (vanzelfsprekend) de energieprijzen in DSMR-reader. In het verleden waren er al wat verzoeken om de berekeningen van energieleveranciers te spiegelen qua kortingen en alles en dat is met het energieplafond nogmaals gevraagd, maar daar is geen beginnen aan. Sterker nog, de energieleveranciers zijn de enige die de waarheid weten qua berekeningen.
Het goede nieuws is dat door de huidige druk op de energieleveranciers ze hard werken (en vorig jaar ook al) aan nog meer/beter live inzicht in verbruik en deels ook de stand qua kosten. Dat zijn allemaal dingen die er 6 a 7 jaar terug veel minder waren, maar waar de consument/gebruiker echt een hoop aan heeft. Helemaal in deze tijd qua crisis en alles.

Dus dat betekent dat ik uiteindelijk de scope van dit project vooral verschuif naar de kern en ik wil zo min mogelijk dingen maken of onderhouden die iemand anders al (vaak veel beter) doet. Het is ook veel handiger om tools te hebben die goed zijn in specifieke dingen, dan eentje die van alles wat doet of probeert te doen. Dit zal overigens niet van de ene op de andere dag veranderen voor DSMR-reader hoor.

De kern van DSMR-reader wordt dan simpelweg het laagdrempelig bijhouden en vergelijking van (historisch) verbruik door de jaren heen. Waar het ook ooit mee begonnen is.
Want dat is het enige wat ik niet snel "opgelost" zie worden door energieleveranciers of andere clouddiensten met een P1-dongel. Als je overstapt dan is er geen garantie dat je je historische data daar mee kan nemen of het "overgezet" wordt. Zowel een HA als InfluxDB als een DSMR-reader kunnen daar mooi wat in betekenen voor gebruikers. Uiteraard ten koste van het zelf moeten hosten op een Pi of iets.

closed 2023-01-02 by dennissiemensma