dsmrreader/dsmr-reader #1783: 💡 Integratie dynamische prijs voor energiecontract (Tibber, Frank, Zonneplan)

henkjanwithaar, 2023-01-04:
### Feature

Zou het mogelijk zijn om een optie te ontwikkelen die gebruikers met een dynamisch energiecontract via een API ([zoals die van Tibber](https://developer.tibber.com/docs/overview)) eens per etmaal de geldende stroomtarieven laat importeren?

Ik maak zelf gebruik van de Tibber API om het verbruik en prijzen in Home Assistant te krijgen. Dit is in een dashboard goed te combineren met data die HA (via MQTT) uit DSMR ontvangt. Functioneel heb ik dus de data en het zou zoveel mooier zijn als de dag/maand/jaar prijsdata in DSMR óók volledig en kloppend kan worden gemaakt.

Dank voor het overwegen!

dennissiemensma, 2023-01-04:
Zie #1729

dennissiemensma, 2023-01-04:
En voor een recenter uitgebreider verhaal: https://github.com/dsmrreader/dsmr-reader/issues/1782#issuecomment-1369120245

closed 2023-01-04 by dennissiemensma