dsmrreader/dsmr-reader #1784: 🙋 Mindergas.nl bulk import van eerdere gegevens

ChefkeGremmen, 2023-01-05:
### Description

Hoi allemaal,

Allereerst; wat een geweldige tool! Helemaal in deze tijden helpt het me om gedegen met data ondersteunde beslissingen te nemen.
Maar goed, ik heb dus een vraag; ik heb de mindergas.nl API aangezet en dit werkt prima. Maar nu wil ik mijn eerder gegevens ook uploaden. Ja ja, ik weet dat die optie niet in DSMRReader zit en er ook niet in zal komen ;-)

Echter zegt mindergas.nl zelf wel het volgende:

> Ik wil een lijst met historische meterstanden importeren. Kan dit?
> 
> Je kan een Excel of CSV bestand aanleveren met daarin een kolom datum en een kolom meterstand om 00:00 uur (beginstand van de dag). Deze kunnen wij dan importeren. Neem hiervoor even contact op via het contactformulier.

Dus ik contact gelegd en ik mag een CSV terug sturen. Maar hoe genereer ik het makkelijkste een CSV? Ondanks dat ik software ontwikkelaar ben, heb ik veel te weinig ervaring met databases om 'even een query' o.i.d. op te zetten.

Alvast bedankt voor de hulp! 🙏

### DSMR-reader version

5.8

### DSMR-reader platform

Docker (e.g. Xirixiz)

dennissiemensma, 2023-01-05:
Bedankt voor je verzoek. Het is wat lastiger door een recent issue #1770, maar opzich maakt dat wellicht niet uit als de meterstanden elke dag van hetzelfde moment zijn (al kan die een uur afwijken, maar dan wel consequent elke dag).

Ik denk dat het het makkelijkste is als je `/export` opent qua URL in DSMR-reader en dan de bovenste/standaard optie pakt. Je hebt dan uit de CSV export alleen de eerste en laatste kolom nodig volgens mij. De eerste is de datum, de laatste de dagmeterstand (al kan die dus een uur afwijken, maar dan is het wel consequent bij alle records en zou het voor de per-dag-diff niet veel moeten uitmaken).

dennissiemensma, 2023-01-05:
Als je overigens een DSMR 5 slimme meter hebt met elke 5 minuten een gasstand, is de afwijking in #1770 vermoedelijk sowieso beperkt. In het ergste geval heb je dan een gasstand in de CSV die 5 a 10 minuten achterloopt en niet een uur, zoals met oude meters.