dsmrreader/dsmr-reader #1792: Waarschuwingen worden niet goed weergegeven in dark mode

Glodenox, 2023-01-13:
### Description

De informatieberichten die extra benadrukt worden met een gele kleur worden in dark mode volledig onleesbaar omdat de tekst daar in het lichtgrijs staat. Zie onderstaande screenshot als voorbeeld. Het is waarschijnlijk beter om een CSS "warning" class of iets dergelijks toe te voegen die er in beide modussen goed uit ziet zodat ook de element-specific CSS vervangen kan worden.

`<div class="submit-row" style="text-align: left; background: #FFFFCC;">`

[image]

Dit lijkt voor te komen in deze bestanden, maar het kan zijn dat er nog andere zulke element-specific regels zijn voor andere redenen:

> dsmr_api/templates/dsmr_api/api_settings/change_form.html (1 hit)
> dsmr_backend/templates/dsmr_backend/email_settings/change_form.html (1 hit)
> dsmr_backup/templates/dsmr_backup/backup_settings/change_form.html (1 hit)
> dsmr_backup/templates/dsmr_backup/dropbox_settings/change_form.html (1 hit)
> dsmr_backup/templates/dsmr_backup/email_backup_settings/change_form.html (1 hit)
> dsmr_consumption/templates/dsmr_consumption/consumption_settings/change_form.html (1 hit)
> dsmr_consumption/templates/dsmr_consumption/energy_supplier_prices/change_form.html (2 hits)
> dsmr_datalogger/templates/dsmr_datalogger/datalogger_settings/change_form.html (1 hit)
> dsmr_datalogger/templates/dsmr_datalogger/meter_statistics/change_form.html (1 hit)
> dsmr_datalogger/templates/dsmr_datalogger/retention_settings/change_form.html (1 hit)
> dsmr_influxdb/templates/dsmr_influxdb/influxdb_settings/change_form.html (2 hits)

### DSMR-reader version

5.9

### DSMR-reader platform

Docker (e.g. Xirixiz)

dennissiemensma, 2023-01-13:
Bedankt voor je melding. Ik zal er een keer wat dieper naar kijken en het meenemen in een release!

dennissiemensma, 2023-01-16:
Ik heb het wat versimpeld voor de volgende release.

closed 2023-01-16 by dennissiemensma