dsmrreader/dsmr-reader #1794: 🙋 Belgische Gas meter data niet weergegeven

svendaems, 2023-01-19:
### Description

Ik woon in België & heb DSMR-Reader via docker geïnstalleerd, maar ik krijg geen weergave van mijn gas verbruik.
Ik heb in de instellingen de DSMR-versie op 'België - Fluvius', maar toch krijg ik via `/admin/dsmr_consumption/gasconsumption/` geen gasverbruik te zien.

Hieronder vind je m'n laatste telegram gegevens. Ik zie daar dat tag `0-1:24.2.3` oploopt wanneer de gasketel aan staat, maar toch krijg ik nergens het verbruik te zien

```
/FLU5\253769484_A

0-0:96.1.4(***********)
0-0:96.1.1(***********)
0-0:1.0.0(230119124638W)
1-0:1.8.1(004423.770*kWh)
1-0:1.8.2(002607.237*kWh)
1-0:2.8.1(001194.693*kWh)
1-0:2.8.2(000755.554*kWh)
0-0:96.14.0(0001)
1-0:1.4.0(00.000*kW)
1-0:1.6.0(230116090000W)(11.173*kW)
0-0:98.1.0(1)(1-0:1.6.0)(1-0:1.6.0)(230101000000W)(221230114500W)(13.603*kW)
1-0:1.7.0(00.000*kW)
1-0:2.7.0(00.204*kW)
1-0:21.7.0(00.000*kW)
1-0:41.7.0(00.177*kW)
1-0:61.7.0(00.109*kW)
1-0:22.7.0(00.491*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
1-0:32.7.0(234.3*V)
1-0:52.7.0(234.2*V)
1-0:72.7.0(234.3*V)
1-0:31.7.0(002.18*A)
1-0:51.7.0(000.93*A)
1-0:71.7.0(000.60*A)
0-0:96.3.10(1)
0-0:17.0.0(999.9*kW)
1-0:31.4.0(999*A)
0-0:96.13.0()
0-1:24.1.0(003)
0-1:96.1.1(37464C4F32313231313630373732)
0-1:24.4.0(1)
0-1:24.2.3(230119124616W)(00734.607*m3)
0-2:24.1.0(003)
0-2:96.1.1(37464C4F32313231313630373734)
0-2:24.4.0(1)
0-2:24.2.3(230119124005W)(00000.000*m3)
!0EB7
```

### DSMR-reader version

5.9

### DSMR-reader platform

Docker (e.g. Xirixiz)

svendaems, 2023-01-19:
Hallo Dennis

Ik gebruik maar 1 gasmeter, maar er zijn er wel 2 geïnstalleerd. Vroeger waren er 2 aansluitingen in onze woning, maar de 2e is nu niet meer actief. Ik veronderstel dat dit hetzelfde effect zal hebben bij appartementsgebouwen waar elk appartement zijn eigen elektriciteit en gasmeter heeft.

dennissiemensma, 2023-01-19:
Ik denk dat DSMR-reader de nulmeter pakt en dat je daarom niets ziet. Ik zal wel een keer kijken waar dat aan ligt en of het een kleine fix is voor een volgende release.

svendaems, 2023-01-20:
Bedankt alvast!

dennissiemensma, 2023-01-23:
Ik heb zojuist je telegram in een test gegooid en het lijkt er inderdaad op dat die de andere (nul) gasmeter pakt.

Het heeft deels raakvlakken met de onderliggende parser in https://github.com/ndokter/dsmr_parser/issues/51, dus ik moet nog even kijken hoe ik het exact oplos.

dennissiemensma, 2023-01-23:
Ik zie nu in een comment van de parser van DSMR-reader dat er ooit eerder het omgekeerde gebeurde: #778
De fix mag ook daar rekening mee houden.

dennissiemensma, 2023-01-23:
Ik heb een workaround gemaakt voor de volgende release. Straks is er in de datalogger config specifiek voor Fluvius gasmeters een "kanaal" in te stellen, standaard op "automatisch".

In het geval van dit issue moet het hardcoded kanaal 1 zijn, omdat kanaal 2 onbedoeld automatisch geselecteerd wordt door DSMR-reader.

----

[image]

----

closed 2023-01-23 by dennissiemensma

svendaems, 2023-01-23:
Bedankt. Ik zal het zo snel als mogelijk testen wanneer de nieuwe versie online komt & feedback geven!

svendaems, 2023-01-30:
Bedankt, na de update wordt het gasverbruik bij mij nu wel meegegeven!

dennissiemensma, 2023-01-30:
Fijn om te horen!