dsmrreader/dsmr-reader #1795: 🚨upgrade to 5.10 failed: TypeError: unsupported operand type(s) for -: 'NoneType' and 'decimal.Decimal'

rodeho, 2023-01-29:
### Description

Hallo Dennis,

de upgrade van 5.9 naar 5.10 faalt er loopt wel een migratie voordat het uiteindelijk stopt.

```
Data migration: Retroactively checking/recalculating gas consumption for day statistics: 2020-05-17 (679/1656)
2020-05-17 | Before: 0.359 m³ - Recalculated: 0.445 m³ | Difference: 0.086 m³
Data migration: Retroactively checking/recalculating gas consumption for day statistics: 2020-05-18 (680/1656)
Data migration: Retroactively checking/recalculating gas consumption for day statistics: 2020-05-19 (681/1656)
Data migration: Retroactively checking/recalculating gas consumption for day statistics: 2020-05-20 (682/1656)
Data migration: Retroactively checking/recalculating gas consumption for day statistics: 2020-05-21 (683/1656)
Data migration: Retroactively checking/recalculating gas consumption for day statistics: 2020-05-22 (684/1656)
Data migration: Retroactively checking/recalculating gas consumption for day statistics: 2020-05-23 (685/1656)
Data migration: Retroactively checking/recalculating gas consumption for day statistics: 2020-05-24 (686/1656)
Data migration: Retroactively checking/recalculating gas consumption for day statistics: 2020-05-25 (687/1656)
Data migration: Retroactively checking/recalculating gas consumption for day statistics: 2020-05-26 (688/1656)
Data migration: Retroactively checking/recalculating gas consumption for day statistics: 2020-05-27 (689/1656)
Data migration: Retroactively checking/recalculating gas consumption for day statistics: 2020-05-28 (690/1656)
2020-05-28 | Before: 0.237 m³ - Recalculated: 0.238 m³ | Difference: 0.001 m³
Data migration: Retroactively checking/recalculating gas consumption for day statistics: 2020-05-29 (691/1656)
Data migration: Retroactively checking/recalculating gas consumption for day statistics: 2020-05-30 (692/1656)
Data migration: Retroactively checking/recalculating gas consumption for day statistics: 2020-05-31 (693/1656)
```

```
Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/core/management/__init__.py", line 419, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/core/management/__init__.py", line 413, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/core/management/base.py", line 354, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/core/management/base.py", line 398, in execute
    output = self.handle(*args, **options)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/core/management/base.py", line 89, in wrapped
    res = handle_func(*args, **kwargs)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/core/management/commands/migrate.py", line 244, in handle
    post_migrate_state = executor.migrate(
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/db/migrations/executor.py", line 117, in migrate
    state = self._migrate_all_forwards(state, plan, full_plan, fake=fake, fake_initial=fake_initial)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/db/migrations/executor.py", line 147, in _migrate_all_forwards
    state = self.apply_migration(state, migration, fake=fake, fake_initial=fake_initial)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/db/migrations/executor.py", line 227, in apply_migration
    state = migration.apply(state, schema_editor)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/db/migrations/migration.py", line 126, in apply
    operation.database_forwards(self.app_label, schema_editor, old_state, project_state)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.10/site-packages/django/db/migrations/operations/special.py", line 190, in database_forwards
    self.code(from_state.apps, schema_editor)
  File "/home/dsmr/dsmr-reader/dsmr_stats/migrations/0020_day_statistics_fix_total_gas_consumption_retroactive.py", line 61, in regenerate_data
    current_day_gas_diff = hours_gas_sum - current.gas
TypeError: unsupported operand type(s) for -: 'NoneType' and 'decimal.Decimal'
>>>>> [!] Executing database migrations failed again! <<<<<
        - Running dsmr_sqlsequencereset did not resolve the problem.
        - Create an issue on GitHub and attach the exception trace listed above.
[!] Halted. Post-deployment script exited with non-zero code
```

### DSMR-reader version

5.10

### DSMR-reader platform

Native (e.g. manual installation)

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                       v5.9 / v3.10.6 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                                 "42" / "4"
    (!) Database migrations pending                                                                     1 (!)

DATA
    Telegrams total (est.)                                                                             305467
    Consumption records electricity / gas (est.)                                               303992 / 36164
```

dennissiemensma, 2023-01-29:
Bedankt voor je melding! Zou je eens willen proberen of dit het fixt?

```
sudo su - dsmr

git fetch
git checkout -b development origin/development

./deploy.sh
```

dennissiemensma, 2023-01-29:
Thanks, dan zal ik die releasen!

----

Om terug te gaan kun je dit doen:

```
sudo su - dsmr

git checkout v5

./deploy.sh
```

closed 2023-01-29 by dennissiemensma

FutureCow, 2023-01-29:
ik heb hetzelfde probleem, opgelost met de nieuwe 5.10.1. Maar na de migrate krijg ik wel deze output:

2017-03-11 | Before: 4.569 m³ - Recalculated: 1289.238 m³ | Difference: 1284.669 m³
2017-03-12 | Before: 2.677 m³ - Recalculated: 2121.984 m³ | Difference: 2119.307 m³
2017-03-13 | Before: 3.802 m³ - Recalculated: 2194.978 m³ | Difference: 2191.176 m³
2017-03-14 | Before: 2.903 m³ - Recalculated: 2282.002 m³ | Difference: 2279.099 m³
2017-03-15 | Before: 2.287 m³ - Recalculated: 1839.808 m³ | Difference: 1837.521 m³
2017-04-09 | Before: 2.229 m³ - Recalculated: 2.315 m³ | Difference: 0.086 m³
2017-04-10 | Before: 2.168 m³ - Recalculated: 29.986 m³ | Difference: 27.818 m³
2017-04-11 | Before: 2.192 m³ - Recalculated: 89.646 m³ | Difference: 87.454 m³
2017-04-12 | Before: 3.274 m³ - Recalculated: 141.726 m³ | Difference: 138.452 m³
2017-04-13 | Before: 2.966 m³ - Recalculated: 230.300 m³ | Difference: 227.334 m³
2017-04-14 | Before: 4.143 m³ - Recalculated: 307.467 m³ | Difference: 303.324 m³
2017-04-15 | Before: 2.905 m³ - Recalculated: 170.091 m³ | Difference: 167.186 m³
2017-04-30 | Before: 1.806 m³ - Recalculated: 20.250 m³ | Difference: 18.444 m³
2017-05-01 | Before: 3.202 m³ - Recalculated: 30.243 m³ | Difference: 27.041 m³
2017-07-08 | Before: 0.392 m³ - Recalculated: 8.534 m³ | Difference: 8.142 m³
2017-09-23 | Before: 0.027 m³ - Recalculated: 0.132 m³ | Difference: 0.105 m³

Dit lijkt me niet helemaal correct?

dennissiemensma, 2023-01-29:
Bedankt voor je aanvulling. Zijn dat de enige die je ziet en van vele jaren terug? Of ook recentere? Wellicht wil je dit in #1770 zetten, want dat is het onderliggende issue.

FutureCow, 2023-01-29:
Dit zijn de enige van de afgelopen jaren, dus dat is wellicht vreemd? Na september 2017 dus geen meldingen. Heb een DSMR 5.0 meter.

dennissiemensma, 2023-01-29:
Zie #1770. Voor nu zal ik nog even die migratie uitschakelen totdat er meer duidelijkheid is dan.