dsmrreader/dsmr-reader #1796: 🙋Sagemcom S211 ( Sweden ) RuntimeError: It took too long to detect a telegram. Check connection params. Bytes currently in buffer: 829

pbiswal, 2023-01-31:
### Description

Hello,

I am using the [official docker image](https://hub.docker.com/r/xirixiz/dsmr-reader-docker) to deploy dsmr on a raspberry pi zero 1.

I have a [Sagemcom S211](https://www.ellevio.se/globalassets/content/el/elmatare-produktblad-b2c/ellevio_produktblad_fas1_s211_web2.pdf) smart meter installed.

While the setup goes by smoothly, the datalogger is not receiving any data from the P1 serial port connected through USB.

Here's what the datalogger configuration looks like
[image]

Here's the error in dsmr container I see.

```
2023-01-31 13:10:20,798 ERROR    mixins       run_once                        107 | dsmr_datalogger.management.commands.dsmr_datalogger: [!] Exception raised. Traceback (most recent call last):
  File "/app/dsmr_backend/mixins.py", line 96, in run_once
    self.run(data=self.data, **options)
  File "/app/dsmr_datalogger/management/commands/dsmr_datalogger.py", line 29, in run
    telegram = next(self.telegram_generator)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/dsmr_datalogger/scripts/dsmr_datalogger_api_client.py", line 51, in read_telegram
    raise RuntimeError(
RuntimeError: It took too long to detect a telegram. Check connection params. Bytes currently in buffer: 829
```

Here's the docker config:

```
version: "3.3"

services:
  dsmrdb:
    image: postgres:12.4-alpine
    container_name: dsmrdb
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - /docker/dsmr/dsmrdb:/var/lib/postgresql/data
    restart: always
    environment:
      - PG_TZ=Europe/Stockholm
      - TZ=Europe/Stockholm
      - POSTGRES_USER=dsmrreader
      - POSTGRES_PASSWORD=dsmrreader
      - POSTGRES_DB=dsmrreader
    healthcheck:
      # postgres is the default user, please update with
      # the DJANGO_DATABASE_USER used for dsmr-reader-docker
      # default for DSMR Reader is dsmrreader
      test: [ "CMD-SHELL", "pg_isready -U dsmrreader" ]
      interval: 10s
      timeout: 5s
      retries: 10

  dsmr:
    container_name: dsmr
    image: xirixiz/dsmr-reader-docker:latest
    container_name: dsmr
    depends_on:
      - dsmrdb

    cap_add:
      - NET_ADMIN
    links:
      - dsmrdb
    restart: always
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - /docker/dsmr/backups:/home/dsmr/app/backups
    environment:
      - TZ=Europe/Stockholm
      - DJANGO_TIME_ZONE=Europe/Stockholm
      - DJANGO_DATABASE_HOST=dsmrdb
      - VIRTUAL_HOST=localhost
      - DATALOGGER_DEBUG_LOGGING=true
      - DSMRREADER_LOGLEVEL=DEBUG
      - DSMRREADER_ADMIN_USER=*
      - DSMRREADER_ADMIN_PASSWORD=*
    ports:
      - 7777:80
      - 7778:443

    devices:
      - "/dev/ttyUSB0:/dev/ttyUSB0"

    healthcheck:
      test:
        [
          "CMD",
          "curl",
          "-Lsf",
          "http://127.0.0.1/about",
          "-o",
          "/dev/null",
          "-w",
          "HTTP_%{http_code}"
        ]
      interval: 10s
      timeout: 5s
      retries: 10
```

### DSMR-reader version

5.10.2

### DSMR-reader platform

Docker (e.g. Xirixiz)

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                    v5.10.2 / v3.11.1 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                               "None" / "4"

DATA
    Telegrams total (est.)                                                                                  -
    Consumption records electricity / gas (est.)                                                        - / -

UNRESOLVED ISSUES
    No day statistics found                                                                               now
    Waiting for the first reading ever                                                                    now
    Process behind schedule: Calculate quarter hour electricity peaks                             2 hours ago
```

pbiswal, 2023-01-31:
On a side note, when I run the following command from my raspberrypi, I get garbage data as opposed to the readable data that I see [here](http://gejanssen.com/howto/Slimme-meter-uitlezen/) 

Command I use to see the data in raspberrypi
`cu -l /dev/ttyUSB0 -s 115200 --parity=none -E q`

dennissiemensma, 2023-01-31:
Hi, would you be able to provide a sample of the data received by CU command? If the output of CU is garbage, DSMR-reader probably is reading the same tainted data. The error indicates DSMR-reader was unable to read a telegram start and end within a decent timeframe, but was able to read something nonetheless...

pbiswal, 2023-01-31:
Hey Dennis,

I have dumped the output to file and this is what it looks like.
```
h]�V��[Y[,m6Vv^VU^Ukek^_%^_^K^]#^_#^_/^[^Y^_^]^Y^]^[^]^[^Y
Q-ek^]%^_^K^]#^O#^_/^_^_^_^_^W^_^]^O#^U+)Q/-ek^]%^_^K^[#^O#^_/^_^_^_^_^_^_^_^_#^_^_^_+)Q/-ek^]%^_^K^Y#^O#^_/^_^_^_^_^_^_^[>
^Uyek^@h]�V��
```

dennissiemensma, 2023-01-31:
Well, unfortunately I am unable to assist until there is a readable telegram. I tried searching previous issues and on a (Dutch) forum as well, but no clues.

Maybe you can check on a Swedisch forum for what is causing this, or ask whoever is responsible for managing your smart meter if you need something else. 
If you were a Dutch user, I'd say your cable is broken or suffers from (eletric?) interference, but I do not know any internals for the Swedisch market, so I'd keep away from that conclusion, for now.

dennissiemensma, 2023-01-31:
It _may_ also be caused by encryption, as at least one other country (Luxembourgh?) requires users to decrypt the cable traffic with some kind of personal key. However, DSMR-reader does not support decryption.

Also, like I said before, it could be totally something else. I would not want to push your search in an incorrect direction.

closed 2023-04-04 by dennissiemensma