dsmrreader/dsmr-reader #1797: 🙋

CptChaos, 2023-01-31:
### Description

Recently, I noticed my DSMR installation wasn't working anymore. I switched to Docker and must've removed everything I had before as well. All in all: I needed to create a new installation. A lot has changed, so after some trying around, I came to the point I had my installation running: two instances of DSMR. One on my Pi, as `api_client`, the other as `api_server`. For both instances, I use Docker (xirixiz's container).

After enabling the API on the server, and setting the API key to the `api_client` and (to be sure) a container restart, the `api_client` is sending the telegrams to `api_server`. But upon doing that, it receives an `HTTP 500` every time a telegram is posted. I don't know what I am doing wrong, and I am hoping you guys can help me out.

For both DSMR instances, I use `docker-compose`. For the `api_server` the Docker-compose file is as follows:
```version: '3'

services:
  dsmrdb:
    # When using Postgres, release 13.x is supported only
    # due to the limited availability of client packages, especially for arm32v7
    image: postgres:14-alpine
    container_name: dsmrdb
    restart: always
    volumes:
      - ./dsmrdb/database:/var/lib/postgresql/data
    environment:
      - TZ=Europe/Amsterdam
      - PG_TZ=Europe/Amsterdam
      - POSTGRES_USER=dsmrreader
      - POSTGRES_PASSWORD=dsmrreader
      - POSTGRES_DB=dsmrreader
    healthcheck:
      # postgres is the default user, please update with
      # the DJANGO_DATABASE_USER used for dsmr-reader-docker
      # default for DSMR Reader is dsmrreader
      test: [ "CMD-SHELL", "pg_isready -U dsmrreader" ]
      interval: 10s
      timeout: 5s
      retries: 10

  dsmr:
    image: xirixiz/dsmr-reader-docker:latest
    depends_on:
      - dsmrdb
    container_name: dsmr
    links:
      - dsmrdb
    cap_add:
      - NET_ADMIN
    restart: always
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - ./dsmr/app/backups:/app/backups
    environment:
      - TZ=Europe/Amsterdam
      - DJANGO_TIME_ZONE=Europe/Amsterdam
      - VIRTUAL_HOST=localhost
      - DSMRREADER_OPERATION_MODE=api_server
    ports:
      - 7777:80
    # healthcheck:
    #   disable: true
    healthcheck:
      test:
        [
          "CMD",
          "curl",
          "-Lsf",
          "http://127.0.0.1/about",
          "-o",
          "/dev/null",
          "-w",
          "HTTP_%{http_code}"
        ]
      interval: 10s
      timeout: 5s
      retries: 10
``` for the `api_client` instance, the `docker-compose.yml` file is looking like this: ```version: '3.6'
services:
  dsmr:
    image: xirixiz/dsmr-reader-docker:latest
    container_name: dsmr
    restart: always
    cap_add:
      - NET_ADMIN
    volumes:
      - /etc/localtime:/etc/localtime:ro
    environment:
      - TZ=Europe/Amsterdam
      - DJANGO_TIME_ZONE=Europe/Amsterdam
      - DSMRREADER_REMOTE_DATALOGGER_API_KEYS=** REDACTED API KEY HERE **
      - DSMRREADER_REMOTE_DATALOGGER_API_HOSTS=http://192.168.0.122:7777
      - DSMRREADER_OPERATION_MODE=api_client
      - DSMRREADER_REMOTE_DATALOGGER_SLEEP=5
      - DSMRREADER_REMOTE_DATALOGGER_DEBUG_LOGGING=false
      - DSMRREADER_LOGLEVEL=WARNING
      - DSMRREADER_REMOTE_DATALOGGER_INPUT_METHOD=serial
      - DSMRREADER_REMOTE_DATALOGGER_SERIAL_PORT=/dev/ttyUSB0
      - DSMRREADER_REMOTE_DATALOGGER_SERIAL_BAUDRATE=9600
      - DSMRREADER_REMOTE_DATALOGGER_SERIAL_BYTESIZE=7
      - DSMRREADER_REMOTE_DATALOGGER_SERIAL_PARITY=E
    devices:
      - /dev/ttyUSB0:/dev/ttyUSB0
```

It maybe has something to do with the database. When I change the admin password and restart the `api_server` instance, the password is back to default too. Maybe I am missing something, but I am not sure, as the Postgres folder structure is created in the volume mount.

### DSMR-reader version

5.10.2

### DSMR-reader platform

Docker (e.g. Xirixiz)

### Debug info dump

```shell
-READER
    App / Python / Database                                                    v5.10.2 / v3.11.1 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                               "None" / "4"

DATA
    Telegrams total (est.)                                                                                 -1
    Consumption records electricity / gas (est.)                                                      -1 / -1

UNRESOLVED ISSUES
    Geen dagstatistieken gevonden                                                                          nu
    Nog nooit een meting ontvangen                                                                         nu
```

dennissiemensma, 2023-01-31:
Unfortunately this seems like a Docker issue. You should be able to check the container logs for more details. One thing I already see is the comment "When using Postgres, release 13.x is supported only" and you're using v14. You might want to try using v13 instead, as it could be the cause of the issues. Fortunately docker makes it really easy to switch container versions, but you'll lose any data stored so far (if any).

CptChaos, 2023-01-31:
> Unfortunately this seems like a Docker issue. You should be able to check the container logs for more details. One thing I already see is the comment "When using Postgres, release 13.x is supported [...]

Hmmm, that is interesting. That would mean that the example file from Xirixiz that I used, is using the wrong version of Postgres by default as well. I will try if that works, thanks for the tip!

dennissiemensma, 2023-01-31:
@Xirixiz weet jij of PostgreSQL 14 inmiddels al goed ondersteund wordt? Of geldt de comment in je sample nog steeds en staat die sample onbedoeld op 14? Los van of dat überhaupt met het huidige issue te maken heeft uiteraard. 😁

dennissiemensma, 2023-01-31:
@cptchaos any clue what telegram is sent to the server? Unfortunately the error is quite generic.

CptChaos, 2023-01-31:
The only clue I have is that it sends a telegram with both my electric and gas usages etc. As you've probably seen by the settings from `api_client`, I have an older "slimme meter" and have to use different settings in order to get a reading out of it. These settings I managed to retrieve from the settings I had before, so I know those settings work and give a readable result, it worked in the past.

xirixiz, 2023-01-31:
> @xirixiz weet jij of PostgreSQL 14 inmiddels al goed ondersteund wordt? Of geldt de comment in je sample nog steeds en staat die sample onbedoeld op 14? Los van of dat überhaupt met het huidige issue te maken heeft uiteraard. 😁

Hi, I`m using PG14 for quiet some time now, and it's working fine. I guess the line with the note about PG13 is there because a docker-compose file from the past was used. @CptChaos better to just remove it....

Looking at the issue it could be related that data is incorrectly formed that is send to the DSMR Reader (or api server in your case). It could be the serial values are incorrect. Maybe you can set ```DSMRREADER_LOGLEVEL``` to debug? @dennissiemensma do you have some sort of curl command with dummy data to validate the api_server is accepting posted data (telegrams)?

CptChaos, 2023-01-31:
@xirixiz I just used the sample mentioned on this page: https://github.com/xirixiz/dsmr-reader-docker.

Where do you want me to put the loglevel to debug? On the `api_client` or on `api_server`, or on both?

xirixiz, 2023-01-31:
Both

xirixiz, 2023-01-31:
Je kan het ook even met een Python scriptje testen wanneer ```cu``` niet beschikbaar is als package.
Je kan dan een beetje met de seriele settings spelen wanneer je geen data kan lezen, mogelijk zijn de default waarden het issue. Let op dat de api_client Docker container dan wel gestopt moet zijn, zowel met ```cu``` of met het onderstaande script.

Script:
```
#!/usr/bin/env python
import time
import serial

ser = serial.Serial(
        port='/dev/ttyUSB0',
        baudrate = 9600,
        parity=serial.PARITY_NONE,
        stopbits=serial.STOPBITS_ONE,
        bytesize=serial.EIGHTBITS,
        timeout=1
)

while 1:
        x=ser.readline()
        print (x)
```

Output moet ongeveer zo zijn iedere paar seconden:
```
xirixiz@NAS /volume1/onedrive/smarthome                                                                                                                                                                                        [7/341]-> python3 test.py
b'\x00\n'
b''
b'\xaf\xc9S\xeb5\\\xb2M\xd43\xb8\xb2-\xb1003\x8d\n'
b'\x8d\n'
b'0-0:96.\xb1.\xb1(5A\xb4\xb2\xb4\xb2\xb4\xb43030353\xb13\xb43\xb135303\xb8393\xb133\xa9\x8d\n'
b'\xb1-0:\xb1.\xb8.\xb1(\xb19\xb26\xb7.\xb850\xaa\xeb\xd7\xe8\xa9\x8d\n'
b'\xb1-0:\xb1.\xb8.\xb2(\xb1650\xb1.\xb15\xb2\xaa\xeb\xd7\xe8\xa9\x8d\n'
b'\xb1-0:\xb2.\xb8.\xb1(00000.00\xb1\xaa\xeb\xd7\xe8\xa9\x8d\n'
b'\xb1-0:\xb2.\xb8.\xb2(00000.00\xb4\xaa\xeb\xd7\xe8\xa9\x8d\n'
b'0-0:96.\xb1\xb4.0(000\xb1\xa9\x8d\n'
b'\xb1-0:\xb1.\xb7.0(0000.\xb45\xaa\xeb\xd7\xa9\x8d\n'
b'\xb1-0:\xb2.\xb7.0(0000.00\xaa\xeb\xd7\xa9\x8d\n'
b'0-0:\xb1\xb7.0.0(0999.00\xaa\xeb\xd7\xa9\x8d\n'
b'0-0:96.3.\xb10(\xb1\xa9\x8d\n'
b'0-0:96.\xb13.\xb1(\xa9\x8d\n'
b'0-0:96.\xb13.0(\xa9\x8d\n'
b'0-\xb1:\xb2\xb4.\xb1.0(3\xa9\x8d\n'
b'0-\xb1:96.\xb1.0(3\xb23\xb83\xb1303\xb1353\xb43\xb130303\xb835393\xb4333\xb13\xb2\xa9\x8d\n'
b'0-\xb1:\xb2\xb4.3.0(\xb230\xb13\xb1\xb2\xb10000\xa9(00\xa9(60\xa9(\xb1\xa9(0-\xb1:\xb2\xb4.\xb2.\xb1\xa9(\xed3\xa9\x8d\n'
b'(\xb1553\xb7.0\xb10\xa9\x8d\n'
b'0-\xb1:\xb2\xb4.\xb4.0(\xb1\xa9\x8d\n'
b'!\x8d\n'
b''
b''
b''
b'\xaf\xc9S\xeb5\\\xb2M\xd43\xb8\xb2-\xb1003\x8d\n'
b'\x8d\n'
b'0-0:96.\xb1.\xb1(5A\xb4\xb2\xb4\xb2\xb4\xb43030353\xb13\xb43\xb135303\xb8393\xb133\xa9\x8d\n'
b'\xb1-0:\xb1.\xb8.\xb1(\xb19\xb26\xb7.\xb85\xb1\xaa\xeb\xd7\xe8\xa9\x8d\n'
b'\xb1-0:\xb1.\xb8.\xb2(\xb1650\xb1.\xb15\xb2\xaa\xeb\xd7\xe8\xa9\x8d\n'
b'\xb1-0:\xb2.\xb8.\xb1(00000.00\xb1\xaa\xeb\xd7\xe8\xa9\x8d\n'
b'\xb1-0:\xb2.\xb8.\xb2(00000.00\xb4\xaa\xeb\xd7\xe8\xa9\x8d\n'
b'0-0:96.\xb1\xb4.0(000\xb1\xa9\x8d\n'
b'\xb1-0:\xb1.\xb7.0(0000.\xb4\xb4\xaa\xeb\xd7\xa9\x8d\n'
b'\xb1-0:\xb2.\xb7.0(0000.00\xaa\xeb\xd7\xa9\x8d\n'
b'0-0:\xb1\xb7.0.0(0999.00\xaa\xeb\xd7\xa9\x8d\n'
b'0-0:96.3.\xb10(\xb1\xa9\x8d\n'
b'0-0:96.\xb13.\xb1(\xa9\x8d\n'
b'0-0:96.\xb13.0(\xa9\x8d\n'
b'0-\xb1:\xb2\xb4.\xb1.0(3\xa9\x8d\n'
b'0-\xb1:96.\xb1.0(3\xb23\xb83\xb1303\xb1353\xb43\xb130303\xb835393\xb4333\xb13\xb2\xa9\x8d\n'
b'0-\xb1:\xb2\xb4.3.0(\xb230\xb13\xb1\xb2\xb10000\xa9(00\xa9(60\xa9(\xb1\xa9(0-\xb1:\xb2\xb4.\xb2.\xb1\xa9(\xed3\xa9\x8d\n'
b'(\xb1553\xb7.0\xb10\xa9\x8d\n'
b'0-\xb1:\xb2\xb4.\xb4.0(\xb1\xa9\x8d\n'
b'!\x8d\n'
b''
```

Een andere optie zou kunnen zijn deze waarden gewoon te testen op de api_client:
```
DSMRREADER_REMOTE_DATALOGGER_SERIAL_BAUDRATE=115200
DSMRREADER_REMOTE_DATALOGGER_SERIAL_BYTESIZE=8
DSMRREADER_REMOTE_DATALOGGER_SERIAL_PARITY=N
```

CptChaos, 2023-01-31:
Met `cu -l /dev/ttyUSB0 -s 9600 --parity=none` krijg ik de melding dat 'de lijn' in gebruik is en het Python script valt over module serial:
```
pi@raspberrypi:/home/pi/docker# python test.py
Traceback (most recent call last):
  File "/home/pi/docker/test.py", line 3, in <module>
    import serial
ModuleNotFoundError: No module named 'serial'
```

xirixiz, 2023-01-31:
Ja klopt, je moet de container stoppen voordat je ```cu``` gebruikt of het Python script. Er mag maar 1 proces het device uitlezen. Python werkt met libraries/modules. Je kan het package eenvoudig installeren met ```pip3 install serial --user``` of ```pip```, al moet het package pip dan ook geinstalleerd zijn :). Aangezien cu al geinstalleerd is en werkt zou ik dat gebruiken.

dennissiemensma, 2023-01-31:
Als je `cu` gebruikt (of iets anders er naast) zul je wel even DSMR-reader moeten stoppen. De seriele poort kan maar door 1 applicatie tegelijkertijd uitgelezen worden.

CptChaos, 2023-01-31:
Ah ja, dat was ik vergeten te zeggen: DSMR draaide niet.

xirixiz, 2023-01-31:
Oke, dan lijkt het er dus op dat er met de poort als iets niet goed zit. Wanneer er geen enkel proces /dev/ttyUSB0 gebruikt, dan moet je hoe dan ook het device uit kunnen lezen. Je kan even kijken met ```lsof /dev/ttyUSB0```

CptChaos, 2023-01-31:
Met `lsof /dev/ttyUSB0` krijg ik geen resultaten terug. Ik gebruikte de nieuwste versie van het package, want heb net `lsof` moeten installeren op mijn Pi.

xirixiz, 2023-01-31:
Aaah, geen rechten.... kan je ```cu``` met sudo uitvoeren.... of ```sudo chmod 777 /dev/ttyUSB0``` uitvoeren.

CptChaos, 2023-01-31:
Bingo, het was blijkbaar een rechten dingetje. Mijn telegrammen zien er als volgt uit:
```
/XMX5XMXABCE100085144

0-0:96.1.1(31333836353130372020202020202020)
1-0:1.8.1(13987.433*kWh)
1-0:1.8.2(12843.754*kWh)
1-0:2.8.1(00000.000*kWh)
1-0:2.8.2(00000.000*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(0000.45*kW)
1-0:2.7.0(0000.00*kW)
0-0:96.13.1()
0-0:96.13.0()
0-1:96.1.0(3238303131303038333037313834393133)
0-1:24.1.0(03)
0-1:24.3.0(230131220000)(00)(60)(1)(0-1:24.2.0)(m3)
(05370.189)
```

xirixiz, 2023-01-31:
Rechten zou dan ook het issue kunnen zijn op de docker container.... ik zou de boel eens herstarten nu :)

CptChaos, 2023-01-31:
Wel gek, want ik heb niets aangepast en had hiervoor ook gewoon DSMR draaien, met een iets andere setup want oudere versie.

xirixiz, 2023-01-31:
Inhoudelijk weet ik verder niet precies hoe het werkt, maar misschien dat @dennissiemensma meer kan met deze melding.

CptChaos, 2023-01-31:
Voor de volledigheid de gehele entry in de log:
```
dsmr      | 2023-01-31 22:44:42,070 DEBUG    datalogger   telegram_to_reading              81 | Received telegram:
dsmr      | /XMX5XMXABCE100085144
dsmr      |
dsmr      | 0-0:96.1.1(31333836353130372020202020202020)
dsmr      | 1-0:1.8.1(13987.433*kWh)
dsmr      | 1-0:1.8.2(12843.787*kWh)
dsmr      | 1-0:2.8.1(00000.000*kWh)
dsmr      | 1-0:2.8.2(00000.000*kWh)
dsmr      | 0-0:96.14.0(0002)
dsmr      | 1-0:1.7.0(0000.44*kW)
dsmr      | 1-0:2.7.0(0000.00*kW)
dsmr      | 0-0:96.13.1()
dsmr      | 0-0:96.13.0()
dsmr      | 0-1:96.1.0(3238303131303038333037313834393133)
dsmr      | 0-1:24.1.0(03)
dsmr      | 0-1:24.3.0(230131220000)(00)(60)(1)(0-1:24.2.0)(m3)
dsmr      | (05370.189)
dsmr      | !
dsmr      | 2023-01-31 22:44:42,071 WARNING  datalogger   telegram_to_reading              90 | Rejected telegram: Failed to perform CRC validation because the telegram is incomplete. The checksum and/or content values are missing.
dsmr      | 192.168.0.4 - - [31/Jan/2023:22:44:42 +0100] "POST /api/v1/datalogger/dsmrreading HTTP/1.1" 500 24 "-" "python-requests/2.28.1" "-"
```

Nu moet ik ook wel zeggen dat ik het wat raar vind dat ik volgens mij wel een gevulde database heb, maar wanneer ik de container van `api_server` herstart, wel weer de default inloggegevens heb. Worden die niet opgeslagen in de database?

xirixiz, 2023-01-31:
Die ```500``` error kan wel kloppen, de server ontvangt een invalid object, wat ook matcht met die crc error. Het wachtwoord wordt inderdaad niet opgeslagen, die moet je dan als vars zetten, dat kan je wel in de README vinden en staat iig los van dit issue.

CptChaos, 2023-01-31:
Thanks, @xirixiz dan ga ik dat alvast even fixen! :) Thanks voor alle hulp zover!

xirixiz, 2023-01-31:
Graag gedaan en succes verder!

dennissiemensma, 2023-02-01:
@CptChaos ik denk dat je aan de API-server zijde de Datalogger instellingen binnen de DSMR-reader admin op `Netherlands - DSMR version 2/3` moet zetten. 

Je telegram lijkt aardig overeen te komen met die legacy DSMR: https://github.com/dsmrreader/dsmr-reader/blob/v5/dsmr_datalogger/tests/datalogger/test_iskra.py#L21

```
Rejected telegram: Failed to perform CRC validation because the telegram is incomplete. The checksum and/or content values are missing
```

CptChaos, 2023-02-01:
@dennissiemensma Dat was het inderdaad, het werkt nu! Bedankt allemaal!

closed 2023-02-01 by CptChaos