dsmrreader/dsmr-reader #1799: 🚨dsmr_parser.exceptions.ParseError na update v5.10.2

SenH, 2023-02-01:
### Description

Volgende error in Docker log om de de 5 sec.

```
dsmr    | ignore line with signature \d-\d:1\.6\.0.+?\r\n, because parsing failed.
dsmr    | Traceback (most recent call last):
dsmr    |   File "/app/dsmr_parser/parsers.py", line 120, in parse
dsmr    |     telegram[signature] = parser.parse(current_match)
dsmr    |                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
dsmr    |   File "/app/dsmr_parser/parsers.py", line 238, in parse
dsmr    |     return MBusObject(self._parse(line))
dsmr    |                       ^^^^^^^^^^^^^^^^^
dsmr    |   File "/app/dsmr_parser/parsers.py", line 212, in _parse
dsmr    |     raise ParseError("Invalid '%s' line for '%s'", line, self)
dsmr    | dsmr_parser.exceptions.ParseError: ("Invalid '%s' line for '%s'", '1-0:1.6.0)(1-0:1.6.0)(230201000000W)(230114124500W)(03.332*kW)\r\n', <dsmr_parser.parsers.MBusParser object at 0x7f6e39c73b50>)
```

Lijkt iets niet goed te gaan met het parsen van 1.6.0 register (België - Sibelga)?
Zie MUCS_P1_Ed_1_7_1.pdf

```/FLU5\253769484_A

0-0:96.1.4(50217)
0-0:96.1.1(REDACTED)
0-0:1.0.0(230201155056W)
1-0:1.8.1(000143.608*kWh)
1-0:1.8.2(000170.853*kWh)
1-0:2.8.1(000002.963*kWh)
1-0:2.8.2(000000.427*kWh)
0-0:96.14.0(0001)
1-0:1.4.0(00.078*kW)
1-0:1.6.0(230201020000W)(01.566*kW)
0-0:98.1.0(1)(1-0:1.6.0)(1-0:1.6.0)(230201000000W)(230114124500W)(03.332*kW)
1-0:1.7.0(00.182*kW)
1-0:2.7.0(00.000*kW)
1-0:21.7.0(00.230*kW)
1-0:41.7.0(00.000*kW)
1-0:61.7.0(00.000*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.048*kW)
1-0:32.7.0(227.8*V)
1-0:52.7.0(000.0*V)
1-0:72.7.0(231.5*V)
1-0:31.7.0(001.60*A)
1-0:51.7.0(001.66*A)
1-0:71.7.0(001.41*A)
0-0:96.3.10(1)
0-0:17.0.0(999.9*kW)
1-0:31.4.0(999*A)
0-0:96.13.0()
!BEF2```

### DSMR-reader version

v5.10.2

### DSMR-reader platform

Docker (e.g. Xirixiz)

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                    v5.10.2 / v3.11.1 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                             "None" / "101"

DATA
    Telegrams total (est.)                                                                             383837
    Consumption records electricity / gas (est.)                                                38560 / 17840
```

dennissiemensma, 2023-02-01:
Bedankt voor je melding. In v5.10 is de [onderliggende parser](https://github.com/ndokter/dsmr_parser) bijgewerkt, juist om later ondersteuning te bieden voor dat veld.

Echter ontbreekt er nog zo'n telegram in de tests voor DSMR-reader, dus achteraf had ik die beter eerst kunnen toevoegen.

Ik zal kijken of de fix makkelijk is. Hoe dan ook kun je overwegen om te downgraden naar v5.9 tot die tijd.

## Downgraden
- Je zult dan dit script (of de commands erin) moeten uitvoeren in je installatie: https://github.com/dsmrreader/dsmr-reader/blob/v5.10.2/dsmrreader/provisioning/downgrade/v5.9.0.sh
- Daarna zou je in principe de container van v5.9 moeten kunnen draaien

dennissiemensma, 2023-02-01:
Oke de fix is relatief simpel voor DSMR-reader, namelijk dat veld hard in de parser negeren, gezien DSMR-reader er toch (nog) niets mee doet.

Ik zal kijken of ik er een v5.10.3 fix voor kan maken.

SenH, 2023-02-01:
Ik wacht nog even de fix af dan voor ik downgrade. Buiten de error logs draait het wel verder. 

Voor zover ik de error begrijp zou de `MaxDemandParser` class  moeten aangeroepen worden om dit veld te parsen in plaats van de `MBusParser`.

dennissiemensma, 2023-02-01:
Bedankt voor je pointer, ik heb hem aan #1764 toegevoegd voor later. Voor nu is die regel hard uit de kopie van de parser in DSMR-reader gehaald.

dennissiemensma, 2023-02-01:
Als alle tests slagen dan zal ik hem mergen en releasen: https://github.com/dsmrreader/dsmr-reader/actions/runs/4068640849

Ik heb je telegram met een random serienummer toegevoegd als test, zodat dit een volgorde keer niet onopgemerkt gebeurt: https://github.com/dsmrreader/dsmr-reader/commit/d88cbe8d4be551d17a7bb61d5f6ddb27ad640b56#diff-583616bc5af1bf7227f1124872ce725a3d5703dc452721874a85a8f50a6e874bR25

SenH, 2023-02-01:
Mijn DSMR meter is nog maar een paar dagen naar v1.7 (waar die velden met kwartier piek waardes inzitten) ge-upgrade. Het zou kunnen dat de error zich al vroeger manifesteerde maar ik bekijk de logs van DSMR-Reader niet elke dag want het draait verder als een zonnetje. Veel dank alvast!

dennissiemensma, 2023-02-01:
De vorige versie van DSMR-reader (of de gebruikte parser) kende dat veld niet, dus ik denk dat het veld eerder genegeerd werd.

dennissiemensma, 2023-02-01:
Het is uitgebracht: https://github.com/dsmrreader/dsmr-reader/releases/tag/v5.10.3

Wellicht dat @xirixiz morgenavond de bijbehorende Docker-release voor je wilt taggen

xirixiz, 2023-02-02:
Released!

closed 2023-02-02 by dennissiemensma