dsmrreader/dsmr-reader #1806: 🚨   Proces loopt achter op schema: Retention data rotation

nilsx, 2023-02-06:
### Description

Sinds enige tijd heb ik steeds de terugkerende foutmelding "Proces loopt achter op schema: Retention data rotation". Alles lijkt echter (vooralsnog) prima te werken.

Als ik via gepland proces - Retention data rotation handmatig het proces start is de melding even weg maar komt altijd snel weer terug.

In de gebruikelijke logs zie ik volgens mij geen errors. 
In het postgresql log zie ik wel elke minuut deze error voorbij komen:

2023-02-06 12:25:23.590 GMT [26530] ERROR:  failed to re-find parent key in index "dsmr_datalogger_dsmrreading_pkey" for deletion target page 16663
2023-02-06 12:25:23.590 GMT [26530] CONTEXT:  automatic vacuum of table "dsmrreader.public.dsmr_datalogger_dsmrreading"

Ik heb een vermoeden dat het hier mee te maken zou kunnen hebben?!

Enig idee hoe ik dit probleem kan oplossen?

### DSMR-reader version

5.10.3

### DSMR-reader platform

Native (e.g. manual installation)

### Debug info dump

```shell
SMR-READER
    App / Python / Database                                                     v5.10.3 / v3.7.3 / postgresql
    BE sleep / DL sleep / Retention / Override                                   10.0s / 15.0s / 672h / False
    Latest telegram version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                             182090
    Consumption records electricity / gas (est.)                                                74030 / 41529

UNRESOLVED ISSUES
    Proces loopt achter op schema: Retention data rotation                                      1 dag geleden
```

dennissiemensma, 2023-02-06:
Je kunt dit proberen: https://dba.stackexchange.com/a/135885

```
sudo su - postgres

reindexdb dsmrreader -v
vacuumdb -v -f dsmrreader
```

Kan even duren.

----

Check ook even of je schijf niet vol zit.
```
df -h
```

De kolom `Use%` kun je checken:
```
Filesystem                 Size  Used Avail Use% Mounted on
tmpfs                      1,6G  3,2M  1,6G   1% /run
/dev/mapper/vgubuntu-root  898G  349G  504G  41% /
tmpfs                      7,8G  537M  7,3G   7% /dev/shm
tmpfs                      5,0M  4,0K  5,0M   1% /run/lock
/dev/sda5                  703M  344M  308M  53% /boot
/dev/sda1                  511M  4,0K  511M   1% /boot/efi
tmpfs                      1,6G  144K  1,6G   1% /run/user/1000
```

Het gaat vooral om de root partitie, in bovenstaand voorbeeld:
```
/dev/mapper/vgubuntu-root  898G  349G  504G  41% /
```

dennissiemensma, 2023-02-06:
Check ook even of je een backup hebt voordat je dit doet, al zou het niet meer stuk moeten maken dan nu eventueel al is.  Vooral corrupte SD-kaartjes doen rare dingen.

nilsx, 2023-02-07:
Een reindex van de DB heeft het probleem voor nu verholpen. Dank je wel.

closed 2023-02-07 by nilsx

dennissiemensma, 2023-02-07:
Fijn, dank voor de update!