dsmrreader/dsmr-reader #1808: Custom graph from plugin / Custom grafiek via plugin💡

GVDmeijde, 2023-02-11:
### Feature

Hallo,

Ik begrijp dat er een bewuste keuze is gemaakt om geen zonnepanelen te integreren in de DSMR reader, hoofdzakelijk omdat dit lastig/niet te testen is. Als alternatief wordt voorgesteld om DSMR reader te gebruiken als 'doorgeefluik' voor P1 data.

De huidige DSMR reader voor veel mensen meer dan een simpele API gateway voor hun energie gegevens. Het heeft een fijne en overzichtelijke UI, en uitgebreide opties om gegevens toe te voegen (energie prijzen, temperatuur gegevens).
Tevens is er plugin support, deze lijkt echter beperkt tot het importeren en exporteren van de data beschikbaar via de P1 verbinding op de slimme meter.

Ik denk dat het een mooie toevoeging zou zijn, als plugins zelf grafieken kunnen toevoegen op de live views pagina. Mijns inziens is iedere grafiek een simpele lijst aan getallen (waardes) met een timestamp. Als een plugin een 'array' aan data exposed, of misschien zelfs gewoon custom data pushed naar de DSMR reader core, moet het vrij eenvoudig zijn om de bestaande infrastructuur te gebruiken om nieuwe grafieken te tonen. De temperatuur grafiek is immers ook niet zichtbaar als deze niet geïnstalleerd is.

Met deze toevoeging kunnen developers zelf plugins maken voor bijvoorbeeld hun zonnepanelen (ik zelf zal er een maken voor SolarEdge), of lokale temperatuur sensoren, energie meters (slimme stekkers), enzovoort.

dennissiemensma, 2023-02-12:
Bedankt voor je verzoek. Een beknopt antwoord is dat DSMR-reader daar niet voor gemaakt is, zowel qua architectuur als interface als plug & plug als keuze voor opslag (relationele database).

Het lange antwoord is dat DSMR-reader uit een tijd komt (~eind 2015) waarin de slimme meter nog in opkomst was en er weinig tot geen partijen waren om je slimme meter uit te lezen. Zelfs de energieleveranciers niet. 
Die tijd is (gelukkig) voorbij en dat betekent dat ik daardoor ook makkelijker keuzes kan maken. In het begin heb ik her en der nog wel wat niet-slimme-meter gerelateerd zaken ingebouwd, zoals de temperatuurmetingen/Dropbox/MinderGas/PVOutput. 
Echter, als ik DSMR-reader vandaag de dag opnieuw zou bouwen, zou denk ik geen enkele van die keuzes er nog in zitten. Ook de plugins niet.
Daarnaast zou ik het ook stukken simpeler maken met wat tools zoals een timeseries-database en grafiekentool. Die hebben zich immers bewezen en zijn ook meer bedoeld voor wat DSMR-reader doet.

De functionaliteit die je voorstelt lijkt me dan ook meer iets voor Home Assistant of soortgelijke modulaire tool. Het uitbreiden van DSMR-reader met dynamische databronnen is niet zo triviaal als dat het wellicht lijkt. Helemaal qua onderhoud.

GVDmeijde, 2023-02-12:
Jammer maar helaas.
Ik gebruik momenteel grafana om alle data te combineren. Het werkt wel maar de uiteindelijke UI is gewoon wat minder dan die van de DSMR reader.
Misschien moet ik zelf maar een tijd steken in een tool. In ieder geval bedankt!

closed 2023-02-12 by GVDmeijde