dsmrreader/dsmr-reader #1812: 💡Change all live graphs 'time scale' concurrently

gctwnl, 2023-02-17:
### Feature

Wat een geweldige tool.

Het zou prettig zijn om alle graphs 1 een keer van tijdschaal te kunnen veranderen zodat je naar meerdere graphs in combinatie kan kijken (dat is nu wat omslachtig omdat je ze allemaal apart moet veranderen)

Het zou ook prettig zijn om de graphs in hoogte te kunnen aanpassen zodat je meer graphs gelijktijdig kunt bekijken en uit de combinatie conclusies trekken

dennissiemensma, 2023-02-19:
Bedankt voor je suggestie! Ik zal er wellicht op de lange termijn een keer naar kijken. Het hangt vooral van de grafieken lib af in hoeverre dit te bouwen is.

dennissiemensma, 2023-10-23:
Ik kom helaas niet aan deze feature toe. Het vereist een rework van de hele UI.

closed 2023-10-23 by dennissiemensma