dsmrreader/dsmr-reader #1813: Support voor digitale watermeter (België)💡

verstrepeng, 2023-02-23:
### Feature

In België is roll-out gestart van slimme watermeters welke via een draadloze verbinding hun data doorgeven aan de slimme meter. Voor wat ik ervan kan terugvinden is deze data vervolgens via P1 poort beschikbaar.
https://maakjemeterslim.be/aanbieder
Zijn er plannen om dit op te nemen ?

dennissiemensma, 2023-02-23:
Bedankt voor je verzoek! Als het gegevens uit de P1-poort zelf zijn, dan is dat zeker een goede reden om het op te nemen.
Hiervoor heb ik echter tzt wel wat voorbeelddata nodig, gezien ik niet mijn eigen meter kan gebruiken.

bezigebever, 2023-04-11:
@dennissiemensma in de raw telegrams zie ik volgende helemaal onderaan
```
0-2:24.1.0(007)
0-2:96.1.1(3853455430303030323038323138)
0-2:24.2.1(230411183906S)(00028.062*m3)
```
28,062 is de stand van de watermeter.
Alvast bedankt !

bezigebever, 2023-05-29:
@dennissiemensma even een ping... Kan ik dit zelf voorzien of is er een planning ?

dennissiemensma, 2023-05-29:
Het daadwerkelijk ondersteunen van watermeters zal nog een aanzienlijke tijd duren, gezien wat bugfixes hogere prio hebben en ik er afgelopen tijd niet aan toe kom.

dennissiemensma, 2024-12-10:
@brainkiller bedankt voor je aanvulling, ik zet hem even hier:

```shell-script
.7.0(002.31*A)
0-0:96.3.10(1)
0-0:17.0.0(999.9*kW)
1-0:31.4.0(999*A)
0-0:96.13.0()
0-1:24.1.0(003)
0-1:96.1.1(37464C4F32313232313439303732)
0-1:24.4.0(1)
0-1:24.2.3(221028214000S)(00032.932*m3)
0-2:24.1.0(007)
0-2:96.1.1(3853455430303030323136343232)
0-2:24.2.1(221028213843S)(00004.332*m3)
!4394
/FLUX253770234_A
```

De focus ligt eerst op DMSR-reader overkrijgen op Docker (upgraden van DSMR-reader is teveel gedoe voor gebruikers als ze het niet in 1x werkend krijgen). Pas daarna is er ruimte voor features.

Helaas heb ik zelf zeer beperkte tijd en vele tegenslagen, waardoor het tempo er momenteel uit is.

stefaanbolle, 2025-02-09:
Hier ook +1 om de slimme watermeter te integreren