dsmrreader/dsmr-reader #1815: Ik heb dsmr reader in docker geïnstalleerd op een RP 3en kom niet ingelogd.

Kakes, 2023-02-27:
### Description

Wat mij opvalt is dat er in Container details altijd staat bij status Running for a few seconds.
[dsmr log.txt](https://github.com/dsmrreader/dsmr-reader/files/10842334/dsmr.log.txt)

### DSMR-reader version

v5.10.2

### DSMR-reader platform

None

closed 2023-02-27 by Kakes

dennissiemensma, 2023-02-27:
Bedankt voor je melding. Voor issues met de docker-versie, specifiek bij de processen in de container (die niet over de werking van DSMR-reader zelf gaan), kun je het beste door (oude) issues bij deze kijken: 
- https://github.com/xirixiz/dsmr-reader-docker/issues

Ik denk dat je sowieso in de logs zelf mag kijken, want als er iets niet goed start dan staat er meestal wel een reden in de container logs.

Kakes, 2023-02-27:
Sorry Dennis.
hartelijk dank voor je snelle reactie.
dit was mijn eerste post op github ik had te snel op enter gedrukt.
ik wilde nog melden dat de docker constant meldt  *Running for a few
seconds*
en zoals de log hieronder dat ook meldt.
Ook dat ik de volgende docker tag heb geinstalleerd docker pull
ualex73/dsmr-reader-docker

Vriendelijke groet Jan Kakes

2023-02-27 20:07:48 INFO: Start DSMR Reader - Mode=SERVER

2023-02-27 20:07:48 INFO: Executing '/usr/bin/pg_isready -h
192.168.2.25 -p 5432 -U dsmrreader -d dsmrreader -t 1'

...........................................................
2023-02-27 20:08:50

2023-02-27 20:08:50 INFO: Start DSMR Reader - Mode=SERVER

2023-02-27 20:08:50 INFO: Executing '/usr/bin/pg_isready -h
192.168.2.25 -p 5432 -U dsmrreader -d dsmrreader -t 1'

...........................................................
2023-02-27 20:09:52

2023-02-27 20:09:52 INFO: Start DSMR Reader - Mode=SERVER

2023-02-27 20:09:52 INFO: Executing '/usr/bin/pg_isready -h
192.168.2.25 -p 5432 -U dsmrreader -d dsmrreader -t 1'

...........................................................
2023-02-27 20:10:55

2023-02-27 20:10:55 INFO: Start DSMR Reader - Mode=SERVER

2023-02-27 20:10:55 INFO: Executing '/usr/bin/pg_isready -h
192.168.2.25 -p 5432 -U dsmrreader -d dsmrreader -t 1'

...........................................................
2023-02-27 20:11:57

2023-02-27 20:11:57 INFO: Start DSMR Reader - Mode=SERVER

2023-02-27 20:11:57 INFO: Executing '/usr/bin/pg_isready -h
192.168.2.25 -p 5432 -U dsmrreader -d dsmrreader -t 1'

...........................................................
2023-02-27 20:12:59

2023-02-27 20:12:59 INFO: Start DSMR Reader - Mode=SERVER

2023-02-27 20:12:59 INFO: Executing '/usr/bin/pg_isready -h
192.168.2.25 -p 5432 -U dsmrreader -d dsmrreader -t 1'

{
    "AppArmorProfile": "",
    "Args": [],
    "Config": {
        "AttachStderr": false,
        "AttachStdin": false,
        "AttachStdout": false,
        "Cmd": [
            "/app/run.sh"
        ],
        "Domainname": "",
        "Entrypoint": null,
        "Env": [
            "DSMRREADER_ADMIN_PASSWORD=admin",
[... 281 lines cut ...]
            "UserAccesses": [],
            "TeamAccesses": [],
            "Public": false,
            "AdministratorsOnly": true,
            "System": false
        }
    },
    "ProcessLabel": "",
    "ResolvConfPath":
"/var/lib/docker/containers/9bbe26bd250ebcce793765aafb3ae69b62faa50c2f3f6b1fc75d2c2b1c1672f2/resolv.conf",
    "RestartCount": 1,
    "State": {
        "Dead": false,
        "Error": "",
        "ExitCode": 0,
        "FinishedAt": "2023-02-27T17:17:22.988972245Z",
        "OOMKilled": false,
        "Paused": false,
        "Pid": 1228570,
        "Restarting": false,
        "Running": true,
        "StartedAt": "2023-02-27T17:17:24.200332179Z",
        "Status": "running"
    }
}

dennissiemensma, 2023-02-27:
Je kunt anders het beste deze image proberen: https://github.com/xirixiz/dsmr-reader-docker

Kakes, 2023-02-28:
Hoi Dennis
helaas een vergelijkbaar resultaat met de geadviseerde tag
alleen deze blijft al in de docker als oranje starting staan
hieronder staat de logging

Vriendelijke groet Jan Kakes

     nc: getaddrinfo: Name does not resolve

Testing database connectivity: 1 second(s) of 30 seconds...
Testing database connectivity: 2 second(s) of 30 seconds...
Testing database connectivity: 3 second(s) of 30 seconds...
Testing database connectivity: 4 second(s) of 30 seconds...
Testing database connectivity: 5 second(s) of 30 seconds...
Testing database connectivity: 6 second(s) of 30 seconds...
Testing database connectivity: 7 second(s) of 30 seconds...
Testing database connectivity: 8 second(s) of 30 seconds...
Testing database connectivity: 9 second(s) of 30 seconds...
Testing database connectivity: 10 second(s) of 30 seconds...
Testing database connectivity: 11 second(s) of 30 seconds...
Testing database connectivity: 12 second(s) of 30 seconds...
Testing database connectivity: 13 second(s) of 30 seconds...
Testing database connectivity: 14 second(s) of 30 seconds...
Testing database connectivity: 15 second(s) of 30 seconds...
Testing database connectivity: 16 second(s) of 30 seconds...
Testing database connectivity: 17 second(s) of 30 seconds...
Testing database connectivity: 18 second(s) of 30 seconds...
Testing database connectivity: 19 second(s) of 30 seconds...
Testing database connectivity: 20 second(s) of 30 seconds...
Testing database connectivity: 21 second(s) of 30 seconds...
Testing database connectivity: 22 second(s) of 30 seconds...
Testing database connectivity: 23 second(s) of 30 seconds...
Testing database connectivity: 24 second(s) of 30 seconds...
Testing database connectivity: 25 second(s) of 30 seconds...
Testing database connectivity: 26 second(s) of 30 seconds...
Testing database connectivity: 27 second(s) of 30 seconds...
Testing database connectivity: 28 second(s) of 30 seconds...
Testing database connectivity: 29 second(s) of 30 seconds...
Testing database connectivity: 30 second(s) of 30 seconds...[ FAIL ]
Database connectivity couldn't be verified! Please verify your
settings. Exiting...

[cont-init.d] 20-set-app-defaults: exited 1.

[cont-finish.d] executing container finish scripts...

[cont-finish.d] done.

[s6-finish] waiting for services.

[s6-finish] sending all processes the TERM signal.

[s6-finish] sending all processes the KILL signal and exiting.

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.

[s6-init] ensuring user provided files have correct perms...exited 0.

[fix-attrs.d] applying ownership & permissions fixes...

[fix-attrs.d] done.

[cont-init.d] executing container initialization scripts...

[cont-init.d] 10-adduser: executing...

usermod: no changes

-------------------------

[... 165 lines cut ...]
[ INFO ] Verifying if the DSMR Reader web credential variables have been set...

[ INFO ] Verifying database connectivity to host: dsmrdb with port: 5432...

nc: getaddrinfo: Name does not resolve

nc: getaddrinfo: Name does not resolve

nc: getaddrinfo: Name does not resolve

nc: getaddrinfo: Name does not resolve

nc: getaddrinfo: Name does not resolve

nc: getaddrinfo: Name does not resolve

nc: getaddrinfo: Name does not resolve

nc: getaddrinfo: Name does not resolve

nc: getaddrinfo: Name does not resolve

nc: getaddrinfo: Name does not resolve

nc: getaddrinfo: Name does not resolve

dennissiemensma, 2023-02-28:
Het lijkt erop dat je database niet draait. 

Het alleen wisselen van een tag is niet afdoende, je zult echt de complete docker compose config moeten gebruiken van whatever je gebruikt:
- https://github.com/xirixiz/dsmr-reader-docker
- https://github.com/ualex73/dsmr-reader-docker

DSMR-reader heeft sowieso een database nodig en mogelijk nog andere images, afhankelijk van wat je gebruikt (influxdb, etc).

Voor inhoudelijke hulp kun je het beste bij bovenstaande projecten zelf aankloppen, maar linksom of rechtsom kun je het beste de voorbeelden daar gebruiken.

Kakes, 2023-02-28:
Oke Dennis
ik ga het nog een keer proberen
Dankjewel voor de snelle reactie