dsmrreader/dsmr-reader #1816: 🚨 	Teruglevering blijft op nul staan in live grafieken

bertus24, 2023-02-28:
### Description

In het overzicht live grafieken blijft de terug levering op 0 staan. In het dashboard wordt het wel goed weergegeven. De grafieken zelf werken wel goed alleen het teruglever totaal blijft op 0 staan.

### DSMR-reader version

v5.10.3

### DSMR-reader platform

Native (e.g. manual installation)

dennissiemensma, 2023-02-28:
Bedankt voor je melding. 

- Is het een nieuwe installatie?
- Heb je een screenshot van het dashboard en grafieken? 
- Heb je een voorbeeldtelegram?

bertus24, 2023-02-28:
Installatie heeft al verschillende upgrades gehad. Is dus niet nieuw.

[image]

Als je mij helpt met hoe ik een voorbeeld telegram kan toevoegen komt deze ook.

dennissiemensma, 2023-02-28:
Bedankt voor je aanvulling.

Ik zie dat je die issue om 8:39 hebt aangemaakt en volgens je grafiek leek je op dat moment nog geen teruglevering te hebben. Doelde je daarop?
 
[image]

Heb je ook een screenshot van het dashboard? Omdat het nu avond is, kun je dit wellicht het beste morgen doen zodra je weer meer opwekt dan verbruikt.

dennissiemensma, 2023-02-28:
Idem voor het telegram, mocht het morgen op dat moment niet kloppen. Bij de pagina SUpprt of Ondersteuning kun je onderaan de pagina je laatste telegram inzien,

bertus24, 2023-03-01:
Vanwege mijn eigen stommiteit heb ik gewoon op het verkeerde moment gekeken. Sorry voor de tijd die dit je heeft gekost.

closed 2023-03-01 by bertus24

dennissiemensma, 2023-03-01:
Bedankt voor je snelle update, het had mijzelf ook kunnen overkomen hoor 🤣