dsmrreader/dsmr-reader #1824: Hallo🙋, DSMR-reader data import vanuit een werkende Domoticz Raspberry pi

GH-Jan, 2023-03-13:
### Description

Ik heb eindelijk na lang wachten de stoute schoenen aangetrokken en DSMR-reader geïnstalleerd. Ik heb hem nog niet aangesloten omdat ik al vanaf 2017 Domoticz gebruik om de slimme meter uit te lezen. Ik wil de DSMR-reader ook niet eerder aansluiten voordat ik zeker weet dat ik geen data kwijt ben en helemaal opnieuw moet beginnen (ik ben al één jaar kwijt en dat laat ik me niet nog een keer gebeuren).

De hele installatie voor de DSMR-reader was geweldig maar over de import van de Domoticz database kan ik geen duidelijk beeld krijgen wat ik moet doen.

Wie kan mij helpen?

### DSMR-reader version

v5.10.3

### DSMR-reader platform

Native (e.g. manual installation)

### Debug info dump

```shell
<!-- COPY OUTPUT BELOW -->

DSMR-READER
    App / Python / Database                                                     v5.10.3 / v3.9.2 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                               "None" / "4"

DATA
    Telegrams total (est.)                                                                                  -
    Consumption records electricity / gas (est.)                                                        - / -

UNRESOLVED ISSUES
    No day statistics found                                                                               now
    Waiting for the first reading ever                                                                    now

<!-- COPY OUTPUT ABOVE -->
```

GH-Jan, 2023-03-13:
> Bedankt voor je vraag. Je kunt het beste wat oude issues doorzoeken, zoals deze:
>
> #908 (comment)
>
> Er is namelijk vanuit mij geen officiele manier om de import te doen, dat is door eerdere Domoticz-gebruikers in elkaar gezet.
> DSMR-reader biedt alleen wat mogelijkheden via de API voor het aanmaken van data, niet specifiek voor Domoticz, dus daar zit meestal een scriptje of omzetting aan vast.
>
> Verder is het goed om te weten dat:
>
> je wellicht beiden zelfs naast elkaar kan draaien, ik raad dat sowieso altijd in het begin aan (mtis dat kan). Als Domoticz (net als DSMR-reader) ook een seriele netwerkpoort via ser2net kan uitlezen, zitten ze elkaar niet in de weg.  ser2net leest dan je meter uit en stelt de data beschikbaar op je Pi op een bepaalde poort, met het grootste veschil dat die meerdere lezers ondersteunt. Dit is echter een software-matig proces en gaat dus niet via een hardware-splitter.
>
> het wel of niet verliezen van data hangt sterk samen met de hardware die je gebruikt en backups. Ik ken niet de oorzaak van je jaar dataverlies, maar DSMR-reader is daar niet perse beter in. In 99% van de gevallen raakt de SD-kaart corrupt en ben je alsnog alles kwijt.
>
> DSMR-reader doet een hoop dingen prima, maar ook daar zitten momenteel wat imperfecties in qua dataverwerking, dus manage je verwachtingen goed

Oh wouw, dat is een snelle reactie. Dat had ik niet verwacht.Het verlies van een jaar kwam door m’n eigen stomme schuld. Domoticz had een update en ik denk dat ik iets verkeerds heb gedaan en ik had dus ook geen backup gemaakt.Beide draaien nu op ‘n raspberry pi.Ik zal die #908 eens gaan bekijken. Ik heb er al enkele gelezen maar snap niet wat ze willen. Ik heb ook chatGPT gevraagd en ook dat klopt niet (waarschijnlijk vanwege een oudere versie o.i.d.)Alvast bedankt en als jij de handleiding hebt geschreven dan bij deze mijn complimenten. Erg duidelijk en alles klopte.Met vriendelijke groet,Jan van Tiggelen

dennissiemensma, 2023-03-13:
Bedankt voor je aanvulling. Je hebt hier helaas wel wat (oppervlakkige) kennis van applicaties/databases voor nodig om het allemaal werkend te krijgen. Het overzetten van data tussen systemen is zeker niet triviaal.

En anders kun je nog overwegen om DSMR-reader gewoon ernaast te draaien vanaf dit moment. 
Zolang je de dataopslag van beide applicaties regelmatig backupt, kun je altijd in de toekomst er opnieuw naar kijken.

GH-Jan, 2023-03-17:
Goede avond Dennis,

 

Ik denk dat ik een API import heb kunnen uitvoeren.

Hoe zou ik dat kunnen controleren dat er enkele jaren aan data in zit?

 

Groet, Jan van Tiggelen

 

Closed #1824 <https://github.com/dsmrreader/dsmr-reader/issues/1824>  as not planned.

dennissiemensma, 2023-03-17:
Je kunt het beste handmatig checken op de URL `/archive` of een CSV bekijken via `/export`.