dsmrreader/dsmr-reader #1825: 🙋 Hoe te gebruiken met API en zonder USB/netwerk uitleesmethode

hmmbob, 2023-03-16:
### Description

Ik gebruik een SlimmeLezer die ik alle telegram-parsing laat doen (o.a. omdat ik "current L1/L2/L3" rechtstreeks naar mijn laadpaal wil pushen). Nu wil ik die waardes via de dsmr-reader API naar dsmr-reader pushen, zodat mijn statestieken toch mooi bij elkaar blijven. Het ESPHome gedeelte om te pushen werkt, maar de datalogger herstart zichzelf steeds omdat hij geen verbinding kan maken met de seriele output.

Hoe kan ik DSMR-reader vertellen om de seriele datalogger te negeren en alleen naar de API te luisteren?

### DSMR-reader version

5.10.3

### DSMR-reader platform

Docker (e.g. Xirixiz)

dennissiemensma, 2023-03-16:
Wellicht zoek je dit: https://github.com/xirixiz/dsmr-reader-docker#dsmr-datalogger-related

hmmbob, 2023-03-16:
Ja! Dank, eens kijken of ik dit aan de praat krijg.

hmmbob, 2023-03-16:
Toch heel de docs doorgespit, en hier maar een paar keer ofzo overheen gelezen. Maar: het werkt!

```
[20:18:40][D][sensor:126]: 'Energy Consumed Tariff 1': Sending state 4035.79492 kWh with 3 decimals of accuracy
[20:18:40][D][sensor:126]: 'Energy Produced Tariff 1': Sending state 0.00000 kWh with 3 decimals of accuracy
[20:18:40][D][sensor:126]: 'Energy Consumed Tariff 2': Sending state 4072.12988 kWh with 3 decimals of accuracy
[20:18:40][D][sensor:126]: 'Energy Produced Tariff 2': Sending state 0.00000 kWh with 3 decimals of accuracy
[20:18:40][D][sensor:126]: 'Power Consumed': Sending state 2.31200 kW with 3 decimals of accuracy
[20:18:40][D][sensor:126]: 'Power Produced': Sending state 0.00000 kW with 3 decimals of accuracy
[20:18:40][D][sensor:126]: 'Voltage Phase 1': Sending state 234.00000 V with 1 decimals of accuracy
[20:18:40][D][sensor:126]: 'Voltage Phase 2': Sending state 233.00000 V with 1 decimals of accuracy
[20:18:40][D][sensor:126]: 'Voltage Phase 3': Sending state 231.00000 V with 1 decimals of accuracy
[20:18:40][D][sensor:126]: 'Current Phase 1': Sending state 0.00000 A with 1 decimals of accuracy
[20:18:40][D][sensor:126]: 'Current Phase 2': Sending state 8.00000 A with 1 decimals of accuracy
[20:18:40][D][sensor:126]: 'Current Phase 3': Sending state 0.00000 A with 1 decimals of accuracy
[20:18:40][D][sensor:126]: 'Power Consumed Phase 1': Sending state 0.08600 kW with 3 decimals of accuracy
[20:18:40][D][sensor:126]: 'Power Consumed Phase 2': Sending state 2.04900 kW with 3 decimals of accuracy
[20:18:40][D][sensor:126]: 'Power Consumed Phase 3': Sending state 0.17300 kW with 3 decimals of accuracy
[20:18:40][D][sensor:126]: 'Power Produced Phase 1': Sending state 0.00000 kW with 3 decimals of accuracy
[20:18:40][D][sensor:126]: 'Power Produced Phase 2': Sending state 0.00000 kW with 3 decimals of accuracy
[20:18:40][D][sensor:126]: 'Power Produced Phase 3': Sending state 0.00000 kW with 3 decimals of accuracy
[20:18:40][D][sensor:126]: 'Gas Consumed': Sending state 2022.44800 m³ with 3 decimals of accuracy
[20:18:40][D][text_sensor:067]: 'DSMR Timestamp': Sending state '230316201831W'
[20:18:41][D][text_sensor:067]: 'Timestamp': Sending state '2023-03-16T19:18:40Z'
[20:18:42][D][http_request:099]: HTTP Request completed; URL: http://192.168.x.x/api/v2/datalogger/dsmrreading; Code: 201; Duration: 133 ms
```

closed 2023-03-16 by hmmbob