dsmrreader/dsmr-reader #1826: dropbox

style2k, 2023-03-17:
### Description

🙋wat een bagger gedoe . ik zit nog op een oudere versie en mij dropbox backup werkt niet meer heb alles geprobeerd maar helemaal nix. nu ben ik op een nieuwe pi de hele zooi aan instaleren en moet nu de backup importeren maar ik heb geen backup bestand omdat dropbox niet werkt mail werkt blijkbaar ook niet en als ik handmatig importeer krijg ik csv bestanden . echt wat een gedoe om die zooi te instaleren . en de lokale backup van de huidige pi krijg ik niet overgezet om een of andere reden . waarom is er niet gewoon een image die je kan flashen een import functie voor een backup en klaar . waarom moet een linux pakket altijd zo ingewikkeld

### DSMR-reader version

3.x

### DSMR-reader platform

Other

style2k, 2023-03-18:
niemand ?????? ik moet de backup bestanden ervan af halen om een nieuwe instalatie te doen maar als ik manueel exporteer krijg ik csv bestanden en bestanden die ik volgens de handleiding moet gebruiken zijn .gz bestanden . ??? hoe krijg ik die bestanden eraf als ik remote desktop inlog hangt de hele pi .

dennissiemensma, 2023-03-18:
Hier vind je informatie voor het maken van de nodige backup:
- https://github.com/dsmrreader/dsmr-reader/issues/1817#issuecomment-1448676680

dennissiemensma, 2023-03-18:
DSMR-reader is open-source en gratis, maar vraagt enige Linux-kennis van de gebruiker. Er zijn ook container/Docker-varianten (soort van images) die het installeren en upgraden supersimpel maken, zoals dit project:
- https://github.com/xirixiz/dsmr-reader-docker

Echter vereist dat alsnog Linux know-how, dus ben je daar vermoedelijk ook niet mee geholpen.

Ik raad je verder sterk aan om over te stappen naar een gebruiksvriendelijker alternatief voor DSMR-reader. Tegenwoordig zijn er tig plug & play oplossingen die het je erg makkelijk maken en je een hoop frustratie kunnen schelen.

style2k, 2023-03-18:
nah het is de eerste keer ook gelukt en nu ben ik het op een andere pi aan instaleren . het probleem is dat ik niet bij de database backup kan . naar docker wil die niet meer backuppen heb verschillende keren een andere token aangemaakt maar het werkt nie ten de email backup werkt ook gewoon niet .  en zonder backup kan ik gewoon niks terug zetten.  docker heb ik geprobeerd en ook hier gevraagd eerder maar geen antwoord gekregen en daarbij is er nergens een uitleg gegeven om de docker versie te instaleren. ik heb meerdere docker container geinstalleerd zonder problemen alleen deze is nooit gelukt. was geinstaleerd maa rwerkte gewoon weg niet , dus ja kan kennis hebben wat ik wil maar het werkt gewoon niet en nergens is uitleg te vinden . dat is het probleem

style2k, 2023-03-18:
March 18, 2023: Unable to authenticate with Dropbox, removing credentials. Error: AuthError('68a51960c6654b5f8a34d67c0a35a347', AuthError('invalid_access_token', None))

deze melding krijg ik terwijl ik de acces token vernieuwd heb 
maar na het opslaan als ik weer t erug dta menu in ga is dat veld gewoon leeg 
hij slaat de token gewoon niet op

dennissiemensma, 2023-03-19:
- In welke installatie krijg je die foutmelding, je oude of je nieuwe?
- En is het inmiddels gelukt om de database backup **handmatig** te maken op de command-line in je oude installatie en over te zetten/importeren in je nieuwe installatie?

style2k, 2023-03-19:
in de oude , de nieuwe ben ik bij de stap dat ik de database moet importeren maar krijg de locale backup er niet echt vanaf. ik kan het kaartje niet in windows lezen. als ik via remote desktop inlog en de bestanden kopieer en in windows plak werkt dat niet en als ik het bestand via remote desktop op me nas probeer te zetten loopt de hele boel een beetje vast . ook met een usb stik wil het om wat voor reden niet lukken .

style2k, 2023-03-21:
ja heb iets gevonden wat ik al had . winscp die logt in via ssh maar dan in een soort ftp style en daarmee kun je gewoon bestanden over zetten. dus ga zo ff kijken of ik de backups in de database kan terug zetten

style2k, 2023-03-25:
sudo -u postgres psql -c "alter user dsmrreader with password 'dsmrreader';"  wat is nu het commando het geen wat tussne haakjes staat is rood wat houd dat in?? heb het vorige keer ook goed gedaan maar de handleiding is veranderd ?

style2k, 2023-03-26:
[image]

ik ben bij het maken van een wachtwoord van de database maar wat is nu goed ik krijg in principe met 2 keer precies hetzelfde commando alleen een ander wachtwoord zeg maar 2 verschillende resultaten.

style2k, 2023-03-26:
ik heb de database nu overgezet alleen nu zegt de handleiding dat ik sudo su - dsmr moet doen alleen die user bestaat nog niet kan ik hierna gewoon met de snelle instalatie van dsmr reader verder of ? want nu klopt de volgorde niet in de instalatie staat instal database dan data base importeren en daarna word er pas een user aangemaakt ? 

[image]

dennissiemensma, 2023-03-27:
Je kunt bovenstaande stappen na die lange regel _skippen_ en dan verder gaan met de gewone installatie. Die `dsmr` gebruiker en deploy en alles volgt dan later tijdens de installatie alsnog.