dsmrreader/dsmr-reader #1828: 🚨Dag/maand/jaartotalen wijken structureel af van daadwerkelijk verbruik

Tralapo, 2023-04-01:
### Description

Ik weet niet of hier sprake is van een bug of bewust design, maar ik loop er tegenaan dat totalen (dag/maand/jaar) in de "Archief" sectie niet stroken met de daadwerkelijke verbruik. Voorbeelden:

> **Gas, februari 2023**
> Meterstand 01-02-2023 00:00:00: 1.589,073
> Meterstand 01-03-2023 00:00:00: 1.680,887
> 
> Dat zou moeten uitkomen op een totaal verbruik van: 91,814 m³
> 
> Wat DSMR aangeeft op Archief > Maanden > feb: 87,851 m³.
> 
> **Gas, maart 2023**
> Meterstand 01-03-2023 00:00:00: 1.680,887
> Meterstand 01-04-2023 00:00:00: 1.768,591
> 
> Dat zou moeten uitkomen op een totaal verbruik van: 87,704 m³
> 
> Wat DSMR aangeeft op Archief > Maanden > mrt: 80,568 m³.

Bij stroomverbruik zie ik ook afwijkingen, maar wel iets minder extreem:

> **Stroom, januari 2023**
> Meterstand 01-01-2023 00:00:00 (gecombineerd): 4.223,180
> Meterstand 01-02-2023 00:00:00 (gecombineerd): 4.395,427
> 
> Dat zou moeten uitkomen op: 172,247
> 
> Wat DSMR aangeeft op Archief > Maanden > jan: 172,167
> 
> **Stroom, februari 2023**
> Meterstand 01-02-2023 00:00:00 (gecombineerd): 4.395,427
> Meterstand 01-03-2023 00:00:00 (gecombineerd): 4.551,298
> 
> Dat zou moeten uitkomen op: 155,871
> 
> Wat DSMR aangeeft op Archief > Maanden > jan: 155,794

Zoals al aangegeven weet ik dus niet of het hier een bug betreft of een "by design" ding? Worden de totalen berekend op basis van de daadwerkelijke meterstanden (lijkt mij het meest betrouwbaar), of maakt DSMR een som van alle binnengekomen metingen?

Het komt best eens voor dat DSMR (tijdelijk) uit is door bijvoorbeeld een update van de software zelf, of het OS met bijbehorende reboot, of een andere overload van het systeem (komt helaas af en toe voor op mijn Pi). 

Dat na zo'n moment het actuele verbruik voor die dag niet meer klopt komt logisch op mij over, maar ik heb altijd gedacht dat voor de uiteindelijke archief gegevens gekeken werd naar de meterstanden omdat daarvoor maar een som van twee opnames nodig is.

### DSMR-reader version

v5.10.3

### DSMR-reader platform

Docker (e.g. Xirixiz)

dennissiemensma, 2023-04-01:
Bedankt voor je melding. Ik vermoed dat de oorzaak #1811 is, waar ik nog niet aan toe ben gekomen. Ik verwacht uiteindelijk wel dat de fix automatisch met terugwerkende kracht kan.

Sinds 5.10.x staan de meterstanden wel in de dag-weergave van het archief, dus daar kun je voor nu de afrekening(en) mee vergelijken. Totdat de fix klaar is.

dennissiemensma, 2023-04-01:
## Follow-up in #1811

closed 2023-04-01 by dennissiemensma