dsmrreader/dsmr-reader #1836: 🙋

aroe0546, 2023-04-14:
### Description

Is het mogelijk om de oude database te verwijderen en opnieuw te beginnen. 

Ik ben verhuist en wil graag weer van 0 beginnen met meten. 

### DSMR-reader version

5.10.3

### DSMR-reader platform

Native (e.g. manual installation)

dennissiemensma, 2023-04-16:
Het makkelijkste is om je hele database te verwijderen, alleen ben je dan ook instellingen kwijt. 

-----

Dit zou ook moeten werken door alleen specifieke data weg te gooien:
```
sudo su - postgres
psql dsmrreader

truncate table dsmr_datalogger_dsmrreading;
truncate table dsmr_consumption_electricityconsumption;
truncate table dsmr_consumption_gasconsumption;
truncate table dsmr_consumption_quarterhourpeakelectricityconsumption;
truncate table dsmr_datalogger_meterstatistics;
truncate table dsmr_stats_daystatistics;
truncate table dsmr_stats_electricitystatistics;
truncate table dsmr_stats_hourstatistics;
```

closed 2023-04-17 by dennissiemensma

dennispracht, 2023-08-02:
Ik krijg dit te zien als ik die script gebruik. Waarschijnlijk doe ik iets verkeerd.

[image]