dsmrreader/dsmr-reader #1837: 🙋 Dagstatistieken lopen achter

syphernl, 2023-04-14:
### Description

Gisterochtend is mijn 1-fase meter vervangen door een een 3-fase (en DSMR4 => SMR 5.0). Na het opnieuw inpluggen zag ik netjes alle data (inclusief L2/L3) en leek alles te werken.

Wel een flinke negatieve piek, die ik vandaag wilde fixen. 
Maar... blijkbaar is er iets mis met het genereren van de dagtotalen binnen mijn DSMR-Reader instance.

De webinterface geef 1 issue weer:  "Dagstatistieken lopen achter  (1 dag, 12 uur geleden)". 
Normaliter ontvang ik om 06:00 een telegram bericht met het verbruik van vorige dag, maar donderdag heb ik dat ook niet gehad.

Aangezien de wissel minder dan 36 uur geleden was lijkt het haast alsof dit probleem wellicht niet gerelateerd is aan de wissel? Maar het is mij verder niet zo duidelijk hoe ik dat kan doen. 

Herstarten en reboots hebben helaas niet geholpen. De debug logs laten ook niets geks zien voor backend of datalogger.

### DSMR-reader version

v5.10.3

### DSMR-reader platform

Native (e.g. manual installation)

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                   v5.10.3 / v3.10.11 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 168h / False
    Latest telegram version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                             234376
    Consumption records electricity / gas (est.)                                                82349 / 33575

UNRESOLVED ISSUES
    Day statistics are lagging behind                                                     1 day, 12 hours ago
```

syphernl, 2023-04-14:
Ik heb `./manage.py dsmr_stats_reconstruct_missing_day_statistics` gedraaid die ik toevallig tegen kwam. Daarna is de melding weggegaan (en kreeg ik direct een Telegram over verbruik van donderdag (met negatieve meetwaarde)

dennissiemensma, 2023-04-16:
Bedankt voor je vraag. Welke meldingen blijven nu nog over? 

Je kunt het beste de logs induiken:
- https://github.com/dsmrreader/dsmr-reader/issues/1230#issuecomment-742636653

Daar zul je al een reden kunnen vinden. Mocht er geen gasmeting binnenkomen, dan is het goed om een telegram te delen. Soms zijn ze niet goed aangesloten na een wissel.

syphernl, 2023-04-17:
Ik heb in totaal 2 dagen achter elkaar de `dsmr_stats_reconstruct_missing_day_statistics` moeten draaien, daarna is het ineens weer gaan lopen. 
Dat is ook iets wat degene van het gelinkte topic ook had. Wellicht iets wat in de "meter vervangen" FAQ toegevoegd kan worden?

Wat betreft de gas metingen, de monteur heeft blijkbaar de gasmeter niet gekoppeld. Dat heb ik gisteravond zelf gedaan dus ga ik er vanuit dat ik binnenkort weer gaswaardes door ga krijgen.

closed 2023-04-17 by syphernl

dennissiemensma, 2023-04-17:
Dank voor je update, fijn dat het weer werkt!

Doorgaans is dat command niet nodig, omdat het reguliere proces dat ook al elke nacht doet. De oorzaak is dan de gasmeter geweest. 
DSMR-reader weigert dagstatistieken te genereren wanneer er nog geen gasmeting op de volgende dag is geweest (het kan soms meer dan een uur duren).

syphernl, 2023-04-18:
> De oorzaak is dan de gasmeter geweest. DSMR-reader weigert dagstatistieken te genereren wanneer er nog geen gasmeting op de volgende dag is geweest (het kan soms meer dan een uur duren).

Ik heb alleen op dit moment nog steeds geen gaswaardes maar de dagstatistieken worden al wel weer een paar dagen netjes gegenereerd. 
Van wat ik begreep kan het 2 dagen duren voordat de gasmeter na activatie waardes door gaat sturen dus die zou het morgen wel weer moeten doen 🤞🏻