dsmrreader/dsmr-reader #1838: Migratie / Upgrade van v3.12 naar v5.x (docker)🙋

MarsWarrior, 2023-04-17:
### Description

Ik heb nog een wat oude docker installatie draaien met v3.12 die ik wil migreren/upgraden naar v5.x.
De bestaande maakt tevens gebruik van een remote datalogger op een RPI (aangesloten op LAN). Ook dit wil ik wijzigen door ser2net te gaan gebruiken op de RPI. Dat maakt het beheer wat makkelijker.

Aangezien ik docker draai, kan ik beide versies tegelijkertijd draaien. De nieuwe omgeving gebruikt postgres 14.

Ik zag in de documentatie dat een upgrade van v4 naar v5 handmatig is vanwege wijzigingen in oa omgevingsvariabelen. Of is de database ook een punt?

Ik kan er niet echt achter komen wat nu het meest eenvoudige upgrade pad in mijn geval is:

- Moet ik dat via een v4 doen?
- Of kan ik een export database (v3.12) en import database (v5.x) doen?
- Moet ik nog wat scripts draaien om alle nieuwe totalen enzo ook te krijgen over alle jaren dat ik data heb?

Dat ik de omgevingsvariabelen moet aanpassen is een feit, net als dat ik nu geen remote datalogger meer ga gebruiken, maar de lokale datalogger die via een ip:port adres is gekoppeld aan de RPI.

Mijn voornaamste punt zit dus in de database en het bepalen van alle nieuwe totalen!

### DSMR-reader version

v3.12

### DSMR-reader platform

Docker (e.g. Xirixiz)

dennissiemensma, 2023-04-17:
Elke major upgrade (zoals 3.x -> 4x) van DSMR-reader bevat (verzamelde) wijzigingen die niet compatible zijn met de vorige versie. 
Soms is het een config die bijgewerkt moet worden, vaak is het een oude Python versie die ik drop. Soms upgrade ik het onderliggende Django-framework, wat weer gevolgen heeft voor andere dependencies.

In alle gevallen is een upgrade via Docker het makkelijkste, mits je een beetje door hebt hoe containers werken. 

----

Dit zijn de regulieren paden, maar dat is nog zonder Docker:
- https://dsmr-reader.readthedocs.io/en/v5/tutorial/upgrading/to-v4.html
- https://dsmr-reader.readthedocs.io/en/v5/tutorial/upgrading/to-v5.html

Voor **v4** heb je denk ik niets nodig, gezien de Docker container dat allemaal al doet in nieuwere versies. Voor **v5** moet je even kijken of je nog influxDB gebruikt. 
Zo niet, dan is het denk ik beperkt aan env vars instellen, maar dat moest je sowieso al doen. Zie daarvoor:
- https://github.com/xirixiz/dsmr-reader-docker#setup--parameters

----

- Ik denk dat je kunt proberen om een DSMR-reader v5.x container naast je huidige te draaien. Ook als die nog niet aan een seriele poort kan, kun je kijken of je in ieder geval de interface draaiende krijgt.

- Daarna is het zaak om te zorgen dat `ser2net` werkt, zodat je daarna je dummy v5.x installatie kan aansluiten en zien of dat werkt.

- Als de metingen binnenkomen (draai het voor een paar minuten), dan kun je die DSMR-reader 5.x container stoppen en de bijbehorende DSMR-reader 5.x database (waar bijna niets in staat) verwijderen.

- Je kunt daarna een export maken van je DSMR-reader v3.x database en importeren in je (nieuwe) Postgres. 

- Als je daarna weer de DSMR-reader 5.x container start of herbouwt, doet die als het goed is een migratie van de v3.x 
 database zodat die compatible is naar 5.x.

- Als dat lukt en de DSMR-reader 5.x interface werkt met je oude data, kun je direct weer `ser2net` instellen als bron in je nieuwe DSMR-reader 5.x installatie.

Als een van de migratiestappen niet lukt, kun je altijd weer je oude installatie opspinnen voor een andere poging later.

MarsWarrior, 2023-04-18:
Ik heb als eerste `ser2net` geprobeerd. Dat was ff wat werk want mijn RPI 2 draait nog op Jessie. Die wordt niet meer ondersteund, dus ik moest eerst wat repositories aanpassen om überhaupt `ser2net` te kunnen installeren.

Daarna `systemctl stop supervisor` om de remote datalogger te stoppen en getest met
 `ser2net -C 2000:raw:600:/dev/ttyUSB0:115200:7DATABITS:NONE:1STOPBIT`.

Dat werkt zowaar:
[image]

Ik zie wel verschillende `ser2net` configuraties: sommigen gebruiken 8 databits. Ik heb nu de info uit de remote datalogger gebruikt. Overigens werken zowel 7 als 8 databits. Ik zie geen verschil en DSMR Reader v5.x kan in beiden gevallen de data lezen.

De oude v2.19 `ser2net` op Jessie ondersteund helaas (nog) geen meerdere verbindingen, dus de optie om zowel de oude als nieuwe DSMR Reader container tegelijkertijd te koppelen is ff geen optie.

Volgende stap wordt kijken naar database export/import.

MarsWarrior, 2023-04-18:
Export en weer import 2.5GB data is goed gegaan. Moest wel de pg14 instelling van SCRAM naar MD5 terugzetten om dsmr nog te laten inloggen op pg14.

De scripts in de docker container om de database bij te werken doen het dus uitstekend!

Alles lijkt er te zijn. Ik mis enkel de lopende dagtotalen. Worden die niet achteraf berekend, of mis ik een instelling, want ook als ik ff live data laat binnenkomen zie ik die niet.

[image]

En ik zie ook dat uurtotalen zijn berekend, maar daar zijn geen grafieken van?

En de kwartiertotalen is typisch iets voor Belgie? Of zou ik die ook moeten zien, want die grafiek is nu leeg. Ik heb een dsmr4.2 meter uit 2016 als ik het goed heb.

[image]

Ik loop nog verder dingen na, voordat ik alles definitief overzet. Vanzelfsprekend deed de live verbinding het niet, omdat ik natuurlijk de remote datalogger config ook overneem met die export/import 😄

MarsWarrior, 2023-04-21:
Nadat ik Home Assistant met Zigbee2Mqtt heb overgezet heb ik nu definitief ook DSMR Reader overgezet. 

Met al het exporteren, importeren, database (laten) migreren en het configureren van de datalogger ben ik totaal 10 minuten bezig geweest. Het exporteren van 2.5GB duurde al ruim 2 minuten in de oude omgeving 😨 

Ik heb de volgende stappen uitgevoerd:

**Database:**

1. Exporteren database in oude omgeving:
``` 
docker exec -i <dsmr containernaam> ./manage.py dsmr_backup_create --full
```
Daarmee wordt in de backup map `backups\_data\manually` de backup neergezet.

2. Database via Windows Verkenner / WinSCP overzetten naar nieuwe omgeving in de `\backups` map.
3. DSMR container in nieuwe omgeving stoppen
4. In de terminal van de nieuwe omgeving naar de `backup` map gaan waar de database backup staat.
5. Bestaande database verwijderen:
```
docker exec -i <dsmrdb containernaam> dropdb -U dsmrreader dsmrreader
```
6. Nieuwe database aanmaken:
```
docker exec -i <dsmrdb containernaam> createdb -U dsmrreader -O dsmrreader dsmrreader
```
7. Gezipte database in de database importeren
```
zcat dsmrreader-postgresql-backup-Friday.sql.gz | docker exec -i ha-dsmrdb psql -U dsmrreader dsmrreader
```

En dat was het qua export / import.

**De datalogger:**

1. Op de RPI waar de P1 meter aanhangt omzetten van de remote datalogger naar ser2net:
```
systemctl stop supervisor
systemctl start ser2net
```

2. Datalogger configuratie aanpassen om over een IP adres en poort de P1 data op te halen.

**MQTT:**
1. Configuratie van MQTT server aanpassen. Ook deze draait in een nieuwe omgeving namelijk.

En toen draaide alles 🥳 

Nu alleen nog alle foutmeldingen / waarschuwingen in de Home Assistant log oplossen. Er is namelijk het een en ander gewijzigd sinds ik 6 maanden geleden heb geupdate 😁 

Bedankt voor de uitleg en het stappenplan.
Verder werkt de automatisch docker migratie blijkbaar als een trein. Heb dit namelijk 3x geoefend om alle stappen juist te hebben zodat ik ze direct na elkaar kan uitvoeren.

Elke keer had ik alle data!

closed 2023-04-21 by MarsWarrior

dennissiemensma, 2023-04-23:
Dank voor de uitgebreide toelichting, fijn om te horen!