dsmrreader/dsmr-reader #1839: 🙋Geen teruglevering zichtbaar in live / dashboard

BjornB2, 2023-04-17:
### Description

Sinds vanmiddag zonnepanelen (yay!) maar ik denk dat ik ergens een instelling in DSMR niet goed heb want ik zie de teruglevering niet terug in Live en dashboard. Ik zie de teruglevering wel in admin -> DSMR readings en in MQTT. Moet ik ergens iets aanzetten? Ik heb me rotgezocht maar nog niets kunnen vinden. 
[image]
[image]
[image]

### DSMR-reader version

5.10 docker op Synology

### DSMR-reader platform

Docker (e.g. Xirixiz)

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                    v5.10.3 / v3.11.2 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 0.5s / 672h / False
    Latest telegram version read / Parser settings                                                 "42" / "4"

DATA
    Telegrams total (est.)                                                                             310938
    Consumption records electricity / gas (est.)                                               112387 / 34639

UNRESOLVED ISSUES
    Proces loopt achter op schema: Calculate quarter hour electricity peaks                      52 minuten geleden
```

dennissiemensma, 2023-04-17:
Je zou kunnen kijken in de admin in `/admin/dsmr_backend/backendsettings/` of daar het vinkje aanstaat om teruglevering geforceerd te verbergen.

BjornB2, 2023-04-17:
Ahh dat was hem! Ik draai DSMR al eeuwen en zal dat ooit aangezet hebben. 

Thanks voor je snelle reactie en uiteraard ook voor DSMR Reader in het algemeen 😊

dennissiemensma, 2023-04-17:
Top, fijn om te horen!

closed 2023-04-17 by dennissiemensma