dsmrreader/dsmr-reader #1841: Notificatie scherm geeft "Server Error" na migratie 3.12 -> 5.10.3🚨

MarsWarrior, 2023-04-22:
### Description

Naar aanleiding van de migratie in #1838 

Alle schermen doen het voor zover ik nu heb nagekeken.

Echter het notifications scherm geeft foutmeldingen:
```
Server Error
Sorry, something unexpected happened.
Exception:
NoReverseMatch: Reverse for 'v4-upgrade-redirect' not found. 'v4-upgrade-redirect' is not a valid view function or pattern name.
Traceback:
  File "/usr/local/lib/python3.11/site-packages/django/core/handlers/base.py", line 204, in _get_response
    response = response.render()
               ^^^^^^^^^^^^^^^^^

  File "/usr/local/lib/python3.11/site-packages/django/template/response.py", line 105, in render
    self.content = self.rendered_content
                   ^^^^^^^^^^^^^^^^^^^^^

  File "/usr/local/lib/python3.11/site-packages/django/template/response.py", line 83, in rendered_content
    return template.render(context, self._request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/local/lib/python3.11/site-packages/django/template/backends/django.py", line 61, in render
    return self.template.render(context)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/local/lib/python3.11/site-packages/django/template/base.py", line 170, in render
    return self._render(context)
           ^^^^^^^^^^^^^^^^^^^^^

  File "/usr/local/lib/python3.11/site-packages/django/template/base.py", line 162, in _render
    return self.nodelist.render(context)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/local/lib/python3.11/site-packages/django/template/base.py", line 938, in render
    bit = node.render_annotated(context)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/local/lib/python3.11/site-packages/django/template/base.py", line 905, in render_annotated
    return self.render(context)
           ^^^^^^^^^^^^^^^^^^^^

  File "/usr/local/lib/python3.11/site-packages/django/template/loader_tags.py", line 150, in render
    return compiled_parent._render(context)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/local/lib/python3.11/site-packages/django/template/base.py", line 162, in _render
    return self.nodelist.render(context)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/local/lib/python3.11/site-packages/django/template/base.py", line 938, in render
    bit = node.render_annotated(context)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/local/lib/python3.11/site-packages/django/template/base.py", line 905, in render_annotated
    return self.render(context)
           ^^^^^^^^^^^^^^^^^^^^

  File "/usr/local/lib/python3.11/site-packages/django/template/loader_tags.py", line 62, in render
    result = block.nodelist.render(context)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/local/lib/python3.11/site-packages/django/template/base.py", line 938, in render
    bit = node.render_annotated(context)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/local/lib/python3.11/site-packages/django/template/base.py", line 905, in render_annotated
    return self.render(context)
           ^^^^^^^^^^^^^^^^^^^^

  File "/usr/local/lib/python3.11/site-packages/django/template/defaulttags.py", line 315, in render
    return nodelist.render(context)
           ^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/local/lib/python3.11/site-packages/django/template/base.py", line 938, in render
    bit = node.render_annotated(context)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/local/lib/python3.11/site-packages/django/template/base.py", line 905, in render_annotated
    return self.render(context)
           ^^^^^^^^^^^^^^^^^^^^

  File "/usr/local/lib/python3.11/site-packages/django/template/defaulttags.py", line 214, in render
    nodelist.append(node.render_annotated(context))
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/local/lib/python3.11/site-packages/django/template/base.py", line 905, in render_annotated
    return self.render(context)
           ^^^^^^^^^^^^^^^^^^^^

  File "/usr/local/lib/python3.11/site-packages/django/template/defaulttags.py", line 315, in render
    return nodelist.render(context)
           ^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/local/lib/python3.11/site-packages/django/template/base.py", line 938, in render
    bit = node.render_annotated(context)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/local/lib/python3.11/site-packages/django/template/base.py", line 905, in render_annotated
    return self.render(context)
           ^^^^^^^^^^^^^^^^^^^^

  File "/usr/local/lib/python3.11/site-packages/django/template/defaulttags.py", line 449, in render
    url = reverse(view_name, args=args, kwargs=kwargs, current_app=current_app)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/local/lib/python3.11/site-packages/django/urls/base.py", line 86, in reverse
    return resolver._reverse_with_prefix(view, prefix, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/local/lib/python3.11/site-packages/django/urls/resolvers.py", line 698, in _reverse_with_prefix
    raise NoReverseMatch(msg)

```

### DSMR-reader version

v5.10.3

### DSMR-reader platform

Docker (e.g. Xirixiz)

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                    v5.10.3 / v3.11.2 / postgresql
    BE sleep / DL sleep / Retention / Override                                       2.0s / 0.5s / -h / False
    Latest telegram version read / Parser settings                                                 "42" / "4"

DATA
    Telegrams total (est.)                                                                           13484217
    Consumption records electricity / gas (est.)                                              2365289 / 37729

POSTGRESQL SIZE OF LARGEST TABLES (> 500 MB)
    public.dsmr_datalogger_dsmrreading                                                                2525 MB
    public.dsmr_consumption_electricityconsumption                                                     803 MB
```

dennissiemensma, 2023-04-22:
In de admin kun je nu de notificaties sinds kort beheren bij weergave instellingen in een apart scherm. Wellicht staat er nog een oude die incompatible is met "v4-upgrade-redirect". Het is vast de notificatie die bij v4.20 aangemaakt wordt om naar v5 te gaan. Als je die kan vinden kun je kijken of je die daar kan verwijderen.

dennissiemensma, 2023-07-21:
Oorzaak zit in [dd4e9b81](https://github.com/dsmrreader/dsmr-reader/commit/dd4e9b819724dade0b516f4a7c6fe9bf29924e87#diff-8fad06ef1878b82cb4b9d4e1cc13ade06fa5f180d8a27e369f312517a6a8377c). Ik zal voor een volgende release dat ding weer herstellen, dattie iig niet meer stuk gaat.

closed 2023-07-21 by dennissiemensma