dsmrreader/dsmr-reader #1842: API Endpoint: contract aanpassimng💡

hrikken, 2023-04-23:
### Feature

Het zou een mooie toevoeging zijn als het mogelijk is om via de API dynamische prijzen te kunnen opvoeren. Die prijzen wisselen per uur. Ik vermoed dat het ook mogelijk is om de contract records in de DB (public.dsmr_consumption_energysupplierprice) aan te passen; dat zou bv met NodeRed icm de ENTSOE API kunnen, echter ik weet niet of ik daar business logic in de DSMR Reader mee verstoor. Een API endpoint zou het mooiste zijn.

dennissiemensma, 2023-04-23:
Bedankt voor je verzoek. Helaas valt dit buiten de scope van het project. Zie ook alles qua prijzen in o.a.:
- #1708
- #1493
- #573
- #1729 
- #1607 
- #1694 
- #1692

closed 2023-04-23 by dennissiemensma

hrikken, 2023-04-24:
> Bedankt voor je verzoek. Helaas valt dit buiten de scope van het project. Zie ook alles qua prijzen in o.a.:
> 
> * [Mogelijkheid tot definiëren prijsverschil voor verbruik 'tot' en gebruik 'vanaf' #1708](https://github.com/dsmrreader/dsmr-reader/issues/1708)
> * [Energiecontract aparte invoer vaste kosten gas/stroom #1493](https://github.com/dsmrreader/dsmr-reader/issues/1493)
> * [Toevoegen vaste kosten levering #573](https://github.com/dsmrreader/dsmr-reader/issues/573)
> * [tibber integratie💡/ dynamische prijzen #1729](https://github.com/dsmrreader/dsmr-reader/issues/1729)
> * [💡 Feature request: variabele energiecontracten van bijv ANWB #1607](https://github.com/dsmrreader/dsmr-reader/issues/1607)
> * [Prijsberekening van kosten van teruglevering stroom boven verbruik💡 #1694](https://github.com/dsmrreader/dsmr-reader/issues/1694)
> * [50% teruglever vergoeding💡 #1692](https://github.com/dsmrreader/dsmr-reader/issues/1692)

Ik heb inmiddels een oplossing (workaround!) gemaakt mbv NodeRed. Een update van de e-prijzen in het contract-record bij de start van ieder uur, waarbij ik gebruikmaak van de API van Energyzero (ANWB doet dat vlg mij ook). Ik laat vaste kosten en inkoopkosten achterwege (alleen kale marktprijzen incl. Btw dus). Lijkt verder goed te werken maar blijft natuurlijk een hack. Interesse? Let me know.

dennissiemensma, 2023-04-24:
Bedankt voor je aanvulling! Ik weet alleen niet of je workaround gaat werken, want DSMR-reader ondersteunt geen uurprijzen. Vermoedelijk pakt DSMR-reader in jouw geval de prijs op het moment dat de statistieken gegenereerd worden (na middernacht).

Ik zou ooit nog de uurstatistieken (achteraf) bewerkbaar kunnen maken via de API, alleen loop je dan tegen bovenstaande issues aan, gezien DSMR-reader daar niet op rekent. 
Het komt er allemaal op neer dat het exact matchen van prijzen (of de berekeningen) niet als doel van DSMR-reader is.