dsmrreader/dsmr-reader #1843: 🙋InfluxDB Measurements pending

solidminds, 2023-04-25:
### Description

Mijn InfluxDB server is down gegaan en ik heb besloten om geen InfluxDB meer te gebruiken voor DSMR reader. Nu zijn er blijkbaar toch een hoop metingen opgeslagen in DMSR reader die nu klaar staan voor verzending. Pending measurements. En ik krijg hier eenmelding van bij het inloggen.

Nu zijn dit voor twee dagen meer dan 45.000 metingen welke verdeeld zijn over pagina's met 100 metingen. Deze kan ik selecteren en vervolgens verwijderen maar helaas enkel per 100 stuks, of per 45.000. In dat laatste geval loopt DSMR reader vast met een timeout. 
Aangezien het onbegonnen werk is om meer dan 450x een pagina van 100 metingen handmatig leeg te gooien (= 3 keer klikken en wachten) vraag ik me af of hier geen simpelere oplossing voor is. Staan dezemetingen in een database of een directory die ik handmatig leeg kan gooien?

Alvast bedankt.

Solidminds

### DSMR-reader version

v5.10.3

### DSMR-reader platform

Other

dennissiemensma, 2023-04-25:
Je kunt dit proberen:

```
sudo su - postgres
psql dsmrreader

delete from dsmr_influxdb_influxdbmeasurement;
```

dennissiemensma, 2023-04-25:
Na afloop kun je nog een vacuumdb doen, al zal het niet veel schijfruimte opleveren. Het kan hoe dan ook geen kwaad:

```
sudo su - postgres
vacuumdb -f -v -z -d dsmrreader
```

closed 2023-07-21 by dennissiemensma