dsmrreader/dsmr-reader #1844: 🚨Gasverbruik en Stroomverbruik in CSV-export (dag-totalen) kloppen niet

jcjdammers, 2023-04-26:
### Description

Ik maak elke maand een CSV-export met dagtotalen van stroom- en gasverbruik en tel die bij elkaar op voor de maandtotalen.
Echter de totalen van de CSV-export en ook het getoonde gasverbruik in het maand Archief scherm kloppen niet.
Het gasverbruik en stroomverbruik in de Home Assistant Energy Dashboard (maandtotalen) kloppen echter wel, met de gegevens van Oxxio en mijn eigen handmatige meteropname.

Zie de bijgevoegde bestanden:
dsmrreader-2023-03.xlsx ==> CSV-export van DSM-reader met dagtotalen van de maand Maart 2023.
oxxio-2023-03.xlsx 	==> CSV-export van Oxxio met dagtotalen van de maand Maart 2023.
DSM-reader Archief Maart 2023.png ==> DSM-reader Archief scherm van de maand Maart 2023.
HA-Stroom-Maart2023.jpg ==> HA energy dashboard met totaal stroomverbruik van Maart 2023.
HA-Gas-Maart2023.jpg ==> HA energy dashboard met totaal gasverbruik van Maart 2023.

Wat is er mis? Graag uw hulp.
Alvast bedankt.
NB. Ik heb DSM-reader geinstalleerd als add-on binnen HAOS.

[image]
[dsmrreader-2023-03.xlsx](https://github.com/dsmrreader/dsmr-reader/files/11334923/dsmrreader-2023-03.xlsx)
[image]
[image]
[oxxio-2023-03.xlsx](https://github.com/dsmrreader/dsmr-reader/files/11334924/oxxio-2023-03.xlsx)

### DSMR-reader version

5.10.3

### DSMR-reader platform

Other

dennissiemensma, 2023-04-26:
Vermoedelijk gerelateerd aan #1811 en #1770

jcjdammers, 2023-04-27:
Ja daar lijkt het wel op.
Hopelijk kan dit worden opgelost.

massis87, 2023-07-19:
ik sluit me aan bij gctwnl: Ik slaagde er maar niet in om mijn kosten te laten overeenkomen met de werkelijke facturen.
Tot bleek dat meterstand bij eind van de maand (2080.393) min meterstand bij eind vorige maand (2052.192) flink afwijkt, want het is werkelijk 28+m³, maar DSMR zegt een maandverbruik van 20m³. Dit is structureel elke maand flink ernaast, zowel in het Archive scherm op maandniveau, als in het Energy Contracts scherm waar mijn contracten per maand (maandelijks variabele prijs) in staan...

dennissiemensma, 2023-07-19:
Dit is helaas een gevolg van https://github.com/dsmrreader/dsmr-reader/issues/1844#issuecomment-1523766555. Ik kom echter in de maanden buiten de winter om hier wat lastig aan toe, omdat het niet een kleine fix is.

massis87, 2023-07-19:
> Dit is helaas een gevolg van [#1844 (comment)](https://github.com/dsmrreader/dsmr-reader/issues/1844#issuecomment-1523766555). Ik kom echter in de maanden buiten de winter om hier wat lastig aan toe, omdat het niet een kleine fix is.

Alle begrip daarvoor, ik wilde maar bevestigen dat het nog steeds speelt :-)

Ps: handig misschien om mee rekening te houden: in België wordt steeds de gas stand van 06:00 gebruikt door netbeheerder Fluvius als ze gaan rekenen, nooit dat van middernacht...

dennissiemensma, 2023-07-21:
Ik sluit deze zodat alles via #1770 gaat.

# Zie #1770

closed 2023-07-21 by dennissiemensma

jcjdammers, 2023-07-21:
Hi Dennis,
Ik snap dat het meer tijd kost, maar ik hoop dat het later dit jaar kan worden opgelost. Het maken van dag/maand totalen is namelijk een belangrijke functie voor mij, en ik zou het jammer vinden als dit niet correct werkt in zo'n goed programma. Alvast bedankt.