dsmrreader/dsmr-reader #1853: 🙋2e meter via api intergreren

onsmam, 2023-05-13:
### Description

Hi,

ik sinds kort een 2e meter voor alleen de teruglevering.
Nu heb ik al een paar issues gevonden dat dsmr reader geen 2e meter/connectie ondersteund, met als antwoord dat het via de api ingevoerd kan worden.

mijn vraag is, 
Moet ik beide telegrams samen voegen en een samengevoegd "telegram" naar DSMR reader sturen.
of kan ik bijv, de verbruiks kabel laten zitten en de teruglevering via de api invoeren of gaat dit dan fout?

### DSMR-reader version

5.10.3

### DSMR-reader platform

Native (e.g. manual installation)

dennissiemensma, 2023-05-13:
Het makkelijkste is om de v2 api te gebruiken om een meting aan te maken. Die api staat losse velden toe ipv een heel telegram. Mogelijk maakt dat het samenvoegen wat makkelijker voor je.

onsmam, 2023-05-15:
gaat het dan goed als ik alleen de teruglevering via de API injecteer?

als ik het goed gezien heb moet ik de meterstanden via de api doorgeven?

dennissiemensma, 2023-05-15:
Je zult beiden (verbruik en teruglevering) in 1 request moeten sturen. Het zijn inderdaad de meterstanden.

closed 2023-07-21 by dennissiemensma