dsmrreader/dsmr-reader #1857: In PVOutput zie ik niet alle gegevens (energy opgeleverd)  🙋

eissens, 2023-05-21:
### Description

In PVOutput zie ik niet alle gegevens, zoals de opgewekte hoeveelheid energie (energy) .

Ik zie wel data er data geupload wordt, maar alleen de kolommen Date, Time, Energy Used, Power Used worden gevuld op PVOutput. 
[image]

### DSMR-reader version

5.10.3

### DSMR-reader platform

Native (e.g. manual installation)

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                     v5.10.3 / v3.9.2 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                             428764
    Consumption records electricity / gas (est.)                                                48367 / 16228
```

dennissiemensma, 2023-05-21:
Bedankt voor je melding. De slimme meter rapporteert alleen netto teruglevering en meer niet. Zie ook #9 voor de hele implementatie.

eissens, 2023-05-21:
Okee, dus alles klopt verder qua werking. Ik zag al wat meldingen van eerdere versie, maar zag geen problemen bij mij, qua meldingen, dus maar ff vragen hoe het zit. Nou alles is in orde, thx voor de snelle reactie!

closed 2023-05-21 by eissens