dsmrreader/dsmr-reader #1858: Geen berekening dagtotalen🚨

brugk000, 2023-05-21:
### Description

DSMR berekent sinds gisteren geen dagtotalen meer.

### DSMR-reader version

5.10.3

### DSMR-reader platform

Native (e.g. manual installation)

### Debug info dump

```shell
2023-05-21 13:12:27,411 ERROR    apps         _on_electricity_consumption_created_signal   99 | publish_json_period_totals() failed: An error occurred in the current transaction. You can't execute queries until the end of the 'atomic' block.
2
```

brugk000, 2023-05-21:
2023-05-21 13:51:49,532 ERROR    apps         _on_electricity_consumption_created_signal   99 | publish_json_period_totals() failed: An error occurred in the current transaction. You can't execute queries until the end of the 'atomic' block.
2023-05-21 13:51:49,533 ERROR    apps         _on_electricity_consumption_created_signal  104 | publish_split_topic_period_totals() failed: An error occurred in the current transaction. You can't execute queries until the end of the 'atomic' block.
2023-05-21 13:51:49,538 DEBUG    services     compact                         211 | Compact: Processed reading: 83948975 @ 2023-05-21 13:51:53+02:00
2023-05-21 13:51:49,541 DEBUG    schedule     reschedule                       84 | SP: Rescheduled "Generate consumption data" to 2023-05-21 13:51:50.538495+02:00 (ETA 0:00:00.999983)
2023-05-21 13:51:49,546 DEBUG    mixins       run_loop                         79 | dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0

dennissiemensma, 2023-05-21:
Bedankt voor je melding. Slaat DSMR-reader verder wel de metingen op, voor zover je kunt zien? Ik heb deze fout zelf nog niet eerder gezien.

brugk000, 2023-05-22:
Ik heb ondertussen een nieuwe installatie gemaakt op basis van Docker. Backup teruggezet en alles loopt weer goed.

dennissiemensma, 2023-05-23:
Fijn om te horen en bedankt voor je terugkoppeling. Houd het wel even in de gaten en zorg dat je goede backups blijft maken, mochten er hardware issues spelen.

closed 2023-05-24 by dennissiemensma