dsmrreader/dsmr-reader #1862: Postgesql error🙋

brugk000, 2023-05-31:
### Description

Ik krijg deze foutmelding:
2023-05-31 09:02:36.815 CEST [73062] ERROR:  nextval: reached maximum value of sequence "dsmr_mqtt_message_id_seq" (2147483647)

### DSMR-reader version

docker tatest

### DSMR-reader platform

Docker (e.g. Xirixiz)

brugk000, 2023-05-31:
Ik ben wat verder in de foutmeldingen gedoken en denk dat het te maken heeft met het MAXvalue van de in de fout genoemde sequence. Heb nu mijn mqtt verbinding uitgevinkt en kijk of dat misschien een oplossing biedt

dennissiemensma, 2023-05-31:
Bedankt voor je melding. Je bent de eerste die hier tegenaan loopt en dat zou betekenen dat je al ruim 2 miljard MQTT berichten via DSMR-reader hebt verstuurd. Ik denk dat de fix voor nu is de tabel leeggooien en de sequence (ID teller) weer te resetten naar 1. Ik heb alleen niet zo 1-2-3 de betreffende queries bij de hand voor je. Daar zou ik thuis naar moeten zoeken en proberen.

brugk000, 2023-05-31:
Is gelukt. pgAdmin4 gebruikt om de database te benaderen. Tabel leeggemaakt en de teller op 1 gezet.
Zal kijken of dit de oplossing is

closed 2023-05-31 by brugk000

dennissiemensma, 2023-05-31:
Ik neem aan dat het gelukt is?

Gebruik je trouwens MQTT in DSMR-reader met een per-veld topic? Want dat zou wel deels verklaren dat je aan de limiet komt, al is 2+ miljard berichten echt heel heel veel

dennissiemensma, 2023-05-31:
Wellicht dat ik dan DSMR-reader moet overzetten naar UUID's ipv numerieke ID's. Dat is wel meer best practice, maar helaas ook niet heel triviaal om "even" te doen. Wellicht ooit.

786534786, 2023-06-09:
Ik heb nu ook dit issue. 
Het zou enorm helpen als iemand de queries heeft om de betreffende tabel leeg te maken en de sequence naar 1 te zetten. Bedankt alvast.

dennissiemensma, 2023-06-09:
Is het gelukt @786534786?

786534786, 2023-06-09:
> Is het gelukt @786534786?

Ja met statement zoals toegevoegd in mijn post. Wellicht heeft iemand anders er ook nog iets aan.

brugk000, 2023-06-10:
Bedankt voor de query. Ik deed het nogal omslachtig via pgAdmin4.

brugk000, 2023-06-14:
Terug van vakantie en de query geprobeerd in een sql window in pgAdmin4.
Deze moest in uitbreiden met het schema in de database: TRUNCATE dsmrreader.public.dsmr_mqtt_message RESTART IDENTITY;