dsmrreader/dsmr-reader #1871: 🙋

style2k, 2023-06-19:
### Description

[image]

wat is dit ? waarom worden er geen backups meer gemaakt op dropbox?????

[image]

en dit staat er ook al een hele tijd terwijl ik die optie aangepast heb ????????

### DSMR-reader version

 v5.10.3

### DSMR-reader platform

Native (e.g. manual installation)

dennissiemensma, 2023-06-20:
Je kunt Dropbox opnieuw instellen als die ontkoppeld raakt

dennissiemensma, 2023-06-20:
Voor dataopschoning zie https://github.com/dsmrreader/dsmr-reader/issues/1854#issuecomment-1546628664

closed 2023-06-21 by dennissiemensma