dsmrreader/dsmr-reader #1872: 🙋

devano1701, 2023-06-24:
### Description

Probleem: bij Dashboard blijft Teruglevering op nul staan.
Hoe kan ik dat verhelpen?

 
[Probleem met dsmr.docx](https://github.com/dsmrreader/dsmr-reader/files/11857210/Probleem.met.dsmr.docx)
 

### DSMR-reader version

v5.10.3

### DSMR-reader platform

Docker (e.g. Xirixiz)

dennissiemensma, 2023-06-26:
Bedankt voor je melding. Ik kan je bijlage helaas niet openen. Wellucht kun je een gewone screenshot maken?

Daarnaast is het handig als je een telegram meestuurt. Die vind je op de Support-pagina: https://dsmr-reader.readthedocs.io/en/v5/explained/about.html#support

devano1701, 2023-06-26:
Beste Dennis,

Misschien lukt het zo.

￼

Voor een telegram moet je even wachten tot morgen overdag wanneer de zonnepanelen energie opleveren.

mvg

Daaf

devano1701, 2023-06-27:
Het lijkt erop dat dit ook niet lukt.
Ik probeer een pdf.

￼

> Begin doorgestuurd bericht:
> 
> Van: ***@***.***" ***@***.***>
> Datum: 26 juni 2023 om 23:21:44 CEST
> Aan: dsmrreader/dsmr-reader ***@***.***>
> 
> Beste Dennis,
> 
> Misschien lukt het zo.
￼

devano1701, 2023-06-27:
Een pdf lukt ook niet.
Ik ga het anders proberen.

dennissiemensma, 2023-06-27:
Je kunt het beste even een meting in DSMR-readings openklikken om te zien of ook de huidige teruglevering/verbruik daarin te zien zijn. Die zouden overeen moeten komen met de metingen. Zo niet, dan zit daar wellicht wat.

Je telegram zelf lijkt namelijk wel te parsen met als resultaat:
```
{   
    'electricity_currently_delivered': Decimal('0.410'),
    'electricity_currently_returned': Decimal('0.000'),
    'electricity_delivered_1': Decimal('8147.640'),
    'electricity_delivered_2': Decimal('11071.698'),
    'electricity_returned_1': Decimal('604.208'),
    'electricity_returned_2': Decimal('1213.540'),
    'extra_device_delivered': Decimal('9304.255'),
    'extra_device_timestamp': datetime.datetime(2023, 6, 27, 9, 0, tzinfo=<UTC>),
    'id': 1,
    'phase_currently_delivered_l1': None,
    'phase_currently_delivered_l2': None,
    'phase_currently_delivered_l3': None,
    'phase_currently_returned_l1': None,
    'phase_currently_returned_l2': None,
    'phase_currently_returned_l3': None,
    'phase_power_current_l1': None,
    'phase_power_current_l2': None,
    'phase_power_current_l3': None,
    'phase_voltage_l1': None,
    'phase_voltage_l2': None,
    'phase_voltage_l3': None,
    'processed': False,
    'timestamp': datetime.datetime(2023, 6, 27, 19, 50, 16, 654300, tzinfo=<UTC>)}
```

devano1701, 2023-06-27:
[image]
[image]

Wel verschil in meterstanden, maar geen teruglevering.
In de andere dsmr zie ik wel teruglevering.

dennissiemensma, 2023-06-28:
Vreemd, ik zie beiden niet en eentje zou altijd een positief getal moeten zijn.

Heb je je datalogger-configuratie wel op DSMR v2/v3 staan? Die had ik nodig om je telegram te parsen, maar kan me bijna niet voorstellen dat het daar aan ligt. Meestal komt er dan helemaal niets uit als die niet compatible is.

devano1701, 2023-06-28:
Configuratie staat ook v2/v3.
Ik zie wel regelmatig staan dat het proces een herstart vereist.

[image]

Ik ging ervan uit dat de waarde berekend werd op basis van de meterstanden. Nu begrijp ik dat de meter deze waarden levert. Kan het aan mijn meter liggen?

dennissiemensma, 2023-06-29:
Die herstart is alleen nodig als je de instellingen aanpast, dus meestal is die zoals in je screenshot en dat is goed.

dennissiemensma, 2023-06-29:
Kun je anders nog een telegram delen? Het hoeft niet perse eentje van overdag te zijn.

devano1701, 2023-06-29:
/XMX5XMXABCD100007286

0-0:96.1.1(4B383257303039373436333035373131)
1-0:1.8.1(08150.315*kWh)
1-0:1.8.2(11076.013*kWh)
1-0:2.8.1(00604.212*kWh)
1-0:2.8.2(01232.571*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(0000.21*kW)
1-0:2.7.0(0000.00*kW)
0-0:96.13.1()
0-0:96.13.0()
0-1:96.1.0(3238303131303031313434333233343131)
0-1:24.1.0(03)
0-1:24.3.0(230629210000)(2C)(60)(1)(0-1:24.2.0)(m3)
(09305.248)
!

devano1701, 2023-06-29:
teruglevering staat altijd op nul.

[image]

devano1701, 2023-06-29:
op mijn andere dsmr (van de vve via remote logger en andere meter) zie ik wel teruglevering

[image]

dennissiemensma, 2023-07-02:
Ik het telegram hierboven zie ik geen huidige teruglevering (kW):
```
1-0:1.7.0(0000.21*kW)
1-0:2.7.0(0000.00*kW)
```

Alleen huidig verbruik (kW). Dus dat verklaart het wel, alleen zou ik wel verwachten dat de onderste teller overdag een keertje oploopt.

Of je moet dusdanig veel stroom gebruiken tijdens de opwekking, dat je niets teruglevert aan het net. Maar dat lijkt me ook vreemd, gezien je kWh tellers voor teruglevering wel wat hebben staan.

devano1701, 2023-07-03:
Ik heb op een andere manier eens gekeken naar het telegram vanaf mijn meter.
Dan zie ik dit:

b'\xaf\xd8M\xd85\xd8M\xd8AB\xc3D\xb10000\xb7\xb2\xb86\x8d\n'
b'\x8d\n'
b'0-0:96.\xb1.\xb1(\xb4B3\xb83\xb25\xb73030393\xb73\xb4363330353\xb73\xb13\xb1\xa9\x8d\n'
b'\xb1-0:\xb1.\xb8.\xb1(0\xb8\xb155.50\xb7\xaa\xeb\xd7\xe8\xa9\x8d\n'
b'\xb1-0:\xb1.\xb8.\xb2(\xb1\xb10\xb76.\xb853\xaa\xeb\xd7\xe8\xa9\x8d\n'
b'\xb1-0:\xb2.\xb8.\xb1(006\xb20.\xb7\xb4\xb8\xaa\xeb\xd7\xe8\xa9\x8d\n'
b'\xb1-0:\xb2.\xb8.\xb2(0\xb1\xb2\xb4\xb4.\xb1\xb29\xaa\xeb\xd7\xe8\xa9\x8d\n'
b'0-0:96.\xb1\xb4.0(000\xb1\xa9\x8d\n'
b'\xb1-0:\xb1.\xb7.0(0000.00\xaa\xeb\xd7\xa9\x8d\n'
b'\xb1-0:\xb2.\xb7.0(0000.00\xaa\xeb\xd7\xa9\x8d\n'
b'0-0:96.\xb13.\xb1(\xa9\x8d\n'
b'0-0:96.\xb13.0(\xa9\x8d\n'
b'0-\xb1:96.\xb1.0(3\xb23\xb8303\xb13\xb130303\xb13\xb13\xb43\xb4333\xb2333\xb43\xb13\xb1\xa9\x8d\n'
b'0-\xb1:\xb2\xb4.\xb1.0(03\xa9\x8d\n'
b'0-\xb1:\xb2\xb4.3.0(\xb230\xb70\xb2\xb160000\xa9(0\xb8\xa9(60\xa9(\xb1\xa9(0-\xb1:\xb2\xb4.\xb2.0\xa9(\xed3\xa9\x8d\n'
b'(09306.009\xa9\x8d\n'
b'!\x8d\n'

Je moet het een beetje 'vertalen'. Het is met veel zon en dus veel opgewekte energie. 
Ik zie op eerdere en latere telegrammen ook dat de terugmeter oploopt.
Maar het vermogen blijft op 0 kW. Zowel opname als teruggave.
De kans daarop is eigenlijk nihil (in ieder geval bij een 1-fase meter).

Het lijkt er dus op dat de meter het niet goed doet.
Maar hoe vertel ik dat aan Stedin.

dennissiemensma, 2023-07-03:
Klopt het dat je een 1-fase meter hebt?

Want als je altijd "verbruik" hebt dan lijkt het erop dat je al je opwekking in je huishouden direct opneemt (en iets meer nodig hebt wat je dan afneemt van het net). Puur gebaseerd op de twee leesbare telegrammen eerder.

Als de beide kW-tellers op nul staan (constant) dan lijkt het me iets met je meter. Alleen is het dan wel goed om een "leesbaar" telegram overdag te hebben die dat heeft.

devano1701, 2023-07-04:
Klopt: ik heb een 1-fase meter (een Landis+Gyr model ZCF120ABd uit 2011).

's Avonds staat de teruglevering op 0,00 kW en is er opname zichtbaar.
Overdag (met zon) staan beide kW-tellers op 0,00.
De kWh-tellers zijn wel goed.

Mijn conclusie, mede na jouw ondersteuning, is dat het aan mijn meter ligt en niet aan de DSMR-reader.
Ik ga maar eens contact opnemen met Stedin.

Wat mij betreft mag je dit afsluiten.
Bedankt voor je hulp.

dennissiemensma, 2023-07-05:
Dank voor alle info. Je bent de eerste waarbij ik dit zie, dus ik heb geen verdere suggesties meer voor je. Hopelijk kan je netbeheerder wat voor je doen. 

Laat het vooral hier weten als ze (iets anders) hebben gedaan voor je. Ook voor eventuele anderen die van hetzelfde last hebben.

closed 2023-07-05 by dennissiemensma

devano1701, 2023-08-30:
De meter is vervangen!
Nadat wij tot de conclusie waren gekomen dat het niet aan de DSMR software lag maar aan de meter, heb ik contact opgenomen met Stedin. Die begrepen niet direct wat mijn probleem was. Ik moest maar een afspraak maken met een monteur.
De monteur is onlangs langs geweest. Hij begreep direct mijn probleem, keek naar de meter (Landis + Gyr ZCF120ABd M40 met DSMR 2.2+): "Ja, dit is een oud model (model 40) en die kan dat niet." Ik heb een nieuwe meter gekregen: een Landis + Gyr E360 SMR 5.0 (en tevens een nieuwe gasmeter i.v.m. de koppeling tussen de meters). 
In eerste instantie zag ik geen telegrammen uit de meter komen. Op internet lees je dan dat een E360 een ander P1-USB kabeltje vereist. Ik heb zo’n nieuw kabeltje besteld, maar deze bleek achteraf toch niet nodig te zijn. Na een paar reboots werkte het weer met het oude kabeltje. Uiteindelijk wel het nieuwe kabeltje geplaatst.
Alles werkt nu dus weer zoals verwacht.

dennissiemensma, 2023-08-30:
Fijn om te horen, dank voor je update!