dsmrreader/dsmr-reader #1881: 🙋dsmr reader resetten

dennispracht, 2023-07-18:
### Description

Goedemorgen,

Ik gebruik dsmr reader nu voor 2 maanden en het werkt echt top. bedankt daarvoor. 

afgelopen zaterdag is mijn nieuwe contract ingegaan voor 1 jaar. In dsmr kan je idd heel netjes een contract jaar aanmaken en hierop zie ik ook bij contract netjes dat die de dagen bij elkaar optelt. Maar ik wil alle waardes in dsmr resetten naar nul. Dit omdat ik in hoas een script heb die doormiddel van dsmr uit kan rekenen hoeveel ik per dag per maand en per jaar kwijt ben aan kosten en de zonnepanelen en verbruik van elkaar aftrekt.

Nu zit ik met nieuwe contract en die 2 maanden en dat is voor de berekening niet juist.

Is er niet ergens in dsmr een reset knop die zorgt dat ik vanaf begin opnieuw start. Mocht het er wel zijn. Ik kan het niet vinden.

Bedankt

Groeten, 

Dennis

### DSMR-reader version

5

### DSMR-reader platform

Native (e.g. manual installation)

dennissiemensma, 2023-07-19:
Als je de contractprijzen op 0 zet, kun je dit daarna proberen: https://dsmr-reader.readthedocs.io/en/v5/how-to/troubleshooting/recalculate-prices.html

Zo niet, dan zal ik kijken hoe je dit via een query in de database kan doen.

dennispracht, 2023-07-19:
Bedankt, Dat zou top zijn Dennis. Je addon werkt echt super goed. Heb je goed gedaan. 

Nu alleen nog even dat ik alles op 0 kan zetten en dan.

dennissiemensma, 2023-07-19:
Werkt het command uit bovenstaande link dan niet? Als je de contractprijzen niet meer nodig hebt in DSMR-reader, zou ik ze gewoon allemaal verwijderen.

dennispracht, 2023-07-19:
Ja ik heb die contract prijzen juist wel nodig. Het systeem werkt super. Heb het nu 2 maanden draaien. Maar afgelopen week ging me nieuwe contract jaar in en heb dus in dsmr een contract aangemaakt. Maar de waardes van die afgelopen 2 maanden blijven staan. Heb alleen een klein stukje waar die de kwh en gas laat zien die vallen binnen het contract wat ik ingevoerd heb. 
[image]

[image]

Zoals je ziet blijft die waardes van afgelopen 2 maanden doorlopen. Dat wil ik juist allemaal op nul. Want nu kan ik niet zien hoeveel geld het jaar daadwerkelijk kost. Kijk maar op de laatste regel voorschot stroom & gas realtime. Door verbruikte kosten huidig jaar van dsmr te gebruiken en die van de voorschot af te trekken. Zie ik altijd hoeveel voorschot er nog is door het jaar heen. 

Die wil ik nog verder uit bereiden met een voorspelling. Maar dat gaat niet omdat die bij het nieuwe contract datum niet op 0 gaat. Maar door loopt met de kuub en kwh.

dennissiemensma, 2023-07-20:
Bedankt voor je aanvulling. Ik begrijp echter nog niet helemaal wat je wilt. Wil je de contractprijzen op nul of de dagprijzen (Archief + Energiecontracten) van de afgelopen maanden op nul of de dagprijzen (Archief + Energiecontracten) van de afgelopen maanden herberekenen of iets anders?

dennispracht, 2023-07-20:
Het archief op nul. Echt het begin. Net zoals je dsmr voor de eerste keer heb geïnstalleerd. 

Wil schoon schip. Zodat mijn contract jaar klopt qua berekening wat ik dar jaar echt heb verbruikt. Nu zit de afgelopen 2 maanden erbij en dat wil ik verwijderen.

dennissiemensma, 2023-07-20:
Zie de volgende link. Echter gooi je daar dus alle data mee weg, niet alleen de prijzen, ook de metingen en het archief!  https://github.com/dsmrreader/dsmr-reader/issues/809#issuecomment-568563028

dennispracht, 2023-07-20:
Dat is juist de bedoeling voor deze keer. Factory reset zegmaar. 

Bedankt voor je reacties

dennissiemensma, 2023-07-21:
Prima. Deze is zelfs nog wat recenter: https://github.com/dsmrreader/dsmr-reader/issues/1836#issuecomment-1510386355

closed 2023-07-21 by dennissiemensma

dennispracht, 2023-07-31:
Hoi Dennis,

Ik heb het geprobeert. Maar alles blijft gewoon erin staan. Heb zelf het stukje gedaan van jou handleiding waarbij staat warning hierdoor verwijder je alle data. Nou niet bij mij dus. In hoas blijft alle verbruikte kwh en gas kuub gewoon staan.

Data verwijderen is goed beveiligd.

dennissiemensma, 2023-07-31:
DSMR-reader doet niets met Home Assistant. Als je de gegevens niet meer ziet in DSMR-reader maar wel in HA, dan zul je daar ook handmatig de gegevens moeten verwijderen.

dennispracht, 2023-08-02:
Ja de pest is dat de waardes gewoon in dsmr blijven staan. Ik heb die sudo su script gedaan. Maar alles blijft staan. Ook in dsmr webpagina.

dennissiemensma, 2023-08-02:
Welke output krijg je bij het uitvoeren van de delete queries?

dennispracht, 2023-08-02:
Ik krijg dit te zien als ik het invoer. Ik doe iets verkeerd.

[image]

dennissiemensma, 2023-08-02:
Die meldingen lijken goed. Wat zie je hier?

```
sudo su - postgres
psql dsmrreader

select count(id) from dsmr_datalogger_dsmrreading;
select count(id) from dsmr_consumption_electricityconsumption;
select count(id) from dsmr_consumption_gasconsumption;
select count(id) from dsmr_consumption_quarterhourpeakelectricityconsumption;
select count(id) from dsmr_datalogger_meterstatistics;
select count(id) from dsmr_stats_daystatistics;
select count(id) from dsmr_stats_electricitystatistics;
select count(id) from dsmr_stats_hourstatistics;
```

dennispracht, 2023-08-02:
Als ik naar het ip adres ga en inlog met de inloggegevens. Dan krijg ik onderstaand te zien. Die zelfde link heb ik in home assistant als sidebar. Daar blijven de gegevens staan. 

Dat is toch raar. Het is toch de zelfde ip adres waar naar gekeken wordt. Wat moet ik nu doen? Dsmr reader verwijderen in home assistant en daarna weer installeren. 

Wat ik nu zie staan als ik naar het ip adres ga. Zoals de afbeelding. Daar gaat die nu ook niet meer opnieuw het stroom en gas berekenen. Dit blijft er staan.

[image]

dennissiemensma, 2023-08-03:
Wat krijg je uit bovenstaande queries? Als de gegevens weg zijn zou je wel tijdelijk de melding uit je screenshots moeten krijgen, dat is de beginstaat

dennispracht, 2023-08-10:
als ik die queries doe dan krijgt ik dit:

 count
-------
     0
(1 row)

Maar hij is al vanaf 0 aan het tellen. Heb in dsmr alles gereset. Rare is dat als ik naar het ip adres ga dan blijft die aangeven geen gegevens gevonden. Wat jij zegt de beginstaat. Maar in home assistant heb ik dsmr als sidebar staan en daar teld die wel vanaf 0 en geeft die data door.

Het is gelukt. Super bedankt voor je input.