dsmrreader/dsmr-reader #1884: pi zero naar pi 2b+🙋

Gerkla, 2023-07-29:
### Description

Ik had een tijdlang dsmr draaien op een pi zero, maar dit is te traag. Er kwam een pi 2b+ vrij, dus die zal ik nu gebruiken.
sd kaart van pi zero in 2b+ gestopt, maar de website laadt niet. Ik krijg op de webpagina '504 gateway time-out, nginx/1.14.2'
Nagezien als dsmr en postgresql draaien en als ik telegrammen kan uitlezen. Deze zijn allemaal ok. 
Ik vermoed dat er nog ergens een configuratie aangepast moet worden?
`sudo supervisorctl status`
```
dsmr_backend                     RUNNING   pid 498, uptime 0:17:27
dsmr_datalogger                  RUNNING   pid 497, uptime 0:17:27
dsmr_webinterface                RUNNING   pid 496, uptime 0:17:27
```

`sudo systemctl is-active postgresql`
`active`

`ls -l /dev/serial/by-id/`
`total 0
lrwxrwxrwx 1 root root 13 Jul 29 10:27 usb-FTDI_FT232R_USB_UART_AQ8VI6EH-if00-port0 -> ../../ttyUSB0`

`cu -l /dev/ttyUSB0 -s 115200 --parity=none -E q`
->toont de telegrammen

raspbian buster

### DSMR-reader version

5.10.3

### DSMR-reader platform

None

dennissiemensma, 2023-07-29:
Je kunt dit nog doen/delen en de rest even doorlezen. https://github.com/dsmrreader/dsmr-reader/issues/1428#issuecomment-891129066. Wellicht zit het daar in, want een Pi 2 is relatief traag, maar is wellicht wel werkend te krijgen met een goed sleep etc.

Gerkla, 2023-07-29:
Foutje van mijnentwege bleek een rpi versie 1 b te zijn. Terug de zero in gebruik genomen en datalogger sleep op 10 gezet ipv 5 standaard. Hopelijk draait het nu wat beter. Thx voor de tip

closed 2023-07-29 by Gerkla