dsmrreader/dsmr-reader #1890: 🙋en weer dropbox problemen

style2k, 2023-08-07:
### Description

ik heb nu deze melding en er is sinds die tijd weer geen backup gemaakt. 
July 12, 2023: Unable to authenticate with Dropbox, removing credentials. Error: InternalServerError('f43aa14c647c4edf8b94472ef364d1aa', 500, '{"error": "The server has either erred or is incapable of performing the requested operation."}')

### DSMR-reader version

DSMR-reader: v5.10.3

### DSMR-reader platform

Native (e.g. manual installation)

dennissiemensma, 2023-08-07:
Foutmelding aan de kant van Dropbox, je zult weer opnieuw moeten koppelen. Het probleem is dat ik niet goed kan detecteren of een fout bij Dropbox zelf fataal is of niet. Vandaar dat je deze melding krijgt (die nog wel wat duidelijker kan).

style2k, 2023-08-07:
opnieuw gekoppeld en doet het zo te zien weer . maar is er ook een optie dat als dsmr een probleem tegen komt zoals deze melding dat die dat doorstuurt via een pushbericht ofzo? zodat je niet heel lang zonder backup zit?

dennissiemensma, 2023-08-07:
Ja wellicht ooit, maar het is allemaal niet heel triviaal om te maken en vooral ook onderhouden, dus het is soms even doorbijten

closed 2023-08-07 by dennissiemensma