dsmrreader/dsmr-reader #1892: 💡Import old / historic data: i created a repositoy for it and could you add it to the Wiki?

heyudude, 2023-08-25:
### Feature

Maybe you can add the import of old data (Domoticz or other sources) to the WIki?

I made a repository for my Python import, ease of maintenance and make it public to others.
I am sure one can use the format and code for other type of sources, either change the CSV format / content or the code

[https://github.com/heyudude/DSMR-reader-import/](url)

dennissiemensma, 2023-08-25:
Bedankt voor je aanvulling! 

Ik denk dat het het handigste is om het allemaal in de issues te houden, om te voorkomen dat specifieke vragen daarover ook weer bij mij komen en ik als tussenluik functioneer.

Als iemand later bij dit issue komt:

# Zie #520
- #520

closed 2023-08-25 by dennissiemensma

heyudude, 2023-08-25:
Prima, snap het helemaal en dank voor de software. Mooie aanvulling op de metingen in welk systeem dan ook.