dsmrreader/dsmr-reader #1899: Probleem met PostgreSQL database🙋

killabeenl, 2023-09-06:
### Description

Hallo,

Ik draai al een tijdje DSMR reader vanuit een Homeassistant addon samen met de TimescaleDB addon voor de PostgreSQL database. Nu heb sinds een aantal dagen deze meldingen in DSMR reader in de logging.

_on_dsmrreading_created_signal  159 | publish_split_topic_meter_statistics() failed: integer out of range

Dit begon op te vallen omdat de MQTT sensoren niet meer werden geupdate. Dus ik heb geen info meer van DSMR in Homeassistant. Nu lijkt DSMR Reader verder wel te werken echter het maken van Backups lukt ook niet meer.

Ik heb de retentie settings aangepast en ook een VacuumDB uitgevoerd zoals vermeld in de handleiding. Echter lost dit nog niets op.

Kan iemand mij hier bij ondersteunen. 

Alvast bedankt

### DSMR-reader version

v5.10.3

### DSMR-reader platform

Docker (e.g. Xirixiz)

killabeenl, 2023-09-06:
Nog even een aanvulling. Dit zie ik ook voorbij komen in de logging:

pg_dump: error: server version: 15.3; pg_dump version: 14.5
pg_dump: error: aborting because of server version mismatch

dennissiemensma, 2023-09-06:
Ik kan je momenteel niet helpen, omdat ik in het buitenland zit.

dennissiemensma, 2023-09-08:
Dank voor je update!