dsmrreader/dsmr-reader #1901: Naamgeving in configuratie van sorted graphs 🚨

VBP8501, 2023-09-23:
### Description

Hallo, ik heb een kleine Big gevonden denk ik in het admin gedeelte. Onder de sorted graphs zie ik een graph die "power" noemt. 

Toen ik die eerste plaatste, bleek dat de fase stromen grafiek te zijn.

### DSMR-reader version

5.10.3

### DSMR-reader platform

None

dennissiemensma, 2023-09-24:
Bedankt voor je melding. Ik weet niet precies wat je voorstel dan is. Mist er een vertaling of is er een betere term? 
In de meeste gevallen heb ik de termen uit de DSMR specs gebruikt, tenminste, als het goed is.

VBP8501, 2023-09-27:
Sorry ik was wellicht niet helemaal duidelijk in mijn originele post. De grafieken in de "live graphs" op het dashboard heten:

|  Live graphs dashboard |  Admin sorted graphs |
|---|---|
|  Recent Electricity usage | Electricity  |
|  Recent quarter hour electricity peak consumption |  Electricity quarter hour peaks |
|  Recent phase currents |  Power |
|  Recent phase usage |  Phases |
|  Recent phase voltages |  Voltage |
|  Recent gas consumption |  Gas |

De "Recent Electricity usage" is eigenlijk een vermogengrafiek. Ofwel in het engels "Power". 

De "Recent phase currents" heeft nu de naam "Power", wat niet klopt want het gaat enkel over de stroom.

Dus het is vooral verwarrend dat de naam 'power' hangt aan een stroomgrafiek en 'electricity' aan een vermogen grafiek.

dennissiemensma, 2023-10-01:
Bedankt voor je aanvulling! Ik zal er naar kijken en het evt gelijktrekken wanneer ik hier weer verder mee ga in het najaar

dennissiemensma, 2023-10-23:
Ik zal een migratie opnemen die in ieder geval de juiste termen overneem. En een extra kolom die onderwater nog wat aanduidt.

[image]

----

Ik kan niet zoveel veranderen aan de Engelse terminologie die DSMR-reader gebruikt, want dan zal ik alles moeten hernoemen om te voorkomen dat verschillende termen hetzelfde aanduiden, ondanks dat de huidige vertalingen wellicht niet ideaal zijn (deel komt uit ook DSMR specs overigens).

closed 2023-10-23 by dennissiemensma