dsmrreader/dsmr-reader #1915: Nieuwe installatie niet meer mogelijk, pyyaml 6.0 niet meer beschikbaar🚨

bertusdebruin, 2023-11-01:
### Description

Heel simpel probleempje en nog simpeler op te lossen.
De ontwikkelaar van pyyaml, lijkt pyyaml versie 6.0 (pyyaml==6.0) van het download netwerk te hebben verwijderd.
Waardoor een nieuwe dsmr-reader niet geïnstalleerd kan worden, omdat je uiteraard een dependency mist.

Oplossing is simpel verander [pyyaml==6.0 in de base.txt naar pyyaml==6.0.1](https://github.com/dsmrreader/dsmr-reader/blob/v5/dsmrreader/provisioning/requirements/base.txt)

Graag een fix zodat iedereen hier van kan genieten.

Overigens en daar kan @dennissiemensma niets aan doen. Is het bovenstaande een van de redenen dat ik een hekel heb aan de manier van dependencies distribueren zoals deze. De ontwikkelaar zoals die van pyyaml trekt de stekker eruit en daar zit je dan.

### DSMR-reader version

5.10.3

### DSMR-reader platform

Native (e.g. manual installation)

dennissiemensma, 2023-11-01:
Bedankt voor je melding! Geldt dit voor een bepaalde distro? Of was het misschien tijdelijk?

Ik kan het niet reproduceren met iets als:
```
pip install pyyaml==6.0 --force-reinstall --no-cache-dir
```

```
Collecting pyyaml==6.0
  Downloading PyYAML-6.0-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (757 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 757.9/757.9 kB 11.5 MB/s eta 0:00:00
DEPRECATION: dropbox 11.36.0 has a non-standard dependency specifier stone>=2.*. pip 24.0 will enforce this behaviour change. A possible replacement is to upgrade to a newer version of dropbox or contact the author to suggest that they release a version with a conforming dependency specifiers. Discussion can be found at https://github.com/pypa/pip/issues/12063
Installing collected packages: pyyaml
  Attempting uninstall: pyyaml
    Found existing installation: PyYAML 6.0
    Uninstalling PyYAML-6.0:
      Successfully uninstalled PyYAML-6.0
Successfully installed pyyaml-6.0
```

En ook hier zie ik hem er nog staan: https://pypi.org/project/PyYAML/6.0/#history

dennissiemensma, 2023-11-01:
Ik zie hem hier ook nog staan: https://github.com/yaml/pyyaml/tags

Al betekent dat niet dattie daarmee ook in Pypi staat, gezien dat weer een aparte storage/partij is.

bertusdebruin, 2023-11-01:
> 
> Bedankt voor je melding! Geldt dit voor een bepaalde distro? Of was het misschien tijdelijk?

Ik kreeg met geen mogelijkheid pyyaml==6.0 binnen. Pyyaml==6.0.1 daarentegen was seconde werk. 

Zou best kunnen dat het gerelateerd is aan Raspberry Pi OS 12 (bookworm), wat ik gebruik op deze nieuwe Raspberry Pi 4.
Maar ook tijdelijk dat een bepaalde CDN mirror even de geest gaf. Dit is altijd achteraf lastig te herleiden naar een probleem in dezen. 

Het heikele punt is dat ik niet meerdere Raspberry's voorradig heb om een en ander te testen. Wanneer ik nieuwe hardware tot mijn beschikking krijg en de oude niet meer van toepassing. Is het vanzelfsprekend zaak om alles zo snel, noch goed over te zetten. En om zo, zoveel mogelijk dsmr-meetdata-verlies te voorkomen. Moet je op een gegeven moment gewoon door.
 
Is het wellicht een optie om de dependencies zelf te hosten? Dan heb je dat in ieder geval de distributie van alle bestanden in eigen hand. 

Sluit ik af met het gegeven dat dsmr-reader met Pyyaml==6.0.1 prima draait. Aangezien het enkel een security update betreft. Lijkt het mij ongeacht de uitkomst van dit 'issue' een gedegen update voor de dsmr-reader.

dennissiemensma, 2023-11-01:
Bedankt voor je aanvulling!

> _Is het wellicht een optie om de dependencies zelf te hosten? Dan heb je dat in ieder geval de distributie van alle bestanden in eigen hand._

Dat is praktisch gezien geen doen (bij elke update) en ook geen best-practice. Als het Python-ecosysteem massaal stopt dan kan ik dat wel doen, maar tot nu toe heb ik maar 1x zelf een dependency hoeven op te nemen in het project. PyYaml zou dan de tweede zijn, als die helemaal niet meer beschikbaar was.

> _Lijkt het mij ongeacht de uitkomst van dit 'issue' een gedegen update voor de dsmr-reader._

Er komt sowieso nog een update die alle dependencies bumpt, alleen was die niet op korte termijn gepland, dus zoek ik even naar de oorzaak van dit specifieke issue. 

- Welke Python-versie draai je?
- Wat is de specifieke foutmelding als je `pip install pyyaml==6.0 --force-reinstall --no-cache-dir` doet?

bertusdebruin, 2023-11-02:
Heb even een Pi geleend. Raspberry Pi OS 12 (bookworm), Python 3.11.2. En nog steeds hetzelfde:

`Collecting pyyaml==6.0
  Downloading PyYAML-6.0.tar.gz (124 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 125.0/125.0 kB 1.7 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error

  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [54 lines of output]
      running egg_info
      writing lib/PyYAML.egg-info/PKG-INFO
      writing dependency_links to lib/PyYAML.egg-info/dependency_links.txt
      writing top-level names to lib/PyYAML.egg-info/top_level.txt
      Traceback (most recent call last):
        File "/home/dsmr/dsmr-reader/.venv/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/home/dsmr/dsmr-reader/.venv/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/home/dsmr/dsmr-reader/.venv/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 118, in get_requires_for_build_wheel
          return hook(config_settings)
                 ^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-kci_x2o2/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 355, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=['wheel'])
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-kci_x2o2/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 325, in _get_build_requires
          self.run_setup()
        File "/tmp/pip-build-env-kci_x2o2/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 341, in run_setup
          exec(code, locals())
        File "<string>", line 288, in <module>
        File "/tmp/pip-build-env-kci_x2o2/overlay/lib/python3.11/site-packages/setuptools/__init__.py", line 103, in setup
          return distutils.core.setup(**attrs)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-kci_x2o2/overlay/lib/python3.11/site-packages/setuptools/_distutils/core.py", line 185, in setup
          return run_commands(dist)
                 ^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-kci_x2o2/overlay/lib/python3.11/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
          dist.run_commands()
        File "/tmp/pip-build-env-kci_x2o2/overlay/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
          self.run_command(cmd)
        File "/tmp/pip-build-env-kci_x2o2/overlay/lib/python3.11/site-packages/setuptools/dist.py", line 989, in run_command
          super().run_command(command)
        File "/tmp/pip-build-env-kci_x2o2/overlay/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/tmp/pip-build-env-kci_x2o2/overlay/lib/python3.11/site-packages/setuptools/command/egg_info.py", line 318, in run
          self.find_sources()
        File "/tmp/pip-build-env-kci_x2o2/overlay/lib/python3.11/site-packages/setuptools/command/egg_info.py", line 326, in find_sources
          mm.run()
        File "/tmp/pip-build-env-kci_x2o2/overlay/lib/python3.11/site-packages/setuptools/command/egg_info.py", line 548, in run
          self.add_defaults()
        File "/tmp/pip-build-env-kci_x2o2/overlay/lib/python3.11/site-packages/setuptools/command/egg_info.py", line 586, in add_defaults
          sdist.add_defaults(self)
        File "/tmp/pip-build-env-kci_x2o2/overlay/lib/python3.11/site-packages/setuptools/command/sdist.py", line 113, in add_defaults
          super().add_defaults()
        File "/tmp/pip-build-env-kci_x2o2/overlay/lib/python3.11/site-packages/setuptools/_distutils/command/sdist.py", line 251, in add_defaults
          self._add_defaults_ext()
        File "/tmp/pip-build-env-kci_x2o2/overlay/lib/python3.11/site-packages/setuptools/_distutils/command/sdist.py", line 336, in _add_defaults_ext
          self.filelist.extend(build_ext.get_source_files())
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "<string>", line 204, in get_source_files
        File "/tmp/pip-build-env-kci_x2o2/overlay/lib/python3.11/site-packages/setuptools/_distutils/cmd.py", line 107, in __getattr__
          raise AttributeError(attr)
      AttributeError: cython_sources
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
`

dennissiemensma, 2023-11-04:
Bedankt voor je aanvulling. Ik moet hier nog wat verder naar kijken. Het bumpen van de versie breekt de CI in #1921, al is de oorzaak vast iets anders

dennissiemensma, 2023-11-20:
Tests & docs uiteindelijk kunnen fixen, onderdeel van https://github.com/dsmrreader/dsmr-reader/releases/tag/v5.10.4

closed 2023-11-20 by dennissiemensma

MrWhammy, 2023-12-23:
Niet dat het er nog toe doet, maar: het lijkt erop dat pyyaml 6.0.1 de dependency naar < cython 3 vastpint. Die [versie](https://cython.readthedocs.io/en/latest/src/changes.html#id38) is gereleaset in juli dit jaar.

[Bovenstaande error](https://github.com/dsmrreader/dsmr-reader/issues/1915#issuecomment-1790557067) is dezelfde als die in [het issue](https://github.com/yaml/pyyaml/issues/601) dat vermeld wordt in de PR voor de workaround in 6.0.1.